tagyoureit/nodejs-poolController #289: VF pump not starting

omo22, 2021-05-02:
**Describe the bug**
pump not running despite working Rs485 and app showing green when clicked

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'dash'
2. Click on 'pool circuit'
3. Scroll down to 'xx.'
4. See error  nothing happening 

**Expected behavior**
pump should start

**Screenshots**
sent with gitter chat

**Packet Capture**
Zip file created after removed RPI from pump
log packet file while connected to pump and clicked pool circuit but regardless same packet was comming over and over, clicking did not dire any packets 

**Pool Equipment**
 - Controller: [e.g. IntelliCenter, EasyTouch, IntelliTouch]        NO
 - Controller Model: [e.g. IntelliCenter i5PS, EasyTouch2 8P]  NO 
 - Controller Firmware Version: [e.g. 1.047]   NO
 - Pump(s) manufacturer and model: [e.g. IntelliFlow 2 VST 011056]  = VF 011012
 - Chlorinator: [e.g. iChlor, IntelliChlor-40]   NO
 - Heater(s): [e.g. gas, solar, heatpump, ultratemp]  NO
 - Chemical controller: [e.g. IntelliChem, homegrown]  NO
 - Valves: [e.g. Intellivalve]  NO
 - Any other relevant equipment:   NO

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]   win 10
 - Browser [e.g. chrome, safari]  chrome
 - Version [e.g. 22] 

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6] 10
 - OS: [e.g. iOS8.1] 14
 - Browser [e.g. stock browser, safari] safari
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.  NO
[packetLog(2021-05-01_19-38-30).log](https://github.com/tagyoureit/nodejs-poolController/files/6411167/packetLog.2021-05-01_19-38-30.log)
[replay (14).zip](https://github.com/tagyoureit/nodejs-poolController/files/6411168/replay.14.zip)

rstrouse, 2021-05-03:
@tagyoureit  is currently working on the Nixie pumps.  As it turns out the flow readings are not being taken into account with that model.  We should have something shortly that will send the proper messages.  It would be great if you could verify because we do not have a VF floating around,

omo22, 2021-05-03:
Sure will confirm
Thanks

tagyoureit, 2021-05-05:
@omo22 Can you pull the latest dashPanel and njsPC?  The restart the apps and go back and save the pump settings.  It will now use the new code that is added to Nixie.

omo22, 2021-05-09:
i re installed both and run it, NO changes still same no response, see packets same as before,

tagyoureit, 2021-05-09:
Ah, we punched this up to a different branch. Follow these instructions. 

In the njsPC directory, 'git checkout NixiePumps'. 

Then restart the app and try again. When we push up the code to the main branch you will run 'git checkout master' but will let you know when that will happen.

tagyoureit, 2021-05-13:
This code has now been pushed to master.  You can ignore my last comment and just pull from the main branch.

omo22, 2021-05-13:
I had master non git installed
Removed all and made it install via git
I tried git pump
Was able to turn on off pump
But not responding to change the speed
Now need trying more this weekend
Thanks

closed 2021-08-05 by tagyoureit