tagyoureit/nodejs-poolController #290: Ultratemp Cooling Mode not Showing Up

Tegguy, 2021-05-05:
Ultratemp heater cooling mode not showing up. I have the Aquacal 166R which has heat mode and cooling mode. This is setup as an ultratemp in the intellicenter panel however, the pool controller dashboard does not have the open to set a cool point that I can see.

**Pool Equipment**
Controller: IntelliCenter
Controller Model: Intellicenter i8PS
Controller Firmware Version:
Pump(s) manufacturer and model: 2X IntelliFlow 2 VST 011056
Chlorinator: IntelliChlor-40
Heater(s): AquaCal Heatpump (Heat/Cool)
Valves: Intellivalve

tagyoureit, 2021-05-05:
Can you attach a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution#-dashpanel).  This will request the new pool config from Intellicenter so we can make sure we have the right configuration bits.  We haven't come across this config yet so we'll need to investigate the packets to make sure we are capturing it correctly.

Once the config reaches 100%, can you go through the different modes with the Ultratemp making sure the heater/cooler turns on each state before moving to the next?

Repeat the following for your pool and then your spa.
1. No heat mode
2. Heat mode with setpoint below current temp
3. Heat mode with setpoint above current temp (wait for heater to turn on)
4. Cool mode with setpoint above current temp
5. Cool mode with setpoint below current temp (wait for cooler to turn on)

If you have the "heater preferred" or "cooler preferred" heat modes, please add those in the mix.

Tegguy, 2021-05-05:
Here is the capture. There is a 3 min compressor delay before turning on in each mode. I also included a screenshot of what the pentair software interface looks like to see if it helps. 
[Pentair Ultratemp.zip](https://github.com/tagyoureit/nodejs-poolController/files/6430153/Pentair.Ultratemp.zip)
[image]

Tegguy, 2021-05-12:
Don't know if this helps but NJSPC sees the "heat mode" and says cooling 

[image]

rstrouse, 2021-05-12:
Alright so I went through the messages to determine where the data is being stored and updated.  There was one thing that puzzled me though.  Each change to the setpoints were affecting the schedules.  This is a Pentair bug in firmware v1.04.

I highly recommend you upgrade your panel to 1.047.  1.04 will drive you batty as the setpoints get modified every time a schedule kicks off no matter whether you have them set to don't change or not.

https://www.pentair.com/en-us/education-support/residential/product-support/pentair-pool-and-spa-software-downloads/intellicenter-download.html

That being said I think I have enough information to build out the cooling setpoints but I have a couple of questions of what is displayed in IntelliCenter for that particular heater.

First, it looks like you do not have a solar heater so are the only heat modes that display in IntelliCenter:
1. Off
2. UltraTemp Only
Or are there more options like heat pump preferred?

Second, is there any indication in IntelliCenter which mode the heater is operating in.  I assume the AquaCal does not include a hybrid gas heat option like the UltraTemp ETi models do.  Is that the case here?

Third, do you have a solar temp sensor hooked up for the heat exchanger output?  Does this appear in the IntelliCenter app?

Tegguy, 2021-05-12:
Thank you for the information on the firmware I'll update that now.

There are only the 2 options you listed (Off and UltraTemp Only)

The aquacal does have a light for heating and cool but the pentair system only displays something for heat (the fire symbol) which I thought was weird. I do not know of a gas heat option for the aquacal unit but I'm new to this brand.

I do have a solar temp sensor hooked up to enable the heater. The solar sensor calibration does show up in the intellicenter settings area.

rstrouse, 2021-05-12:
Is there a gas pipe hooked up to it and an exhaust vent?  That will tell you if it is a hybrid model.  It does look like IntelliCenter is actually reporting the cooling mode so I will make sure the snowflake shows up when it is in this mode.  The solar temp sensor is irrelevant then I just saw temp reporting from it so I assume that it must have to have something connected to it in order to match the temp offsets.

After I get this coded, I will need you to do some very specific captures so I can finalize all the bits flying around.  This initial capture contains almost 7k messages so as you can imagine it won't take us all the way there but it did allow me to find the timeframe where you changed the cooling setpoint from 87 -> 89 -> 90.  Interestingly, it appears there is RS485 traffic directly from the AquaCal that is reporting its status.  And I can see IntelliCenter sending it a message to disable the control panel and demanding heat.

Tegguy, 2021-05-12:
No gas line on this heater.

Yeah the aquacal is hooked up over the rs-485 bus so I'm not surprised you
see traffic with it. I'm glad my capture is helping and just let me know
what else I can do to help. I've never done into RS485 stuff so I'm
probably no help deciphering anything but I can help with the testing.
There is a 3 min compressor delay each time the aquacal turns on so that is
probably one reason the capture is so long.

rstrouse, 2021-05-13:
What are the min and max values for the cooling setpoint?  Is it dependent on the heat setpoint?

Tegguy, 2021-05-13:
Yes the cooling setpoint has to be higher than the heater setpoint. I don't
know what the limits are though I'll try to do some testing tonight

rstrouse, 2021-05-14:
Are you capable of pulling files from the pi and posting them up here?  Also for this next step it is critical that you upgrade your IntelliCenter firmware.  The bug that I mentioned earlier will get in our way for the testing.

Tegguy, 2021-05-14:
Firmware has been upgraded.

Which files do you need?

rstrouse, 2021-05-14:
We are going to generate the files that I need.  So what I would like you to do is first pull the most recent master code for njsPC and dashPanel.  When you are done with that open a browser window and type the same url that you use to access dashPanel but add `/messageManager.html` to the end of it.  This will be what we use to generate the files we are looking for.

We are going to use messageManager to simply listen to the RS485 bus but since there is so much traffic on it I would like to limit the messages we are looking at.  To do this perform the following steps.

#### Step 1:
Click the hamburger menu on the upper left to show the settings dialog then select the logging tab.

#### Step 2:
Turn on all the options as you see them on this screen shot.
[image]

#### Step 3:
Click the filter icon located on the Broadcast option from the logging tab.
[image]

#### Step 4:
Select the following options on the include tab.  After you have selected them click away from the popup so it will close.
[image]

#### Step 5:
Get ready to start capturing by reviewing the things that I want you to do from IntelliCenter (either the mobile app, the web app, indoor controller, or wireless controller).  You can use whichever controller is most convenient.  Set all heat modes on all bodies to off.  Turn off all circuits, bodies, features, or any other items that may be running on the pool.  These will just be noise.

#### Step 6:
On the Settings > Logging tab click the Log to: File option on the packets section.  As soon as you click it njsPC will begin capturing the information we requested.  Once you have clicked the log to file option click away from the logging tab to close the settings popup.  If you want to watch what information is being exchanged you can click the rightmost icon on the Messages bubble.  Each time one of the items that we are looking for appears on the RS485 bus it will show in the messages list.
[image]

#### Step 7:
Perform the following using an IntelliCenter controller
1. Change the heat mode for the pool body to UltraTemp Only
2. Change the heat mode for the spa body to UltraTemp Only
3. Change the cooling setpoint for the pool body and let me know what you changed if from and to.
4. Change the cooling setpoint for the spa body and let me know what you changed it from and to.

#### Step 8:
Re-open the hamburger > Settings > Logging tab and uncheck the Log to File option.  Please remember to do this or njsPC will continue to write the log file eventually filling up your pi with a very large file.

#### Step 9:
There will now be a file in your njsPC logs directory on the pi.  Its name will start with `packetLog(` followed by todays date and time.  Upload that file here.

Tegguy, 2021-05-16:
I ran the git-pull on the controller and the dashpanel but when I go to the "hamburger" at the top left and go to logging the heater option doesn't show up.

[image]

rstrouse, 2021-05-16:
Hard refresh your browser CTRL + F5

Tegguy, 2021-05-16:
Doh! Bonehead move I forgot about the refresh sorry. I think I did the capture correctly and it's attached both pool and spa started at 100 and stopped at 90 for the cooling set point. Let me know if this doesn't look right and I can try it again.

[packetLog(2021-05-16_08-03-19).log](https://github.com/tagyoureit/nodejs-poolController/files/6490063/packetLog.2021-05-16_08-03-19.log)

rstrouse, 2021-05-17:
Well this pretty much nailed down the location for the pool and spa cooling setpoints.  However, it looks like you did a capture replay or you deleted the data directory and reloaded njsPC during the caputure.
For this process I do not want a replay as this generated an extra 700 messages to weed through.

Then since I was able to whittle down the location of the pool cooling setpoint I want you to do some other things regarding the heat settings.

#### Step 1:
Start up njsPC and dashPanel/messageManager and wait for the Ready indicator at the top right.

#### Step 2:
Set up the message inclusions using previous steps 1-5.  Do not capture the replay.  I do not want all the configuration options or the extended status data included in our file.  Turn off everything on the pool.  Including setting the heat modes for both the pool and spa to Off.

#### Step 4:
Click clear messages on the Settings > Logging tab.  This will make sure we only have the stuff we want.

#### Step 3:
Now go ahead and click the Log To file under the Packets section on the Settings > Logging tab.

#### Step 4:
Using any Pentair controller perform the following steps:
1. Change the heat mode to UltraTemp only for the pool
2. Change the heat mode to UltraTemp only for the spa
3. Turn on the pool
4. Allow the heater to go into heating mode by changing the setpoints to make it start heating
5. Allow the heater to go into cooling mode by changing the setpoints to make it start cooling

#### Step 5:
Uncheck the log to file option on the Settings > Logging tab.  And copy the new file you just created here.

Thanks a bunch.  Hopefully, this will nail down the rest of the data items.

Tegguy, 2021-05-17:
Here is the capture. I think I did it the right way this time. I didn't log temp settings but a guess is cool was 100 down to 75, heat was like 70-85 hope this information wasn't critical. If it was I can redo the test while remembering the settings.

[packetLog(2021-05-04_17-17-36).log](https://github.com/tagyoureit/nodejs-
poolController/files/6497295/packetLog.2021-05-04_17-17-36.log)

rstrouse, 2021-05-17:
Ooops the file didn't upload.

Tegguy, 2021-05-18:
Sorry about that let's try this again
[packetLog(2021-05-04_17-17-36).log](https://github.com/tagyoureit/nodejs-poolController/files/6500568/packetLog.2021-05-04_17-17-36.log)

Tegguy, 2021-05-18:
The log looked weird so I recaptured this morning

[packetLog(2021-05-18_06-34-21).log](https://github.com/tagyoureit/nodejs-poolController/files/6500651/packetLog.2021-05-18_06-34-21.log)

rstrouse, 2021-05-18:
Progress update.  The log was very helpful.  It appears that the Ultratemp uses the solar bytes for its data.  
So I have been able to determine the following:
1. The heat status that is sent when the Ultratemp is heating, cooling, and off.
2. The heat mode that is set when changing the body from off to Ultratemp.  It appears that you cannot have a gas heater or a solar heater and an Ultratemp.   If you have assigned one or the other to the body then the option is not available.

But this has begged a few questions.
Out of curiosity what do you have the differential temp set to in your configuration?
There is a byte being returned from the heater that is always 3 are there any settings that might be 3.
`[165, 0, 16, 112, 115, 10][160, 1, 2, 3, 0, 0, 0, 0, 0, 0][2, 72]`

I will let you know when I am done with the testing so you can pull and test it out.

rstrouse, 2021-05-18:
Go ahead and pull njsPC and dashPanel.  Lets see if we have the setpoints showing up for UltraTemp.

Tegguy, 2021-05-18:
"The heat mode that is set when changing the body from off to Ultratemp. It appears that you cannot have a gas heater or a solar heater and an Ultratemp. If you have assigned one or the other to the body then the option is not available."

You can have both at least with the intellitouch. My neighbor has the same heater as me and also a gas heater. I don't have both but he does.

The setpoints for heat and cool do show up in the new dashpanel! 

My differential temp in the config is 3F with the boost temperature set to 0.

rstrouse, 2021-05-18:
Thanks for that I suspected that the differential temp was the 3 and one of those other bytes is likely the boost temp,  I'll bet your neighbor cannot control the cooling setpoint from IntelliTouch.  Can you ask him if he can?

This kinda smacks of something Pentair just now got around to doing since I know the bytes weren't there correctly in fw1.040.

rstrouse, 2021-05-18:
I would also like you to verify that when heating you are getting a status of Heatpump (heat) and when cooling a status of Heatpump (cool).  There should be a flame icon and a snowflake icon to indicate which is underway.

Tegguy, 2021-05-19:
I checked with my neighbor and he says he can control the cooling setpoint from his intellitouch. If I remember correctly there is a specific port for the gas heater but I'd have to dig back into the manual

Tegguy, 2021-05-19:
Here is his screenshot 
[image]

Tegguy, 2021-05-19:
Icons have been verified they show up at the appropriate time. It doesn't appear as though I can control the heater yet from the app but I'm not sure if I'm doing something wrong or if you're still working on this.

rstrouse, 2021-05-19:
What do you mean control the heater?  Are the heat modes not sticking?

Tegguy, 2021-05-19:
So I'm not sure if I'm doing something wrong but I cannot get the heater to turn on with the app. I have tried clicking the green button and I have tried just changing the set points. I have to turn the heater on in the pentair app to get it to work. Also on Chrome the fire symbol was in a weird spot

[image]
[image]

rstrouse, 2021-05-20:
Oh it is missing the UltraTemp heat mode.  That screen should contain Off and UltraTemp.  I'll take a look and see why these are not being sent on as a valid mode for the body.  It also looks like the Heatpump (heat) text is pushing the icon on the panel.  Some clever positioning should take care of that.

rstrouse, 2021-05-20:
Pull both njsPC and dashPanel and your woes should be taken care of.

Tegguy, 2021-05-23:
Sorry for the delay. Pulled the latest files and everything appears to be working great. If I were to ever add a gas heater would all the other options already be there (for example ultratemp preferred)  or will you need more information?

I have a side question (I can open a new item if you'd like) but I have the 5150 port forwarded so I can access my dashpanel remotely but it doesn't seem to load anything other than the initial page with the background and some lines. Any ideas? If I connect via VPN it works.

rstrouse, 2021-05-23:
Yes the preferred options will show up if you add a gas heater.  

You also need to forward port 4200 or whichever port you are using for njsPC.  There are some folks using zerotier to gain access and have had success with it.  If you need suggestions there you might get more help on the gitter page since folks use different things for external access.

rstrouse, 2021-05-23:
Closing this.  If you need further help open another issue.

closed 2021-05-23 by rstrouse