tagyoureit/nodejs-poolController #291: Feature Request: Pump Speed Adjustment Input

Tegguy, 2021-05-05:
Would it be possible to make the pump speed setting box something you could type in rather than clicking the + or - buttons? Similar to how the time box is in the schedule?

tagyoureit, 2021-05-05:
Done.  Please pull the latest and hard refresh the browser (ctrl-F5/command-f5) to ensure the browser downloads the new files.

Tegguy, 2021-05-05:
I tried this (Ctrl+F5) but it still won't allow me to click on the pump
speed and type anything. Is there something else I need to do to get this
to update?

tagyoureit, 2021-05-06:
You need to download the latest dashPanel software.

Assuming you used git, go to the dashPanel directory and run `git pull` and it will download the latest files.  Then restart the app.  Finally, go to the web browser and then do the hard refresh.

Tegguy, 2021-05-06:
Sorry still learning this whole thing. Git pull worked! and the feature works! thank you!

closed 2021-05-06 by Tegguy