tagyoureit/nodejs-poolController #294: HA header toggle doesn't work

acronce, 2021-05-11:
If I use the header toggle to turn all entities in a Home Assistant Lovelace card on or off, it only affects the first entity in the list.

Steps to reproduce the behavior:
1. Create a Lovelace entities card with two or more pool entities. In my case I created one with the pool and spa lights.
2. Click on the header toggle button to turn all of the entities on.

Expected behavior:
I would expect all the entities (lights in my case) to turn on.

Actual behavior:
Only the first entity (pool light in my case) turns on.

I've included the packet capture here, after removing personal and sensitive data.
[2021-05-11_07-15-10.zip](https://github.com/tagyoureit/nodejs-poolController/files/6460090/2021-05-11_07-15-10.zip)

tagyoureit, 2021-05-13:
Can you pull the latest (7.1.0) and see if we have it fixed.  We made some changes to the way the outbound packets are processed.

acronce, 2021-05-13:
Thanks! I'll give it a try.

I'm running 7.0.0 currently, and have never upgraded before. Do I just stop the app, do a pull, then start up again (I'm using pm2 to automatically launch the apps). 

Or is it generally recommended to also do a 'npm i' after a pull?

tagyoureit, 2021-05-14:
General best practice:
1. `git pull`
2. `npm i` (although no new dependencies were introduced with 7.1)
3.  If you, do step 2, you may need to use PM2 to restart the app anyway because we don't watch the compiled directories but otherwise the watch will pick up any new file changes.

acronce, 2021-05-15:
Thanks! I updated, and it *mostly* works. But not always.

The very first time I turned on the lights via header toggle it failed. The first light came on, but not the other.

Then it worked about 10 times, and failed when turning the lights off. Only the first light turned off but not the other.

When there's a failure, there seems to be the following errors in the log:

```
1|poolController            | [5/14/2021, 18:23:12] warn: Comms port is not open.  Message aborted: 165,1,16,33,134,2,4,0,1,99
1|poolController            | [5/14/2021, 18:23:12] error: Comms port is not open.  Message aborted: 165,1,16,33,134,2,4,0,1,99
```

This error message seems to be new with this latest update. I've searched the past few days of logs and there are no instances of it prior to the update.

rstrouse, 2021-05-18:
Did the commit 3 days ago fix your issue?

acronce, 2021-05-18:
Sorry, I didn't see this before. I'll try it tonight and report back.

acronce, 2021-05-19:
Sorry, but after updating I saw the same failure after 3 attempts. Basically the first light turned on, but the second did not.

In case it's helpful (but I don't think it is), here's the pm2 log. Note that I had poolController-dashPanel running in a browser to verify the results of the changes driven by HA:

```
1|poolController  | [5/18/2021, 19:38:49] info: EasyTouch system config complete.
1|poolController  | [5/18/2021, 19:38:49] info: Initializing Nixie Controller
1|poolController  | [5/18/2021, 19:38:49] info: Nixie Controller Initialized
2|poolController-dashPanel  | xxx.xxx.xxx.xxx GET /config/web.services?null
1|poolController            | [5/18/2021, 19:39:52] info: [19:39:52] xxx.xxx.xxx.xxx GET /state/all?null {}
1|poolController            | [5/18/2021, 19:39:53] info: New socket client connected _G7bSUqHC5s7esRwAAAB -- xxx.xxx.xxx.xxx
2|poolController-dashPanel  | xxx.xxx.xxx.xxx GET /config/web.services?null
1|poolController            | [5/18/2021, 19:39:53] info: [19:39:53] xxx.xxx.xxx.xxx GET /state/all?null {}
```
After it failed, I tried again and on the third attempt *no* lights turned on, and there was no new information in the pm2 log.

Honestly the current state seems worse. It's gone from mostly working (9 out of 10 times), to failing more often (about 30%).

I wish I had better news. Let me know if there's anything else you need me to do.

rstrouse, 2021-05-19:
Do another pull.  Perhaps we were throttling a little too hard to keep from overwhelming the OCP.

acronce, 2021-05-22:
Unfortunately it still doesn't work reliably. On my 10th attempt turning on the lights from the header toggle, neither light turned on.

I didn't see anything useful in the normal pm2 log output.

rstrouse, 2021-05-22:
If there is nothing in that log then that tells me EasyTouch is responding that it got the message and is doing nothing with it.  I would like replay for this but I do not want you to reload your configuration.  

Instead I would like you to click the capture replay but when the Select Capture Options dialog comes up I would like you to uncheck Capture Configuration Reload.  Then while it is capturing see if you can get it to fail.  When it does stop then and finish the capture.

fgonza2, 2021-05-29:
> If I use the header toggle to turn all entities in a Home Assistant Lovelace card on or off, it only affects the first entity in the list.
>
> Steps to reproduce the behavior:
>
> 1. Create a Lovelace [...]

how are you interfacing with HA - via MQTT?

acronce, 2021-05-30:
> how are you interfacing with HA - via MQTT?

Yes, I'm using MQTT currently. But it also happened when I was using REST before moving to MQTT.

acronce, 2021-05-30:
> Instead I would like you to click the capture replay but when the Select Capture Options dialog comes up I would like you to uncheck Capture Configuration Reload. Then while it is capturing see if you can get it to fail. When it does stop then and finish the capture.

I haven't had time to try another capture (under a major deadline), but I'll hopefully get to it this weekend.

tagyoureit, 2021-06-16:
Have you been successful?  Or can you get another packet capture?

closed 2021-08-05 by tagyoureit