tagyoureit/nodejs-poolController #296: Install Hanging

gambler2841, 2021-05-12:
**Describe the bug**
While trying to install the package, after running 'npm start&', the connection to the RS485 adapter is successful, but hangs after that:

[image]

Prior to this I was getting the lat/long error, but after entering the information into the poolConfig.json,, this is where i'm at. Not sure why it's hanging up. It is wired to the pool controller, but not connecting.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...' npm start&
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
I expected connection to the pool equipment, or an error of why it cannot connect.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Packet Capture**
Follow the instructions to complete a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) and attach the resulting zip/log files.

**Pool Equipment**
 - Controller: [e.g. IntelliCenter, EasyTouch, IntelliTouch]
 - Controller Model: [e.g. IntelliCenter i5PS, EasyTouch2 8P]
 - Controller Firmware Version: [e.g. 1.047]
 - Pump(s) manufacturer and model: [e.g. IntelliFlow 2 VST 011056]
 - Chlorinator: [e.g. iChlor, IntelliChlor-40]
 - Heater(s): [e.g. gas, solar, heatpump, ultratemp]
 - Chemical controller: [e.g. IntelliChem, homegrown]
 - Valves: [e.g. Intellivalve]
 - Any other relevant equipment:

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

rstrouse, 2021-05-12:
This is not hanging.  It is simply not hearing any information from your EasyTouch.  As a result njsPC is acting as a master controller for a single body system.  Once it detects traffic from an EasyTouch controller it will acquiesce to it and immediately start asking for information from the EasyTouch.

So based on our experience here, this means that either the adapter is bad, the serial port is not on /dev/ttyUSB0 or you have the wires backwards.  Bear in mind we have seen a ridiculous number of failed JBTek adapters lately.  The other thing we have seen are adapters with the printing backwards for the A-/B+ channels on RS485.  The way to check this is to simply swap the wires but you should shut down before you do it.  You do not want the two wires to touch.

gambler2841, 2021-05-13:
Thanks for the quick response. I changed the A-/B+ and it found the controller. Now I'm stuck at this:

[image]

Any ideas?

rstrouse, 2021-05-13:
That line `Pentair EasyTouch System Detected` is actually data from the OCP.  However, the aborted message is common.  What does your poolConfig.json look like in the data directory?  It looks like the configuration actually completed. 

Are you running dashPanel and do you see your circuits and features?  Collisions and aborts are a normal part of an RS485 bus?

gambler2841, 2021-05-13:
I haven't installed the dashPanel yet; I thought I had to have the connection to njsPC up and running completely first. I will try and install now. 

I can try and get the poolCongig.json, but i'm VNC'd into the Pi, so it's hard to get all the information. Is there something in particular you're looking for?

Thanks so much for the help! I'm a newbie and really appreciate it!

rstrouse, 2021-05-13:
I am pretty sure that once you get dashPanel up and running this will all make sense.  The process you are looking at in the console is not locked or hung and it wasn't before.  If you had loaded up dashPanel you would have seen that it was a Nixie Single Body controller with all the fixins around that.  However, it should now say EasyTouch 2 8 in the header and the circuits, features, and schedules configured on your EasyTouch should appear on the screen.

gambler2841, 2021-05-13:
So I installed the dashPanel. It went well. This is what I am getting when logging into the server (granted I am using Chrome on my Windows machine, which I would think would work?)

[image]

rstrouse, 2021-05-13:
Click on the hamburger on the upper left.  Then choose the connection tab and enter the address for njsPC and press apply.  The default port for njsPC will be 4200.  You can also choose to find the server but that is dependent on your network ability to supply upnp packets.

gambler2841, 2021-05-13:
That did it! Thanks so much for your patience and help!

rstrouse, 2021-05-13:
No worries glad you are up and running.

closed 2021-05-13 by rstrouse