tagyoureit/nodejs-poolController #298: How do I set a circuit to be on when starting up the server?

colin-young, 2021-05-19:
Currently if I restart the service it comes up with the sole circuit I have defined in the off state. That's even if it was on when I stopped the service. How do I configure things so that when the nodejs-poolController starts up that circuit is "on" so that my schedules will actually run?

**Pool Equipment**
 - Controller: Raspberry Pi direct to pump - so, Nixie Single Body?
 - Controller Model: n/a
 - Controller: n/a
 - Pump(s) manufacturer and model: Pentair IntelliFlo 3 HP variable speed EC-011028
 - Chlorinator: not connected (yet)
 - Heater(s): n/a
 - Chemical controller: n/a
 - Valves: n/a

tagyoureit, 2021-05-19:
The short answer is that we don't have the schedules implemented yet for Nixie.  If it is a Nixie Single Body it should show up as such:
[image]

The longer term answer is that we will be implementing these shortly.  Just for clarity, though, the schedules turn the circuits on/off and not the other way around.  If any circuit is on that connects to the pump (pool or otherwise) then the pump will be on.  I don't think that's what you want.  

In the interim you can either use cURL or HomeAssistant/Hubitat/HomeSeer/node-Red to turn on/off the pump on a schedule via the api.  See [setState API](https://tagyoureit.github.io/nodejs-poolcontroller-api/#/state/put_state_circuit_setState_).

colin-young, 2021-05-20:
Thanks! That explains it, and makes way more sense. I thought I had managed to understand things this morning, but apparently not (this explains the behavior I was seeing).

My long term plan is to write my own scheduler anyway and use this server to interface to the pump and chlorinator. I'll just have to knock something together in node red tonight/tomorrow.

On another slightly related topic, I've found some inconsistencies between the swagger yaml and the server api. How do you prefer those to be reported? Separate bug for each, one big list?

tagyoureit, 2021-05-24:
Feel free to open up any specific issues for the swagger api/yaml.  We can keep this one open to track the schedules item.

rstrouse, 2021-05-28:
You can now add schedules to your Nixie controller.  These should trigger and operate as well as the egg timer functions.

colin-young, 2021-06-01:
Thanks for the update! I'm currently planning to stick with my current python (yuck) service/nodeRED implementation, but good to know the option is there now.

closed 2021-06-01 by colin-young