tagyoureit/nodejs-poolController #299: Pressure sensor and view

jarrah42, 2021-05-20:
Hi,

I've managed to update everything to the latest version and it seems to be working with dashpanel. I'm using Nixie single body to control my Intelliflo VS and IC60, thanks!

I have a pressure transducer on my filter that is hooked up to an ADC on a Pimoroni automation hat on my RP3. I set up REM and I'm getting voltage readings on the I2C bus. I would like to display the pressure on the dashpanel, preferably on the main panel but I don't see any easy way to do this. I see how to set up a feed to njsPC but there are no socket events for pressure. I'm also not sure how I could customize displaypanel to display this. Is it possible to do this currently? Could you point me at what I'd need to modify to do it?

Sorry if this it not the right place for this issue, but it seems to involve all three applications.

Thanks!

rstrouse, 2021-05-20:
Was REM able to configure the inputs on that device using the ADS1x15 device?

jarrah42, 2021-05-20:
Yes, using the REM web interface, I can see the channel #1 voltage change when the pump speed changes, so the sensor appears to be working fine. The thing I'm grappling with is how to feed this to njsPC and then display it on the displaypanel.

rstrouse, 2021-05-20:
Interesting that Pimoroni just hooked up an ads1015 and called it a day.  I wish other boards did that.

We haven't surfaced these values yet in dashPanel because it requires finding some time to sit down and build the interface.  However, your next step is to create a Generic Pressure Transducer device and feed it from the ADC.  This will convert the voltage to psi.  Then when we get the filter interface created the pressure transducer device will feed njsPC through the filter socket.

Just a little patience as our timeline is dependent upon external forces.

jarrah42, 2021-05-20:
Yeah, I was pleasantly surprised at how easy it was to get it to work. I'll look at creating the device and report back. I presume this is in REM, so I'll start digging through the code to see what it needs.

closed 2021-06-16 by tagyoureit