tagyoureit/nodejs-poolController #300: How to set heatMode via MQTT?

theRealMCQwerty, 2021-05-23:
Seems that to set heat mode via the API:

http://192.168.1.80:4200/state/body/heatMode
`heatmode` is set in the body.

But I can't work out how to do this via MQTT since there is no `body` topic in that is published to from NJSPC and the only mention of heatmode is under temps: `intellicenter-i8ps/state/temps/bodies/2/spa/heatMode`

Could you point me in the right direction?

Thanks!

rstrouse, 2021-05-27:
Topic is `intellicenter-i8ps/state/body/heatMode` and the payload is `{"id":<the body id>, "heatMode":"<off | heater | solar | solarpref | heatpump | heatpumppref | ultratemp | ultratemppref>"}`

You can also use `{"name":"<pool | spa>", "heatMode":"<off | heater | solar | solarpref | heatpump | heatpumppref | ultratemp | ultratemppref>"}`

theRealMCQwerty, 2021-05-28:
Thanks @rstrouse. 
```
{
  "name": "pool",
  "heatMode": "heater"
}
```

Didn't work for me but
```
{
  "id": 1,
  "heatMode": "heater"
}
```

Works great! Thanks again.

closed 2021-05-28 by theRealMCQwerty

tagyoureit, 2021-05-28:
We are doing a javascript string compare on the body names which is case sensitive.  So `pool !== Pool`.  Likely you have it upper case (Pool is the default) or maybe have a different name setup on the body.  Just fyi...