tagyoureit/nodejs-poolController #301: VS pump stopping

jarrah42, 2021-05-23:
Nixie single body with one VS pump and IC30. After a few minutes the VS stops but is still displayed as running in dashpanel.

Packet dump attached.
[packetLog(2021-05-23_09-43-10).log](https://github.com/tagyoureit/nodejs-poolController/files/6528217/packetLog.2021-05-23_09-43-10.log)

rstrouse, 2021-05-27:
When this occurs do you see anything in the njsPC log.  I wonder if we detected a comms lost and didn't report it correctly.

tagyoureit, 2021-05-28:
```
{"id":2958,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,17],[100],[215,16,3]],"ts":"2021-05-23T09:57:14.736-0400"}
{"id":2959,"valid":true,"dir":"in","proto":"chlorinator","for":[2958],"pkt":[[],[],[16,2,0,18],[72,128],[236,16,3]],"ts": "2021-05-23T09:57:14.785-0400"}
{"id":2960,"valid":true,"dir":"out","proto":"pump","pkt":[[],[255,0,255],[165,0,96,33,7,0],[],[1,45]],"ts":"2021-05-23T09:57:15.026-0400"}
{"id":2962,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,33,96,7,15],[10,0,0,7,91,12,38,0,0,0,0,0,1,22,18],[2,3]],"ts": "2021-05-23T09:57:15.088-0400"}
{"id":2961,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,20],[0],[118,16,3]],"ts":"2021-05-23T09:57:15.090-0400"}
{"id":2964,"valid":true,"dir":"in","proto":"chlorinator","for":[2961],"pkt":[[],[],[16,2,0,3],[0,105,67,104,108,111,114,32,32,32,32,32,32,32,32,32,32],[182,16,3]],"ts": "2021-05-23T09:57:15.152-0400"}
{"id":2965,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,0],[0],[98,16,3]],"ts":"2021-05-23T09:57:18.160-0400"}
{"id":2966,"valid":true,"dir":"in","proto":"chlorinator","for":[2965],"pkt":[[],[],[16,2,0,1],[0,0],[19,16,3]],"ts": "2021-05-23T09:57:18.210-0400"}
{"id":2967,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,17],[100],[215,16,3]],"ts":"2021-05-23T09:57:18.513-0400"}
{"id":2968,"valid":true,"dir":"in","proto":"chlorinator","for":[2967],"pkt":[[],[],[16,2,0,18],[72,128],[236,16,3]],"ts": "2021-05-23T09:57:18.564-0400"}
{"id":2969,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,20],[0],[118,16,3]],"ts":"2021-05-23T09:57:18.867-0400"}
{"id":2970,"valid":true,"dir":"in","proto":"chlorinator","for":[2969],"pkt":[[],[],[16,2,0,3],[0,105,67,104,108,111,114,32,32,32,32,32,32,32,32,32,32],[182,16,3]],"ts": "2021-05-23T09:57:18.931-0400"}
{"id":2971,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,0],[0],[98,16,3]],"ts":"2021-05-23T09:57:21.935-0400"}
{"id":2972,"valid":true,"dir":"in","proto":"chlorinator","for":[2971],"pkt":[[],[],[16,2,0,1],[0,0],[19,16,3]],"ts": "2021-05-23T09:57:21.989-0400"}
```
We've lost comms at packet 2962 with timestamp "2021-05-23T09:57:15.088-0400".  This is where you would look in the logs to see if we have any errors there.

jarrah42, 2021-05-29:
Unfortunately there are no errors in the log at this time, only a couple of warnings and infos. It seems to be running reliably now however. I've updated a couple of times since then, so possibly something got fixed. I'll close this for now and reopen if I see a problem again. Thanks!

closed 2021-05-29 by jarrah42