tagyoureit/nodejs-poolController #302: Ultratemp heater not showing up in schedule

Tegguy, 2021-05-27:
When setting a schedule I am unable to select ultratemp as the heater source because it does not show up. I know the functionality with this heater was just setup.

rstrouse, 2021-05-27:
I forgot about that reference.  So now I need you to capture the information just like we did for the heat modes but this time I want you to save a schedule from any Pentair controller that is set to Ultratemp for the heat source.  All I am looking for is the action 168 message that is sent when saving that schedule.  It will contain the heat source information.  I think I know where the cooling information is saved but it would be great to capture that information as well.

Tegguy, 2021-05-27:
I think I did this right... Should be a schedule for Sunday 8AM-5PM temp at 85

I ran this with the same broadcast settings as before so if you need me to set a different setting please let me know.

[packetLog(2021-05-23_09-25-39).log](https://github.com/tagyoureit/nodejs-poolController/files/6557321/packetLog.2021-05-23_09-25-39.log)

rstrouse, 2021-05-28:
Yeah I need different settings.  You can turn off Heater and set the broadcast settings to only include the 168 messages.
[image]

Tegguy, 2021-05-28:
Try this one:

[packetLog(2021-05-23_09-25-39).log](https://github.com/tagyoureit/nodejs-poolController/files/6560895/packetLog.2021-05-23_09-25-39.log)

rstrouse, 2021-05-28:
That is perfect.  Thanks

rstrouse, 2021-05-30:
If you pull njsPC and dashPanel the ultratemp heat sources should show up for you.

Tegguy, 2021-05-31:
This shows up now but the cool option also shows up. In the pentair app it doesn't give me a cool option for the schedule. I'll test it and see if it works.

Not sure if it's related but when I set a one time schedule it sets as 1 month past the date I set. For example if I select June 4th when I hit save it changes it July 4th

rstrouse, 2021-05-31:
Funny, the docs show it when the cooling is enabled for the ultratemp and there is data for it in the schedule message.

Stupid 0 based javascript months strike again.  I'll dig into the settings on njsPC.

Tegguy, 2021-05-31:
The cool function works in the schedule. Thanks!

rstrouse, 2021-06-01:
Thanks there are still things in IntelliCenter that exist only in the documentation or they exist on one platform and not another.  Probably on the dev schedule.

If you pull njsPC I fixed the run-once schedule date.

closed 2021-06-01 by Tegguy