tagyoureit/nodejs-poolController #304: "service", "timeout" monitoring and control via MQTT

fgonza2, 2021-05-29:
I am using HA with MQTT and SOCAT with a RS485 to wifi module. works great mostly :)

Is there a way to monitor and disable "service" or "timeout" mode via MQTT?

dashpanel shows it but you can't change it from there.   Let me know if there is a way or if not if this can be added as a feature

thanks

rstrouse, 2021-05-29:
If you are using a Pentair OCP this can only be triggered from the main panel.  I have yet to find a way to coerce the panel into timeout or service more or kick it out of it.

fgonza2, 2021-05-29:
Got it.  Is there a way to monitor the state via MQTT?

rstrouse, 2021-05-30:
These should now emit on the controller MQTT topic.

fgonza2, 2021-05-30:
great will try it out tonight thanks!

tagyoureit, 2021-06-06:
Please re-open if this is not fixed.

closed 2021-06-06 by tagyoureit