tagyoureit/nodejs-poolController #306: Easytouch 2-8 detected but I have an Easytouch 2-4

johnny2678, 2021-05-30:
continuing from gitter...

Since 5,x, njspc has always detected my OCP as an Easytouch 2-4 (which it is).  As such, that is my root for all MQTT state/config cmds.  However, I just updated to the latest 7.1 and now the OCP is detecting an easytouch 2-8.  

I don't think some kind soul swapped my OCPout in the middle of the night - but that would be nice ;)

Attached a replay of startup - not sure how to get a `2 packet` but can provide whatever else you need to look into this.

Also, this is not urgent for me anymore because I just manually changed my MQTT root back to `easytouch 2-4` and all my automations are working again.

[replay (1).zip](https://github.com/tagyoureit/nodejs-poolController/files/6566414/replay.1.zip)

rstrouse, 2021-06-01:
@johnny2678 this should be finding your EasyTouch 2-4 just fine now.  Please let us know if it still thinks its an 8.

johnny2678, 2021-06-01:
just pulled the latest - easytouch 2-4 is being detected now. thx @rstrouse !

closed 2021-06-01 by johnny2678