tagyoureit/nodejs-poolController #308: Dashboard not pulling in all i10PS Configuration

tclayton10, 2021-05-30:
I am a new njspc user who is trying to use a Pi to connect via RS485 to my new Intellicenter i10PS system.  After following the hardware and software installation and configuration steps, it appears that I am communicating with the i10PS, but the dashboard is not showing any details other than the type of system, date, and time.  I also see a lot of errors when running the npm start& command in the -poolController directory.  In addition, the npm start& command in the -poolController directory never completes.  It appears to be looped on a heliotrope message (lat/long and address info are in the i10PS configuration).  If I try to update the info in the dashboard I get a write error.

I have deleted the nodejs-poolController and nodejs-poolController-dashPanel subfolders several times, rebooted after the deletions, and then reinstalled from the github links and still get the same errors.

Here are two screen shots that will show the dashboard issues.  Happy to upload anything else that would be helpful in troubleshooting.

[image]

[image]

**Packet Capture**
I'll attach the packet capture from my Pi.  I can't paste it from my VNC connection to this Windows browser window.

**Pool Equipment**
 - Controller: IntelliCenter
 - Controller Model: i10PS
 - Controller Firmware Version: 1.047]
 - Pump(s) manufacturer and model: IntelliFlow VSF
 - Chlorinator: IntelliChlor-40
 - Heater(s): MasterTemp 400,000 BTU gas
 - Chemical controller: IntellipH

tclayton10, 2021-05-30:
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/6566975/replay.zip)

tclayton10, 2021-05-30:
Additional screen shots
[image]
[image]

rstrouse, 2021-05-31:
You are not getting reliable rs485 communications. It looks like it is receiving data but cannot send. First make sure njsPC is exclusively accessing the serial port.   Next make sure you are using an rs485 adapter and not a tty adapter.  Finally verify your wiring.

tclayton10, 2021-05-31:
Appreciate the quick response.  Definitely using an RS485 adapter (DSD Tech; only one I could get quickly), and wiring should be good at both ends, but I'll verify again tomorrow in daylight.  Same cable was on the old indoor control panel on the old ET8 system, and that worked fine.  Nothing else on the i10PS using the port that I am aware of.  Is there a command or check I can run to see if anything on the Pi is trying to access that port as well?

rstrouse, 2021-05-31:
There are some troubleshooting tips in the wiki.

https://github.com/tagyoureit/nodejs-poolController/wiki/RS-485-Adapter-Details

tclayton10, 2021-05-31:
I'll run thru that wiki again tomorrow to make sure I didn't miss anything. I have a new adapter coming this week just in case.Sent from my Verizon, Samsung Galaxy smartphone

https://github.com/tagyoureit/nodejs-poolController/wiki/RS-485-Adapter-Details

tclayton10, 2021-05-31:
Wiring is good, and I'm seeing the ffs5ff entries several times in the od -x output string.  I'm gonna guess for now that the adapter isn't working correctly and hope that when the waveshare adapter arrives on Wednesday that it will solve the problem.

rstrouse, 2021-05-31:
Yeah there have been a rash of crappy adapters lately.  I wonder if the company that makes support chips for all of these is having some sort of quality issue.

tclayton10, 2021-06-01:
New adapter arrived, and SUCCESS!!!!!  So I'd say the DSD Tech RS485 adapter is not compatible.  Appreciate the help getting this up and running!!!

rstrouse, 2021-06-01:
Yeah given what I have seen lately with these adapters I would be willing to bet that it is most likely defective.  There has to be some defective chip that all these vendors are using.  Maybe it works at one speed but not another.  We need 9600 but maybe it is fine at 38400.  Who knows, good thing they are cheap.

closed 2021-06-01 by rstrouse