tagyoureit/nodejs-poolController #310: Intellichem status isn't correct in dashPanel

davthomaspilot, 2021-06-01:
**Describe the bug**
Intellichem status isn't correct in dashPanel.  ScreenLogic correctly shows ORP, pH, etc.

**To Reproduce**
Compare dashPanel Chemistry panel to ScreenLogic

**Expected behavior**
dsshPanel Chemistry should report same as dashPanel

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Packet Capture**

[snapshot.log](https://github.com/tagyoureit/nodejs-poolController/files/6579292/snapshot.log)

**Pool Equipment**
 - Controller: EasyTouch
  
 - Controller Firmware Version: 2.140
 - Pump(s) manufacturer and model: [e.g. IntelliFlow 2 VST 011056]
 - Chlorinator: [Intellichem (both chlorinator and acid injector)
 - Heater(s): [e.g. gas, solar, heatpump, Gas Heater
 - Chemical controller: [IntelliChem (injects both chlorine and acid)
  
 -
[image]

davthomaspilot, 2021-06-01:
I can provide a temporary password if you want to connect to the EasyTouch remotely to reproduce/debug.  But, I need a time window for that.

rstrouse, 2021-06-01:
So what is currently appearing in dashPanel? What we are likely missing is that the original info we had did not include the second relay pump for the Chlorine dosing.  Screenshot please.  Also If I can get you to give me a replay dump by clicking on the hamburger menu on the upper left of dashPanel then navigate to the logging tab and click Capture Replay.  We want the entire configuration for this so leave the Capture Configuration Reload checkbox checked.  Press the Begin Capture button.

Give it a couple of seconds and it will begin interrogating the EasyTouch.  During this time the status will change to loading and provide a percentage completion.  As it detects more data the percentage may reset to grab that info as well.  When it stops and the status on the upper right changes to Ready, press the Cancel Capture button and it will download a zip file to your browser.  Upload that here.

rstrouse, 2021-06-01:
Ok so I told you I would show you how to view the message exchange between EasyTouch and IntelliChem.  First make sure you have updated your njsPC and dashPanel installations.

If you open up a browser and type the following url `http://<ip address for dashPanel>:5150/messageManager.html` it will open up the message manager application.  From there you will see the following.
[image]

If you click the hamburger menu and navigate to the logging tab you will see a series of options.  If you select the options as you see below this will filter to only include the communications between EasyTouch and IntelliChem.
[image]

Once you have selected the options as above click away from the settings dialog to close it.  From here you can click the list icon on the upper right of the messages bubble and any communication between EasyTouch and IntelliChem will appear in the messages list.  If you hover over the other options up there it will tell you what they do.

All messages on the RS485 bus can be viewed using this tool and the filters on the settings > log tab are used to determine which ones you want to see.  On the broadcast button there is a filter that will allow you to further refine the broadcast messages that come from EasyTouch to the equipment.

davthomaspilot, 2021-06-01:
> So what is currently appearing in dashPanel? What we are likely missing is that the original info we had did not include the second relay pump for the Chlorine dosing. Screenshot please. Also If I [...]

The log file snapshot.log was in the original post.   Did you miss it, or do I need to do it again?

Originally, there was a panel on the home screen for chemistry, but it is no longer there.  That's probably because of my actions.

I tried adding another Intellichem controller (to try another address).  But, I got a message saying I could have only one.  So, I deleted it.  But, when I tried to create it again, I keep getting the message saying I can have only one (see screen shot below)

[image]

davthomaspilot, 2021-06-01:
Redid the log following your instructions more closely (especially, waiting for 100% and adding filter for intellichem messages)

[packetLog(2021-06-01_10-53-13).log](https://github.com/tagyoureit/nodejs-poolController/files/6580939/packetLog.2021-06-01_10-53-13.log)

davthomaspilot, 2021-06-01:
Here's the zip file when I did it from messageManager.

[replay (4).zip](https://github.com/tagyoureit/nodejs-poolController/files/6580977/replay.4.zip)

rstrouse, 2021-06-01:
Here is the protocol documentation as we know it.

<meta name="ProgId" content="Word.Document">
<meta name="Generator" content="Microsoft Word 15">
<meta name="Originator" content="Microsoft Word 15">
<link rel="File-List" href="file:///C:/Users/rstro/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel="themeData" href="file:///C:/Users/rstro/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel="colorSchemeMapping" href="file:///C:/Users/rstro/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

<h2 style="mso-list:l0 level2 lfo1"><a name="_Toc59351306"><span style="mso-fareast-font-family:Arial"><span style="mso-list:Ignore">1.1<span style="font:7.0pt &quot;Times New Roman&quot;">&nbsp;&nbsp;&nbsp; </span></span></span>IntelliChem
Protocol</a><o:p></o:p></h2>

<p class="MsoNormal">IntelliChem communicates with its controllers via a series
of messages.<span style="mso-spacerun:yes">&nbsp; </span>Some are sent from the
IntelliChem controller to the OCP while others are sent from the OCP to the
controller.<span style="mso-spacerun:yes">&nbsp; </span>For the status information
we will focus on the status message that comes in the form of an action 18
message sent from the IntelliChem controller.<o:p></o:p></p>

<p class="MsoNormal"><o:p>&nbsp;</o:p></p>

Index | Value | Description
-- | -- | --
0-1 | pHLevel | The current integer pH   level. -- 256 x byte(0) + byte(1) (big endian encoding) / 100
2-3 | ORPLevel | The current ORP level. --   256 x byte(2) + byte(3) (big endian encoding)
4-5 | pHSetpoint | The current pH setpoint --   256 x byte(4) + byte(5) (big endian encoding) / 100
6-7 | ORPSetpoint | The current ORP setpoint   -- 256 x byte(6) + byte(7) (big endian encoding)
8 | &nbsp; | Unknown
9 | &nbsp; | Unknown
10 | &nbsp; | Unknown
11-12 | pH Dose time | 256 x byte(11) + byte(12)   (big endian encoding) in seconds
13 | &nbsp; | Unknown
14-15 | ORP Dose time | 256 x byte(14) + byte(15)   (big endian encoding) in seconds
16-17 | pH Dose Volume | The units here are ambiguous   256 x byte(16) + byte(17) (big endian encoding)
18-19 | ORP Dose Volume | The units here are ambiguous   256 x byte(18) + byte(19) (big endian encoding)
20 | pH Tank Level | The byte values are 1-7   and the actual level is byte(20) – 1.&nbsp;   If the value is 0 the tank is not installed.
21 | ORP Tank Level | The byte values are 1-7   and the actual level is byte(21) – 1. If the value is 0 the tank is not   installed.
22 | Saturation Index | If bit(8) is set then the   value is negative (256 – byte(22))/ -100. Otherwise it is positive byte(22) /   100
23-24 | Calcium Hardness | 256 x byte(23) + byte(24)   (big endian encoding) in ppm
25 | Status 2 | ??????
26 | Cyanuric Acid | &nbsp;
27-28 | Total Alkalinity | 256 x byte(27) + byte(28)   (big endian encoding) in ppm
29 | Salt Level/TDS | Byte(29) * 50
30 | Water Flow | &nbsp;
31 | Temperature | &nbsp;
32 | Alarms | Alarm bitmask the bits   are decoded as follows   1: Flow (0=No flow)   2-3: pH Alarm (0 = No   alarm, 1 = Low pH, 2 = High pH)   4-5: ORP Alarm (0 = No alarm,   1 = Low ORP, 2 = High ORP)   6: pH tank Empty (0 = No   alarm, 1 = Tank Empty)   7: ORP tank Empty (0 = No   alarm, 1 = Tank Empty)   8: pH Probe Fault (0 = No   Alarm, 1 = Probe Fault)
33 | Warnings | Warning bitmask the bits   are decoded as follows   1: pH Lockout (0 = No   warning, 1 = pH Locked out)   2: pH Daily Limit (0 = No   warning, 1 = pH Daily Limit Reached)   3: ORP Daily Limit (0 =   No warning, 1 = ORP Daily Limit Reached)   4: Invalid Setup (0 = No   warning, 1 = Invalid Setup)   5: Chlorinator Comms (0 =   No warning, 1 = Chlorinator comms error)
34 | Dosing Status | Bitmask for the current   dosing status. High nibble (bits 5-8) contains the dosing status.   5-6: pH dosing status (0   = Dosing, 1 = Monitoring, 2 = Mixing)   7-8: ORP dosing status (0   = Dosing, 1 = Monitoring, 2 = Mixing)
35 | Setup Data | Bitmask for the setup   information on the controller   The upper nibble bits (5-8)   are a bitmask for the following Boolean values.   1-3: Setup Status (0 = No   Errors, 1 = No Comms, 2 = Setup Error   4: 1 = Manual Dosing   5: 1 = Use Chlorinator   6: 1 = HMI Advanced   Display   7: 0 = Dose Base pH+, 1 =   Dose Acid pH-   8: Not Used
36-37 | Firmware Version | Minor version = byte(36),   Major version byte(37)
38 | Water Chemistry Warnings | Water chemistry warnings.&nbsp; These are straight values where   0 = Ok   1 = Corrosive   2 = Scaling   8 = Unable to calculate
39 | &nbsp; | Unknown – The remaining 3   bytes might actually be for CO2 dosing.
40 | &nbsp; | Unknown
41 | &nbsp; | Unknown

<p class="MsoNormal"><o:p>&nbsp;</o:p></p>

<h3 style="margin-left:0in;text-indent:0in;mso-list:l0 level3 lfo1"><a name="_Toc59351307"><span style="mso-fareast-font-family:
Arial"><span style="mso-list:Ignore">1.1.1 </span></span>Manual
Control</a><o:p></o:p></h3>

<p class="MsoNormal">Manual control of IntelliChem is managed by sending a 146
message to the controller address from 16 (broadcast).<span style="mso-spacerun:yes">&nbsp; </span>All integers are backwards from inbound
message<o:p></o:p></p>

Index | Value | Description
-- | -- | --
0-1 | pHSetpoint | The target pH setpoint.
2-3 | ORPSetpoint | The target ORP setpoint
4 | 1 | &nbsp;
5 | 1 | &nbsp;
6-7 | Calcium Hardness | &nbsp;
8-9 | Cyanuric Acid | &nbsp;
10-11 | Total Alkalinity | &nbsp;
12-20 | 0s | &nbsp;

<p class="MsoNormal"><o:p>&nbsp;</o:p></p>

davthomaspilot, 2021-06-01:
Thanks!  Tomorrow I plan to play with that.

rstrouse, 2021-06-01:
Yeah I needed the support files that went along with the message log.  That is telling me that the message that we are expecting from EasyTouch that says that the IntelliChem is enabled isn't coming through.  I don't have EasyTouch so I will defer this to @tagyoureit to see if he has any insight.

rstrouse, 2021-06-02:
I added a fix to acquire the ET IntelliChem data if it doesn't have it.  Pull njsPC and give it a shot.  Since I don't have the equipment,  it would be greatly appreciated if you could verify the documentation with what we believe the known settings are if you are able.

davthomaspilot, 2021-06-02:
Ok-checking it out now, but having trouble getting going again.

Erased everything and starting over.

davthomaspilot, 2021-06-02:
Working!  

The ph and orp values look correct:

[image]

Also, the setpoints are good.  Nice graphics!!

Should the bubble be green?  Also, it says: Mixing pH and mixing ORP.  Not sure what that means--the mixing delay has long expired.

I'll start looking at packets--

davthomaspilot, 2021-06-02:
I guess I should first focus on inbound messages from Chem[1], action 18, that are flagged as invalid?  (relay log is below)

So, my approach will be to look at code to see where messages get flagged as invalid, and see if I can figure out what's making them invalid.

[replay (5).zip](https://github.com/tagyoureit/nodejs-poolController/files/6584954/replay.5.zip)

davthomaspilot, 2021-06-02:
<quote>Should the bubble be green? Also, it says: Mixing pH and mixing ORP. Not sure what that means--the mixing delay has long expired.</quote>

byte 30 ("Water Flow" from above) in the Chem[1] packets is 0.  But, the flow sensor is definitely indicating that flow is on.

When flow sensor is indicating inactive, screenLogic indicates this by showing "-" (dash) in the ph and ORP fields.   I can turn off the pumps and see what changes-

davthomaspilot, 2021-06-02:
The dosing status fields (byte 34) don't seem right.

I see 0x55, and the mixing delay has expired.  

"Bitmask for the current dosing status. High nibble (bits 5-8) contains the dosing status. 5-6: pH dosing status (0 = Dosing, 1 = Monitoring, 2 = Mixing)"

I think that means 0x55 would be interpreted as mixing for both ORP and pH.  And, the lower nibble is not used.   This might explain why the bubble isn't green.

 I'll get a capture during mixing time.

rstrouse, 2021-06-02:
You can ignore the Invalid messages.  These are normal on this type of bus.  It is half duplex so send and receive are the same wires and there is no signal for ready to send.

You are absolutely correct I regarding the status'.  They should both be monitoring and the bubble should green up when it is dosing.  I'll have a look.

rstrouse, 2021-06-02:
I took another look at the dosing status.  Pull njsPC and let me know if the correct display appears in dashPanel.  At this point can you upload a screenshot of the chemistry settings screen.  There are probably items in there that you cannot control for IntelliChem.  For instance we do not have the data for what was dosed in the last 24 hours, nor do we have the ability to cancel mixing or set a manual mix time.  I don't think there is the ability to initiate a manual dose from EasyTouch.

[image]

davthomaspilot, 2021-06-02:
Ok, just finished up checking the packets (before I pulled your latest).

Most things checked out, except the dosing status.  I made a table with extra column for my observations:

[packet analysis.pdf](https://github.com/tagyoureit/nodejs-poolController/files/6585888/packet.analysis.pdf)

I planned to do some experimentation to further verify some fields--I'll do that after I try your latest.

Let me know if there is something  else specific you'd like me to try to verify.

rstrouse, 2021-06-02:
Wow thanks for the help and the collaboration!  There are a couple of things to note with this.  Since the first time we went through this we spent some time to upgrade our tools hopefully it will allow us to really nail down the data that is packed into the bytes.  Your document is very helpful.

First the extra byte at the end of the documentation was incorrect.  There are 2 messages that are bandied about wrt IntelliChem these include the action 147(41 bytes) message and the 18(40 bytes) message.  These two are identical except the 147 has a preceding 0 in it.  Their use differs in that the OCP sends out the 147 to all controllers out there when they want to know the last status from IntelliChem and the 18 message is the current status.

It appears your IntelliChem is wired differently than some of the others we have seen.  I have no idea which is correct but the differences are this.  We have seen in the past that IntelliChem is constantly powered whereas yours appears to only be powered when the filter circuit is engaged.  This means that we never get a comms lost byte or a no flow byte.  No big deal just an observation.

This leaves the discrepancies.  If I have this right the most glaring is the dosing status.  To get to the bottom of this I would set up messageManager to only show IntelliChem messages.  Then start the log feeding into the display by clicking on the rightmost icon in the Messages header.  If you click the not equal sign (make it pink) it will only show a message when the data has changed.  Also click the thumbtack so it won't change the selected message during the feed.  If you click on one of the action 18 messages it will appear on the right with the bytes that have changed since the last message.  This will tell you what is different from the previous message received.

Another area that probably needs some verification is the alarms and warnings bitmasks.  If we get the bitmasks correct dashPanel will do a pretty good job of showing these on the display.  Any help getting to the bottom of these would be appreciated.  This can only be done with some empirical testing on your end.

It would be great if we could also confirm the dose times and volume.  Perhaps we can synthesize the type of functionality we get from REM Chem regarding dosed amounts and timing if these are figured out.  That way IntelliChem users could enjoy the Grafana displays that we get when using REM Chem.

It looks like the only things that we

davthomaspilot, 2021-06-02:
You are welcome and thanks for the extra info!

It's been a while, but I recall there were some options on wiring the Intellichem.  I like it powered off when the pumps are off--this provides a way for me to clear the dosing limits when I'm far away on vacation.  

"I took another look at the dosing status. Pull njsPC and let me know if the correct display appears in dashPanel. "

Still looks incorrect:
[image]

Byte 34 is 0x95

I did a git pull, but maybe I should check a specific file to make sure I have your change?  

Yes, I plan to verify the alarms and warnings bit patterns.  Several looked good, but I really haven't experimented yet--I'll use your tips on message logging  

Also, ORP and ph dose volumes didn't make much sense.  I'll experiment with that

" I don't think there is the ability to initiate a manual dose from EasyTouch."

I was really hoping there would be some magic for that.   I've wanted that several times while away from the house for a few weeks.

My work-around is having the dose limits reset on power cycle.  Set targets to a extremes then turn pump back on.  After forcing a dose this way, then change targets back to something reasonable.     

There's a way to manually dose on the Intellichem, but there's no interface for it in ScreenLogic (or from the EasyTouch panel).

davthomaspilot, 2021-06-02:
Actually, I don't the Intellichem manual specifies connecting its power through the filter pump relay, to make sure it won't dispense when the pump isn't running.  (But the flow switch should prevent this too)

S```
witch the main house circuit breaker OFF. Connect the supply voltage to
the PUMP SIDE OF THE MAIN FILTER PUMP RELAY located in the load/
power center (see page 40 for the IntelliChem controller AC power wiring
diagram). This ensures that the IntelliChem controller will only dispense
chemicals when the filter pump is ON.
```

davthomaspilot, 2021-06-02:
But, it is possible to have no flow when pump is running--happens all the time.  

When my cleaner is on, there is not enough flow to trigger the flow sensor.  Which is a good thing, because the back pressure on the injectors is too high.  If it tries to inject when the pool cleaner is on, back pressure causes water to mix in the tanks.

rstrouse, 2021-06-02:
I don't know what the values are for the dosing status.  What we thought we knew is fundamentally wrong.

I need an example of the action 18 message and the corresponding dosing status for dosing, mixing, and monitoring with pH and ORP.

davthomaspilot, 2021-06-02:
Ok, I'll work on that.

But, it's pouring down right now and I think I need to do manual dosing at the Intellichem.  I did see a difference in mixing versus monitoring though.  Thought I posted that, but I'll detail what I find out by going through all states.

Probably be tomorrow--it will be easier with help from my wife to push the Intellichem buttons while I stop/start logs.

rstrouse, 2021-06-02:
Yeah we will see blue skies until November although we could use a good dousing.  No worries, once we get the info, the changes will be quick.  It is important that I know what state IntelliChem thinks its in and the corresponding action 18 message.   I think that was something that we struggled with getting when we originally wrote the code originally.  As a result, the values we have are wrong.

davthomaspilot, 2021-06-02:
Rain quit and wife is in a good mood, so I did a bit of experimenting

Low nibble of byte 34 is always 0x5.

When dosing PH, high nibble is 8
When dosing ORP, high nibble is 1

I was confused by "mixing time" versus "flow delay".  I was thinking mixing time was the time after flow sensor indicates on before intellichem will dose.  **Actually, it is the minimum time between doses**.  Hope that I didn't confuse you!

I set different times for ORP mixing time (8 minutes)  mixing and pH mixing time (14 minutes) mixing and did a power cycle.  I expected byte 34 to change after 8 minutes and change again after 14 minutes.   They didn't change.

Now I'm thinking they won't change until after a dose has completed.  Hopefully, a manual dose will initiate the mixing delay 
So, I think to get it in the "mixing" state, there needs to be a dose.  On the todo for tomorrow

**So, the bubble should be green during mixing time, only empty during delay time, right?**

There may be another state reported during flow delay.  I'll look for that too.  

On ORP and ph dose volume (currently, I'm dosing by volume).  There are actually two dose limits, one limit per dose, and one for limit 24 hours.  Also, units can be English (oz) or metric (mL).  I don't know if the field is unitless and any conversion must be in code, or if the client needs to know what units are associated with the volume.  A lot to experiment with there.

Do we know which fields can be written?  I suspect only those documented in the "Manual Control" section above?

rstrouse, 2021-06-02:
The bubble should only be green when the controller is dosing either Acid or Chlorine.

Theoretically, there should be a mix time and a delay time as two separate durations.  The other attributes would be great to verify:
pH Dose Time
ORP Dose Time
pH Dose Volume
ORP Dose Volume

Yes the Manual Control is the only message that we are aware of that we can send.  Bear in mind I have no idea what most of those bytes are.

davthomaspilot, 2021-06-02:
"The bubble should only be green when the controller is dosing either Acid or Chlorine."

Oh!
I thought your intention was to have the bubble green when ScreenLogic is reporting ORP and pH, empty when they are showing dashes.
ScreenLogic shows the dashes when the pumps are off and during the flow delay.  On, the rest of the time.  There is no indication of a current dose in progress.
But, the ORP/pH history shows when doses occurred,  so data must be in the packets.
The screenLogic ORP/ph history file might be useful to compare with timestamped packets from the messageManager.  I'll see if I can find and interpret it on the pc with the ScreenLogic app.

rstrouse, 2021-07-02:
I posted up changes that should fix all the outstanding items with IntelliChem.

closed 2021-08-05 by tagyoureit