tagyoureit/nodejs-poolController #312: superChlorRemaining does not count down

dnikles, 2021-06-04:
When I set SuperChlor through dashpanel, the timer starts counting down on the web page. If the page is refreshed, the timer begins again. I think the countdown is only in the web interface and isn't actually happening on the controller end. When I curl state/chlorinator/1, the superChlorRemaining is always equal, in seconds, to the superChlorHours in hours, it never gets smaller.

**To Reproduce**
Superchlorinate through dashpanel

**Expected behavior**
Dashpanel and API would show amount of time remaining for superchlorinate

**Pool Equipment**
 - Controller:easytouch
 - Controller Model:EasyTouch2 4P
 - Chlorinator: IntelliChlor-20

**Additional context**
I am running the controller in docker from this image https://hub.docker.com/r/msmi/nodejs-poolcontroller

tagyoureit, 2021-06-04:
This is expected behavior.  EasyTouch only ever tells connected clients about the number of hours.  If there are 5 hours left, we can't be sure if there are 4 hours, 59 minutes left or 4 hours, 1 minute left.  If you don't refresh your browser, it should be pretty accurate.  When EasyTouch tells us there are 4 hours left we'll adjust the countdown and start counting down from that time.  

If we tried to synchronize the time between the browser and njsPC there would just be too much traffic and it isn't worth the overhead.

Incidentally, IntelliCenter reports down to the minute so you would see the same behavior if you refreshed the browser before the minute ticked down.

dnikles, 2021-06-04:
Even if I wait hours later it doesn't change. Is that expected?

tagyoureit, 2021-06-04:
No, that is not expected.  

If you go to the OCP (outdoor control panel) or ICP (indoor..) or SL (screenlogic, if you have it) and set the superchlor hours, does it correctly show up in njsPC if you set/change it there?  Do you have any trouble setting the chlorinator set point or any other circuit, pump state?

If that works, then we'll have you submit a packet capture and see what's going on internally.

dnikles, 2021-06-04:
I'll give it a try on the outdoor control panel. I've never set
superchlorinate there, I've only set things to service mode and done it
from the chlorinator itself. I'm not very familiar with this stuff so it
could very well be something I'm doing wrong on my end.

dnikles, 2021-06-04:
I checked it on the outdoor control panel, and now I'm seeing the correct values in the API and the dash panel. I don't know if it had to intialize on the control panel first somehow, I had never been in the chlorinator section on there. Nevertheless, it is working in the manner you suggested, rounding up to the next hour. Thank you.

tagyoureit, 2021-06-04:
Let us know if you continue to have trouble setting it from njsPC/dashPanel.

dnikles, 2021-06-04:
Ok I checked it again, after it should have been done super chlorinating by an hour or two, but it still showed 2 hours left through dash panel and the API. I captured logs and had it reload config to capture the logs. After doing that, it is showing the appropriate status of the chlorinator. I did not check if the chlorinator was actually superchlorinating when it showed it was, before the reload. 
[replay1704.zip](https://github.com/tagyoureit/nodejs-poolController/files/6601240/replay1704.zip)

tagyoureit, 2021-06-04:
Ok, thx.  I'll take a closer look later (I have ET2-8.). I also see it's resetting the timer even if we open up the chlorinator settings pop-up so there are likely some changes we can make here to improve the experience.

tagyoureit, 2021-06-06:
I updated the UI so it won't restart the countdown timer just because you opened the pop-up.

However, I can't tell exactly what's going on with your pool.  At some point, njsPC thought the system was doing a super chlorinate, but in the files you sent me the pump is off and the OCP never sent any updates.  When OCP is in super chlorinate mode, it will force the pumps to be on.  

If you can create a longer packet capture where you can start the super chlorinate, and then get it in a state where it is inconsistent with the OCP that would be helpful.

dnikles, 2021-06-12:
My wife did a super chlorinate 2 days ago but it was still showing super chlorinate in dash panel. It was not still super chlorinating at the chlorinator though. I didn't get to grab logs before she did that, I'll try getting that next time.

tagyoureit, 2021-08-05:
Please re-open if this is still happening.

closed 2021-08-05 by tagyoureit