tagyoureit/nodejs-poolController #314: Can't change temp on heater

Telemachus1977, 2021-06-06:
When I try to change the temperature on the heater its not working. 

Log attached.

[replay (1).zip](https://github.com/tagyoureit/nodejs-poolController/files/6603690/replay.1.zip)

tagyoureit, 2021-06-06:
I added some code in the UI to default the setpoint to the current temp if it isn't set.  Additionally, I added some code so njsPC would not try to send packets with null values.

However, I'm not clear how you got into this pickle.  Can you do a Packet Capture (with reset - it will give you a clean set of configuration files and back up your existing ones to the logs dir)?  If it happens again, I'll be able to track it down but otherwise you seem to be getting all the right packets from the OCP.

Telemachus1977, 2021-06-06:
I think what happened is instead of using the arrows to change the temp I tried typing in the desired temp.  Hasn't worked well since.

tagyoureit, 2021-06-07:
Type in values are fine and we validate they are numeric in the back end. 

Let us know of it is working now and if you happen to recreate this it would be interesting to see but hopefully it's okay now.

Telemachus1977, 2021-06-07:
What do I need to update?  The web UI or the controller?

Telemachus1977, 2021-06-09:
Sorry should I delete the whole ui and reinstall it?

tagyoureit, 2021-06-09:
Do a 'git pull' on both njsPC and dashPanel. Then in your browser do a hard refresh (ctrl–f5) to ensure you have the latest dashPanel version in your browser.

Telemachus1977, 2021-06-11:
Do I need to "npm i" after git pull?

Attached is a log with a reload and then attempts to change temp with heater which gives me this error screen.

tagyoureit, 2021-06-11:
There were no new modules added so you don't need to run `npm i` although it never hurts.

Clearing browser cache on mobile devices is awful.  For Safari (iOS) you need to go to settings and literally erase your life.  For Chrome, you have a little bit more flexibility.  See [link](https://support.google.com/accounts/answer/32050?co=GENIE.Platform%3DiOS&hl=en).  You might want to try the `cached images and files`; it should be there.

closed 2021-06-11 by tagyoureit