tagyoureit/nodejs-poolController #315: Getting no data from IntelliChem which is directly connected to RS-485.

ibehren1, 2021-06-06:
**Describe the bug**
I have raspberry pi setup running poolController and dashPanel. It is connected to the IntelliChem via RS-485 serial to USB controller. I am getting no data. I have swapped the data+ and data- and also ordered/tried a second adapter all to no avail.

I am using this https://smile.amazon.com/gp/product/B081MB6PN2/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1 adapter and have connected Ground to Ground, Data+ to A+ and Data- to B-.

Using outdoor rated Cat 6 cable for the connection -- total cable length is about 5 ft.  Using green for ground and blue/blue white for data.

**Expected behavior**
Expect poolController/dashPanel to see equipment or see some data with `od -x /dev/ttryUSB0`

**Screenshots**
[image]

**Packet Capture**
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/6604959/replay.zip)

**Pool Equipment**
 - Chemical controller: IntelliChem

**Additional context**
Add any other context about the problem here.

rstrouse, 2021-06-06:
Did you click on the gears and add the IntelliChem to your setup?

ibehren1, 2021-06-06:
Upon trying to add the IntelliChem I get the following:

[image]

```
ApiError: Message aborted after 2 attempt(s): 165,0,16,16,146,21,2,218,1,144,0,0,2,193,0,0,0,0,45,0,0,0,0,0,0,0,0,3,201
    at SendRecieveBuffer.writeMessage (/home/ibehr/nodejs-poolController/controller/comms/Comms.ts:357:27)
    at SendRecieveBuffer.processWaitPacket (/home/ibehr/nodejs-poolController/controller/comms/Comms.ts:292:29)
    at SendRecieveBuffer.processOutbound (/home/ibehr/nodejs-poolController/controller/comms/Comms.ts:300:26)
    at SendRecieveBuffer.processPackets (/home/ibehr/nodejs-poolController/controller/comms/Comms.ts:283:21)
    at Timeout.exports.conn.buffer.procTimer.timers_1.setTimeout (/home/ibehr/nodejs-poolController/controller/comms/Comms.ts:328:59)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
```

I was choosing address `1` as there is no `0`.  Per the following screenshot and reference in the manual `HA=00` means it is using address `0` and `R0` means that it is running in standalone mode because it has not seen packets.

rstrouse, 2021-06-08:
We are getting some more information regarding Intellichem.  I believe we only get a response on that message if an OCP is present.

ibehren1, 2021-06-08:
Thanks!  I appreciate the help!

ibehren1, 2021-06-09:
Just a bit of update.  Today I got a cable for the Intelliflo VSF pump and connected it.  It shows up correctly and I am able to control it, so that verifies my RS-485 adapter and config on the Pi...  No change in the behavior of the IntelliChem

rstrouse, 2021-07-02:
Grab a new njsPC and dashPanel.  These should now control a standalone IntelliChem.  I don't have one so I will need help from you to work through any potential issues.

rstrouse, 2021-07-02:
The KA packet is probably the action 210 above.  IntelliCenter only sends 210 and 146 to the controller.  We will expect an 18 in response to the message.  Address 1 is probably the correct one for a non-commercial unit.  Are you sure you have the green and yellow wires correct?  Also, I do not know if it needs the black/red combo for power over the RS485 but that would be really stupid.

rstrouse, 2021-07-02:
I also noticed there is a TX light on the dongle are you getting flashing there?  You should not be getting flashing over the RX since IntelliChem is not responding.  The one other thing although I don't think it matters is that you have the ground hooked up to the black wire.  The black is not necessarily ground on the RS485 bus.  When combined with the red it is power for external controllers and equipment.  A good example is a radio transceiver for remotes.

ibehren1, 2021-07-02:
Did a bit more playing with it...  I do get RX when the pump is connected.  Disconnecting it I do not get RX light.  Also moved the pump to address 2 since it is configurable.

Validated wiring.  I have updated since the earlier pictures.  Also tried switching the IntelliChem wires with no success there either.

Updated pictures.
[image]

The yellow/green wires come from the pump the blue/white from the IntelliChem.
The disconnected pair go to the UltraTemp Heat Pump.

Inside of the IntelliChem
[image]

IntelliChem wiring diagram
[image]

rstrouse, 2021-07-03:
Perhaps the source address is important to IntelliChem.

I changed njsPC to talk to IntelliChem on 16 to see what it has to say about that so let me know how it goes.

ibehren1, 2021-07-03:
Wow!  I think you got it!!!  Awesome!

[image]

And seeing packets back and forth.
[image]

Going to give it a bit of time and see how things go.

Thanks for all the work!!!

rstrouse, 2021-07-03:
Yeah that is exactly what I was hoping to see!

rstrouse, 2021-07-03:
As an aside, address 1 for your pump and address 1 for IntelliChem are two distinct address ranges they will not and cannot conflict.  Each equipment class has its own 16 value range.
At a high level:
Pump Address 1 = 96
IntelliChem Address 1 = 144
UltraTemp Address 1 = 112

ibehren1, 2021-07-03:
Excellent info to know....

An issue has popped up this morning.  Noticed on the IntelliChem that both the pH and ORP tanks were marked empty and had reached dose limit.  I reset/cleared each on the IntelliChem panel.

Came back to the dashPanel and it was not responding.  Investigating, it looks like nJSPC is continually restarting (via systemd).  Manually restarting I am seeing the following output:

```
ibehr@pool-monitor:~/nodejs-poolController $ node dist/app.js
Init state for Pool Controller
[03/07/2021, 12:19:48] info: The current git branch output is master
[03/07/2021, 12:19:48] info: The current git commit output is 9a1c597d2cb826fba878a86bdd31a5fa32704cba
[03/07/2021, 12:19:49] info: Starting up SSDP server
[03/07/2021, 12:19:49] info: Checking njsPC versions...
[03/07/2021, 12:19:49] info: Starting Pool System nixie
[03/07/2021, 12:19:49] info: Initializing Nixie Control Panel for Nixie Single Body
[03/07/2021, 12:19:49] info: Initializing Intake/Return valves
[03/07/2021, 12:19:49] info: Initializing Nixie Controller
[03/07/2021, 12:19:49] info: Initializing body Pool
[03/07/2021, 12:19:49] info: Initializing Filter Filter
[03/07/2021, 12:19:49] info: Initializing Nixie circuit Pool
[03/07/2021, 12:19:49] info: Initializing chemController IntelliChem
[03/07/2021, 12:19:49] info: NCP: Setting Pump Intelliflow VSF to 2000 RPM.
[03/07/2021, 12:19:49] info: Initializing Nixie Pump 50-Intelliflow VSF
[03/07/2021, 12:19:49] info: Nixie Controller Initialized
[03/07/2021, 12:19:50] info: Nixie Single Body control board initialized
[03/07/2021, 12:19:50] info: Server is now listening on 0.0.0.0:4200
[03/07/2021, 12:19:50] warn: Comms port is not open.Message aborted: 165,0,144,16,146,0,2,238,2,238,7,7,2,166,0,2,0,0,30,0,0,0,0,0,0,0,0,1,215 
[03/07/2021, 12:19:51] warn: Comms port is not open.Message aborted: 165,0,97,33,6,1,10,1,56 
[03/07/2021, 12:19:51] error: Error sending setDriveState for Intelliflow VSF : Comms port is not open.Message aborted: 165,0,97,33,6,1,10,1,56 
[03/07/2021, 12:19:51] error: Error running pump sequence for Intelliflow VSF: Comms port is not open.Message aborted: 165,0,97,33,6,1,10,1,56 
[03/07/2021, 12:19:51] error: Nixie Error running pump sequence - OutboundMessageError: Comms port is not open.Message aborted: 165,0,97,33,6,1,10,1,56 
[03/07/2021, 12:19:51] warn: Comms port is not open.Message aborted: 165,0,144,16,210,1,210,2,234 
[03/07/2021, 12:19:51] info: Serial port: /dev/ttyUSB0 request to open successful
[03/07/2021, 12:19:51] debug: Connection opened from init function;
[03/07/2021, 12:19:51] silly: SSDP/UPnP Server started.
[03/07/2021, 12:19:53] verbose: Wrote packet[255,0,255,165,0,97,33,6,1,10,1,56].Retries remaining: 1 
[03/07/2021, 12:19:53] verbose: Wrote packet[255,0,255,165,0,97,33,1,4,2,196,7,208,2,201].Retries remaining: 1 
[03/07/2021, 12:19:54] info: {"id":8,"valid":true,"dir":"out","proto":"intellichem","pkt":[[],[255,0,255],[165,0,144,16,146,21],[2,238,2,238,7,7,2,166,0,2,0,0,30,0,0,0,0,0,0,0,0],[4,162]],"ts":"2021-07-03T12:19:54.700-0500"}
[03/07/2021, 12:19:54] verbose: Wrote packet[255,0,255,165,0,144,16,146,21,2,238,2,238,7,7,2,166,0,2,0,0,30,0,0,0,0,0,0,0,0,4,162].Retries remaining: 3 
[03/07/2021, 12:19:54] info: {"id":9,"valid":true,"dir":"in","proto":"intellichem","for":[8],"pkt":[[],[255,0,255],[165,0,16,144,1,1],[146],[1,217]],"ts": "2021-07-03T12:19:54.826-0500"}
[03/07/2021, 12:19:54] info: {"id":10,"valid":true,"dir":"out","proto":"intellichem","pkt":[[],[255,0,255],[165,0,144,16,210,1],[210],[2,234]],"ts":"2021-07-03T12:19:54.854-0500"}
[03/07/2021, 12:19:54] verbose: Wrote packet[255,0,255,165,0,144,16,210,1,210,2,234].Retries remaining: 3 
[03/07/2021, 12:19:55] debug: Invalid enumeration: val = false map = {}

/home/ibehr/nodejs-poolController/controller/State.ts:2058
                this.doseHistory.unshift(dose);
                                 ^
TypeError: Cannot read property 'unshift' of undefined
    at ChemicalORPState.endDose (/home/ibehr/nodejs-poolController/controller/State.ts:2058:34)
    at Function.processState (/home/ibehr/nodejs-poolController/controller/comms/messages/status/IntelliChemStateMessage.ts:237:27)
    at Function.process (/home/ibehr/nodejs-poolController/controller/comms/messages/status/IntelliChemStateMessage.ts:31:41)
    at Inbound.process (/home/ibehr/nodejs-poolController/controller/comms/messages/Messages.ts:526:41)
    at SendRecieveBuffer.processCompletedMessage (/home/ibehr/nodejs-poolController/controller/comms/Comms.ts:463:17)
    at SendRecieveBuffer.processInbound (/home/ibehr/nodejs-poolController/controller/comms/Comms.ts:502:48)
    at SendRecieveBuffer.processPackets (/home/ibehr/nodejs-poolController/controller/comms/Comms.ts:296:21)
    at Timeout.timers_1.setTimeout [as _onTimeout] (/home/ibehr/nodejs-poolController/controller/comms/Comms.ts:268:109)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)```

rstrouse, 2021-07-03:
I fixed this this morning.  Since there is no history the initial values for the history array was not defined.  Pull njsPC

ibehren1, 2021-07-03:
It did indeed fix the issue.  You are on it!

rstrouse, 2021-07-03:
If you do another pull it should now operate your UltraTemp.

rstrouse, 2021-07-06:
I am going to close this one as I think we have this all worked out.  Open another issue if you see any.

closed 2021-07-06 by rstrouse