tagyoureit/nodejs-poolController #316: Intellichem Packet Analysis

davthomaspilot, 2021-06-07:
I'm doing experiments to verify Intellichem packet understanding and will post results in this issue.  I posted a spreadsheet which shows packets for each experiment, and does the calculations/decodes as documented so far.  Maybe someone can spot how to fix some of the discrepancies from this (e.g, that's in byte x instead of byte y).

I'll have a sheet for each experiment and will color code what I verified ok (green) or known bad (red).  Any other color is just highlighting, usually to indicate another experiment needed to veriy.
Not finished yet, but here's but here are the discrepancies I suspect so far:

Bytes 11-12 dose time.    Time in seconds is in byte 11, not byte 12.  I need to increase the dose time on the Intellichem to something greater than 256 seconds to see what packet byte has the 256x field.

But, ORP Dose time bytes are correct.

Looks like decode of dosing status has "monitoring" and "mixing'  reversed ("dosing" looks good.  I think this (not sure yet) because during a dose, the field indicates "dosing", but after the dose, it goes to "monitoring".  Then, several minutes later (probably the mixing time), it does to "mixing".  

I didn't capture what I have set for mixing time for ph and ORP.  I'll set those to something known (and short) and repeat the experiment to validate my conclusion.

The dose volume fields are a fixed multiplier of the dose time fields.  I suspect the multiplier changes when dose units change.  I want to repeat the experiment for that with the packets captured and in the spreadsheet.

Firmware version isn't consistent with what ScreenLogic reports.

Flow field  (byte 32, bit 0) is always zero.  

More to come.

[Intellichem.xlsx](https://github.com/tagyoureit/nodejs-poolController/files/6610496/Intellichem.xlsx)

davthomaspilot, 2021-06-08:
pH Dose time is in bytes 10-11, not 11-12 as documented.  But, I think dashPanel is using 10-11, so it's just a documentation error.

rstrouse, 2021-06-08:
Is your pH dose time set to 2 minutes?

davthomaspilot, 2021-06-08:
It's 256 x byte 10 + byte 11.  Big Endian.  Just llke what's documented. 

I can post an updated Excel, if you have it.  Here's a screen shot, notice column 10 increments only after column 11 rolls over 256:

[image]

But, dashPanel isn't showing 256*byte11 when present "dosing x of y"  It may already be using bytes 10 and 11.  At least it's using byte 11 NOT multiplied by 256.

davthomaspilot, 2021-06-08:
Here's an updated Excel with extra sheets with dosing experiments.  Just had to keep increasing dose to see byte 10 change.  

Actually, I get "don't support that file type" when I try uploading xlsm.  It has a couple of VBA methods for BITWISE_AND to for decoding alarm and warning fields.  So, an xslx version (no macros) won't work right.  If you're interested, I could change so it doesn't need macros.  Or I could email a version.  Or just post screenshots to support conclusions about issues I see.

davthomaspilot, 2021-06-08:
Found this in controller/comms/messages/status: 

//      11-12 : pH Dose time
        scontroller.ph.doseTime = (msg.extractPayloadByte(11) * 256) + msg.extractPayloadByte(12)

Should be bytes 11 and 12.  But, dashPanel shows "Dosing x of y" in chemistry panel, it seems it's displaying  byte 11 (without multiplying by 256).  I'll try to capture what dashPanel is showing, but from above packet capture, we want bytes 10-11, not bytes 11-12.  

This assumes messages logged from messageManager are accurate! (My analysis is from packets logged to file from messageManager.

davthomaspilot, 2021-06-08:
in controller/boards/SystemBoard.ts:

```
public chemControllerDosingStatus: byteValueMap = new byteValueMap([
    [0, { name: 'dosing', desc: 'Dosing' }],
    [1, { name: 'monitoring', desc: 'Monitoring' }],
    [2, { name: 'mixing', desc: 'Mixing' }]
  ]);
```
Should be:

public chemControllerDosingStatus: byteValueMap = new byteValueMap([
    [0, { name: 'dosing', desc: 'Dosing' }],
    [1, { name: 'mixing', desc: 'Mixing' }],
    [2, { name: 'monitoring', desc: 'Monitoring' }]
 
  ]);

davthomaspilot, 2021-06-08:
The pH and ORP volume columns don't really convey any extra information.  The volume columns are always 2.18 time the time columns (truncated to integer).

Time columns are elapsed time in seconds, even when "dosing by volume".  Same ratio whether ounces or mL are used on the Intellichem panel.

davthomaspilot, 2021-06-08:
The flow bit (bit 0) in byte 32 (Alarms) is accurate when Intellichem is powered on.  (I can get low flow by keeping pump on, but at low rpm).  But, it doesn't indicate when there is no flow because of IntelliChem being powered off--it indicates flow is OK (0).

Byte 35 is 2 when the Intellichem panel on ScreenLogic shows "-" for ph and ORP readings.  This is what it shows when Intellichem has power off (since pump motor off) but also during the turn on delay period.  So, ORP and pH readings should not be displayed whey Byte 35 = 2

davthomaspilot, 2021-06-21:
Just noticed there's a control in Intellichem setup for doser pump rated output.  "0" if unknown, else gallons per day.

I suspect that will affect the ratio of the dosing volume / dosing times.   For the dosing volume column to be useful, the client (like dashPanel) will have to know the rated output.  But, given that, simply using the time column and multiplying by the appropriate volume/sec parameter provices the same thing.  So, I think the client should be parameterized for the pump output (and tank volumes)..

davthomaspilot, 2021-06-21:
I confirmed that the ratio of the dosing volume to dosing time columns is determined by what's been set up in Intellichem for the daily pump capacity.   

For example, if I set my pump volume to 10 gallons per day, the dose volume columns ratio to the dose time column will be .43813 * the dose time colum
 
The dose volume per second (mL) will be that 10 gallons/day*  3.78541 liters/gallon  * day/)3600*24_ seconds  * 1000 mL/Liter .

10 gpd is the what the pump volume for the Stenner pumps shipped with Intellichem.

rstrouse, 2021-06-25:
Ok so I found a stumbling block that I need your help with.  In the past we have had other users that did not report the data between ScreenLogic, and Indoor Control Panel, or a Wireless Control panel for the Touch controllers.

This message is known to exist and we have a pretty good idea as to its payload but we have never seen it.  So if you could, could you create a replay WITHOUT CAPTURING THE CONFIGURATION RELOAD at the time you are changing setpoints on ScreenLogic or a controller that is not the OCP?

davthomaspilot, 2021-06-25:
Sure.  But I'm not sure exactly what to do.   Still just action 18 packets?  Or, some unknow action type?

"WITHOUT CAPTURING THE CONFIGURATION RELOAD".  Not sure what that means, or how to do it?  I don't mind hacking a code fragment to filter packets we don't want to see.

rstrouse, 2021-06-25:
No this should capture everything.  I cannot believe in all the IntelliChem captures I have I do not have one that shows the communication between ScreenLogic and the OCP.  If you go to the Logging tab under the hamburger menu you will see a Capture Replay button.  When the begin capture prompt comes up I want you to uncheck Capture Configuration Reload.  Then begin capture.  After that go to ScreenLogic, an Indoor Panel, or wireless remote and change the pH and ORP setpoints.

We have a good handle on the 18, 19, 210, 147, and the 146 messages but there is one more that is alluding us and I do not know what its signature is.

davthomaspilot, 2021-06-25:
[replay (8).zip](https://github.com/tagyoureit/nodejs-poolController/files/6719137/replay.8.zip)

davthomaspilot, 2021-06-25:
I've hacked my local copy of the code--I hope it doesn't impact what you want to see.  Let me know if the capture doesn't make sense--I can undo my changes are re-install if necessary.

rstrouse, 2021-06-25:
I'll have a look.  I am really looking for something very specific.  Also, does the dosingTime/Volume count up or down?  Have you seen anything that indicates the target amount it wants to dose?

davthomaspilot, 2021-06-25:
dosingTime/Volume counts up.

dosingTime is always just the elapsed time--increments by 2 (2 seconds each time).

the dosingVolume columns always counts up.  It's just the dosingTime column multiplied by a constant.  The constant is based on what is setup in Intellichem for the dosing pump's gallons per day.

I'd suggest that the client has the ability to set the volume dosed per second AND the tank volumes.  That's what I did for my hack in Hubitat.

davthomaspilot, 2021-06-25:
So, the dosingVolume has no useful information, except maybe that you could calculate what Intellichem has set for the gpd for each pump.  Really,  it always doses by time.  If you say dose by volume, it just divides the dose volume by the volume per second.

rstrouse, 2021-06-25:
This is the message I was looking for.  Hmmmm.... the dosing volume and time do not increase together?  If they do then I know the constant.
[image]

davthomaspilot, 2021-06-25:
Yes, the dosing volume and time DO increase together.  Again, dosing volume is always a constant times the dosing time.

davthomaspilot, 2021-06-25:
See my posts above--the constant is based on the what is set up in Intellichem for the pump gallons per day.  Just convert that to mL/sec--that's the constant/.

rstrouse, 2021-06-25:
Yeah so if I know the volume from the previous message and the time from the previous message I can break that down into the units I need (mL/min) which will be somewhere shy of 125.   What I would really like is the target mixing and dosing time so we can show when it should be done and how much it has to go.  Apparently 4 bytes of data is way too much to ask for from the controller.

davthomaspilot, 2021-06-25:
Yes, it would be nice to know "how much to go", but the Intellichem packets (action 18) don't have that.  

Do you think that new packet might have the info?

davthomaspilot, 2021-06-25:
If one really wanted "how much to go", you could keep track of the maximum dose seen when target was still not the same as actual.  I keep track of how much total has been dosed by accumulating the time (or volume) when the (corrected) status changes from dosing to mixing.

rstrouse, 2021-06-26:
No sadly it is a replica of the 146 message with the address of the controller - 144 prepended to the message.  The problem isn't so much dosing because these times are typically pretty short but it is nice to know how long is left in the current dose if you want to interrupt it.  But IntelliChem doesn't give you that option either.  The issue comes with the mixing because it doesn't really tell us the last time it dosed.  So knowing how long it has to go before a reliable change to the pH or ORP is to occur is really braindead.

I am going to try to get as much parity with the REM Chem functionality as I can but the countdowns are probably going to be sacrificed to the limitation gods.  As well manual dosing and manual mixing delays are falling into that category as well.

rstrouse, 2021-06-26:
I am going to auto-calc CSI instead of LSI if you want it as well as manipulate the low/high pH and ORP warnings.

rstrouse, 2021-06-26:
So did you have any luck determining what units the doseVolume is in?  They definitely look like mL but perhaps they are in .1oz?  For instance your Chlorine dose above @151oz pumps over a gallon.  That is not likely.  151mL does seem likely but 15.1oz also makes some sense.

I have finished up the CSI/LSI calcs as well as revamped the messages related to Touch controllers.  There is just this little tidbit left.  I would like to be able to maintain the dose history if can but the volume is a key part of that.

davthomaspilot, 2021-06-26:
Short answer, yes, it's in mL.  But the volume columns really have no informative value.

Again, the volume column is just the time column multiplied by a constant.  That constant is defined when you set up Intellichem pump volumes.  Once option in the set up for the pump volume is "unknown".  That is what I had initially.

When I went back and set it to 10 gallons per day, confirmed:

`The dose volume per second (mL) will be that 10 gallons/day* 3.78541 liters/gallon * day/)3600*24_ seconds * 1000 mL/Liter .`

Rather than use the volume column, it would be much more useful to have the constant defined in the client, in either pump gallons per day or liters per day.  And keep track of the total seconds dosed and present as a volume based on the constant defined.

The Intellichem setting is buried and I doubt installers dig up the pump daily capacity when an option is "unknown".

davthomaspilot, 2021-06-26:
Dose pump daily capacity is in the "Auto Setup Wizard"

[image]

davthomaspilot, 2021-06-26:
I can set up the chlorine pump capacity to verify it works the same as I verified for pH, if you want.  I didn't bother, since I'm just using the dosing time column.

rstrouse, 2021-06-26:
Actually the units of the constant are mL/sec if the time column is in seconds and the volume column is it mL.
So if I have the previous time lets say 1 and the current time lets say 5 and the previous volume lets say 2 and the current volume lets say 10.

I should be able to subtract the previous value from the current value and know what the mL/sec actually is.  So this would be equal to 8/4mL/sec or 120mL/min.  I use the exact pump as you do and mine calibrates to exactly 112.5mL/min.  Since I also have the start of the dose I also would know what the total dose was when it was done.

ibehren1, 2021-06-27:
I believe it it .1 oz per second.

If I do a manual dose, for my pH, it will do 3.0oz and takes 30 sec.  I have max dose set at 3oz.

Looks like it does have option to set it to either oz or ml thought...

Manual: https://www.pentair.com/content/dam/extranet/nam/pentair-pool/residential/manuals/automation/intellichem/intellichem-water-chemistry-controller-manual-english.pdf

Page 12 (in the manual 18 in the PDF)

Dose (Volume): Displays the current dose size (0-9999) in ounces [oz] or milliliters [ml].
Set the amount of ounces (or milliliters) that the IntelliChem controller will feed pH (acid)
each time the IntelliChem controller calls for pH chemicals. The Dose (Volume) is set by
Auto Setup wizard based on pool volume and filter run time.

Dose (Time/Setpoint): Displays the current pH dose (feed) time. If pool volume is less
than 4k gallons, these units automatically get set to milliliters (ml) for better resolution.
Sets the amount of time in hours, minutes and seconds that the IntelliChem controller will
feed pH (acid) each time the IntelliChem controller calls for pH chemicals. The pH dose
time is set by Auto Setup. The Dose (Time) is set by Auto Setup wizard based on pool
volume and filter run time. Dose (to Setpoint) does not have a dose amount, only a time
limit.

rstrouse, 2021-06-27:
What is really odd is that the pump rating (at least on the newer tanks) is 4.5mL/day.  These are stenner pumps 45M and the output is fixed.  If you have the 3 roller design then this is the pump you have.  The older pumps were 2 roller and put out more volume per rotation but there were issues with them.

There are 3 byte locations that are likely candidates for the units on the 18 message.  However, I suspect we have only been looking at these when the setting was mL.  I wonder if byte 8, 9, or 10 change to a different value when dosing in oz which at 2 seconds between messages makes for a very inaccurate measurement.
[image]

ibehren1, 2021-06-27:
Mine is the 3 roller version. [image]

davthomaspilot, 2021-06-27:
Yes, the constant is mL / second.  It can be calculated from the gallons / day that is set up in Intellichem.

_I should be able to subtract the previous value from the current value and know what the mL/sec actually is_

Divide by 2, since the time column increments by 2 seconds each new packet.  This will give you the constant in mL / sec.

Previously, I tried setting dose limits in  both mL and ounces in Intellichem.  It did not change what's in the packets.   Apparently, it just allows you to enter a dose limit in volume (either mL or ounces) instead of seconds.  Intellichem does the math based on what you have set up for the pump capacity.

Bytes 8, 9, and 10 have always been zero.

You seem skeptical about my conclusions.  So, I'll  do some more  packet captures with Intellichem set up with different pump capacities.   I'll duplicate at least one with dose volume at mL and another in ounces.  (I did that at least once already, but did not document) .  

But, I'm confident   the volume columns are just the time column multiplied by the volume per unit time constant that is calculated from the pump capacity set up in Intellichem.

davthomaspilot, 2021-06-27:
I tried uninstalling using HPM, but got a fatal error.  Then I did a repair and it seemed to work ok, but I didn't have an Intellichem device.

Then  I did a configure in the Pool Controller device.  Still no Intellichem.  I see messages like this in the log:

```
2021-06-27 09:15:35.590 am errororg.codehaus.groovy.runtime.metaclass.MissingMethodExceptionNoStack: No signature of method: user_driver_bsileo_Pool_Controller_Intellichem_520.parse() is applicable for argument types: (java.util.HashMap) values: [[calciumHardness:325, cyanuricAcid:0, siCalcType:0, type:[val:2, ...], ...]]
Possible solutions: use([Ljava.lang.Object;), wait(), run(), run(), grep(), any() (parse)
dev:2792021-06-27 09:15:34.294 am errororg.codehaus.groovy.runtime.metaclass.MissingMethodExceptionNoStack: No signature of method: user_driver_bsileo_Pool_Controller_Intellichem_520.parse() is applicable for argument types: (java.util.HashMap) values: [[calciumHardness:325, cyanuricAcid:0, siCalcType:0, type:[val:2, ...], ...]]
Possible solutions: use([Ljava.lang.Object;), wait(), run(), run(), grep(), any() (parse)
dev:2362021-06-27 09:15:33.850 am infoAir Temperature value is 74°F
dev:2372021-06-27 09:15:33.831 am infoSolar Temperature value is 73°F
dev:2792021-06-27 09:15:32.321 am errororg.codehaus.groovy.runtime.metaclass.MissingMethodExceptionNoStack: No signature of method: user_driver_bsileo_Pool_Controller_Intellichem_520.parse() is applicable for argument types: (java.util.HashMap) values: [[calciumHardness:325, cyanuricAcid:0, siCalcType:0, type:[val:2, ...], ...]]
Possible solutions: use([Ljava.lang.Object;), wait(), run(), run(), grep(), any() (parse)
dev:2792021-06-27 09:15:30.265 am errororg.codehaus.groovy.runtime.metaclass.MissingMethodExceptionNoStack: No signature of method: user_driver_bsileo_Pool_Controller_Intellichem_520.parse() is applicable for argument types: (java.util.HashMap) values: [[calciumHardness:325, cyanuricAcid:0, siCalcType:0, type:[val:2, ...], ...

```

I'm going to try uninstalling again, and then doing install instead of repair.

davthomaspilot, 2021-06-27:
After doing the repair, there are two versions of the app to install.   Seems I need to clean everything up manually and try again.

davthomaspilot, 2021-06-27:
I manually deleted all the driver code, app code, devices, and app.  Then, the HPM uninstall was happy.

I then re-installed Pool Controller.  It said 0 Intellichem created, but I see an Intellichem 1 as a child device for EasyTouch 2.4.  I see this in the log file:

`021-06-27 09:29:53.289 am errororg.codehaus.groovy.runtime.metaclass.MissingMethodExceptionNoStack: No signature of method: user_driver_bsileo_Pool_Controller_Intellichem_528.parse() is applicable for argument types: (java.util.HashMap) values: [[calciumHardness:325, cyanuricAcid:0, siCalcType:0, type:[val:2, ...], ...]]`

davthomaspilot, 2021-06-27:
I did auto-setup in Intellichem and set the daily volume for the pH pump at 15 gallons per day and the the ORP pump to 25 gallons per day.   The per dose volume was set to 2 ounces for pH and 10 mL for ORP.

Then I initiated a dose using ScreenLogic by moving pH and ORP setpoints.  Here's a screenshot of the resulting Excel spreadsheet from the packet info:

[image]

I then did a small spreadsheet that calculates the expected doseTime and doseVolume columns given the pump capacity configuration and per dose maximum volume.   They match nicely:

[image]

The conclusions:

1. doseVolume columns do not depend on the units chosen for maximum volume per dose
2. columns 8 and 9 are always zero regardless of units used for maximum volume per dose.  " I wonder if byte 8, 9, or 10 change to a different value when dosing in oz. Note!  Columns 10 and 11 are the ph Dose time, not columns 11 and 12!!
3.

rstrouse, 2021-06-27:
I finally got around to changing the docs for the doseTime.  There are 2 messages that come through that are almost identical but the 147 (which comes from the OCP) is shifted by 1 byte to identify the address of IntelliChem.  Btw I am not doubting you I am just making sure we are squeezing all the juice out of the information IntelliChem provides and fit it into the external interfaces.

davthomaspilot, 2021-06-27:
If the packet was shifted by one byte, all the fields would be inaccurate.  Not just the ph dose time.

Yes, it would be really good if we could find packets that contain the setup information.  Especially mixing time, and max per dose and per day values.  

Note, the docs were  also wrong about dosingStatus and alarms.  Or, maybe you changed those too.

davthomaspilot, 2021-06-27:
Should I explore packets with different action types to look for the Intellichem set up data?   Are there  unknown fields that show up when the configuration packets are sent?

davthomaspilot, 2021-06-27:
The messages from njsPC to Hubitat include tank capacities:

`... equipmentType:tank, capacity:6, master:0], pump:[isDosing:false, ratedFlow:0, type:[val:0, name:none, ratedFlow:false, tank:false, remAddress:false, desc:No Pump], equipmentType:pump, enabled:true, master:0], tolerance:[high:800, low:650, enabled:true], dosingStatus:[val:2, name:Monitoring, desc:Monitoring], flowDelay:false, flowOnlyMixing:true, setpoint:760, maxDosingVolume:0, level:815, delayTimeRemaining:0, probe:[level:815, equipmentType:probe, enabled:true, saltLevel:0, master:0], manualDosing:false, master:0, dosingMethod:[val:0, name:manual, desc:Manual], lockout:false, dailyLimitReached:false, dosingVolumeRemaining:0, doseTime:0, phLockout:7.8, startDelay:1.5, flowReadingsOnly:true, chemType:orp, mixTimeRemaining:0, doseVolume:0], alkalinity:70, master:0, ph:[enabled:true, equipmentType:ph, dosePriority:true, maxDailyVolume:500, maxDosingTime:0, phSupply:[val:1, name:acid, desc:Acid pH-], dosingTimeRemaining:0, tank:[level:6, alarmEmptyEnabled:true, units:[val:0, name:, desc:No Units], alarmEmptyLevel:20, equipmentType:tank, capacity:6, master:0], pump:[isDosing:false, ratedFlow:0, type:[val:0, name:none, ratedFlow:false, tank:false, remAddress:false, desc:No Pump], equipmentType:pump, enabled:true, master:0], tolerance:[high:7.6, low:7.2, enabled:true], dosingStatus:[val:2, name:Monitoring, desc:Monitoring], flowDelay:false, flowOnlyMixing:true, setpoint:7.6, maxDosingVolume:0, level:7.5, delayTimeRemaining:0, acidType:0, probe:[level:7.5, temperature:89, tempUnits:[val:0, name:F, desc:Fahrenheit], equipmentType:probe, enabled:true, master:0], manualDosing:false, master:0, dosingMethod:[val:0, name:manual, desc:Manual], lockout:false, dailyLimitReached:false, dosingVolumeRemaining:0, doseTime:0, startDelay:1.5, flowReadingsOnly:true, chemType:ph, mixTimeRemaining:0, doseVolume:0], name:IntelliChem 1, `

I don't see that in the action 18 packets.  Is it in another, or is it just hardcoded?

rstrouse, 2021-06-27:
It is just hardcoded because IntelliChem only tells us the unitless value of 0-6.  Which is [1-7] - 1.

davthomaspilot, 2021-06-27:
So, I guess the tank levels 1-6 are 1/6 units of full capacity?

For example, I have set tank capacities of 4 gallons for acid and 5 gallons for chlorine.   So a "3" means 1/2 full.  For me, that's 2.5 gallons on the chlorine tank and 2 gallons on the acid tank.

Illustrates on reason why I suggest  setting tank capacity in the client.

rstrouse, 2021-06-27:
Yeah but keeping all that together with IntelliChem will yield really painful support issues between what is set up in IntelliChem and what exists in njsPC.  None of that data is available to njsPC.  Unfortunately when you tell IntelliChem that you have new levels you can only choose increments of 1 up to 6.  

This lack of data is the primary reason I aborted the purchase of IntelliChem and built REM Chem.  I have managed to get some of the robust portions of REM Chem into IntelliChem, but since njsPC is not in control of anything that is limited to the data that is provided by IntelliChem.  Heck you can't even start a manual dose, cancel a current one, or set a countdown for when you want the monitoring to start after adding chemicals.  REM Chem does all of that plus it lets you calibrate the dosing pumps and all the equipment through the REM interface.

davthomaspilot, 2021-06-27:
"Unfortunately when you tell IntelliChem that you have new levels you can only choose increments of 1 up to 6." 

I think you meant what Intellichem reports as tank level.  You don't have that restriction when setting the tank capacity in Intellichem set up.

I'll continue to use the dose time and have a tank capacity and dosing volume per second constants set in my Hubitat app.  All that I need to get accurate dosing is the total dosing time.  I get that by accumulating the dose time whenever dosing status changes from dosing to mixing.

I'll switch to REM at some point--but I'm heading out of town for a month this week.  Probably be late August or later before I get to it.   With Brad's help getting a working Intellichem driver in Hubitat, I'll now be able to monitor the tank levels at very good resolution (about .25 mL resolution)

rstrouse, 2021-06-28:
Actually through the RS485 interface you are limited to the values 0-7 where 0 means the tank is not installed and [1 through 7] - 1 is the tank level.  If you try to send anything else it will create an error.  I have the model saving each dose when it completes as well as the total dosed over the timeframe as well as the acid demand.  Both dosing and mixing count up in IntelliChem.

davthomaspilot, 2021-06-28:
Yes, I ignore the tank level from the Intellichem entirely.  Instead, I keep track of the amount dosed and accumulate it (in the Hubitat driver/app) whenever dosing status changes from dosing to mixing.   

When Intellichem starts dosing again, the values from Intellichem go back to zero.  That's why it's necessary for the client to capture and accumulate the total after each dose.

That's how I get the .25 mL resolution.  

I just do the math for converting to volume in the client--don't need that info from Intellichem.  Same with tank volumes.

So, I get the tank low alarms via Alexa echo devices.

davthomaspilot, 2021-06-28:
It would be different if Intellichem was actually measuring dosed volume and tank levels.  It is  not.  It just doses by time and calculates a corresponding dose volume and remaining tank capacity.  It needs what's set up in Intellichem (dose pump daily capacity and tank volumes) to do that.  

I just do the math instead in the client code.  It's a whole lot easier to set the pump capacity and tank volumes there, than via the Intellichem setup wizard.  At least in my case--the Intellichem display is virtually unreadable when the sun is shining, not the mention the need to kneel to be at eye level with it.

rstrouse, 2021-07-02:
njsPC should now be accumulating each dose.  I will log these doses in the chemDosage_[acid | orp].log files.  There is the ability to clear the log from dashPanel but it does not clear it from the IntelliChem controller.  I don't know how that might be done but I do know that ScreenLogic keeps that information but I don't know if you can actually clear the 24 hour dosage to allow it to start dosing again when it has hit the max daily.

I also, rerouted the communication through the OCP when making a change to any of the attributes for IntelliChem when an OCP is in the mix.  The OCP insists on being in control so I decided that we should acquiesce and let it be the boss.

The configuration in dashPanel also now allows you to choose either LSI or CSI but you must supply your body volume to calculate CSI and it will take borates into account if you use boric acid.  Unfortunately, IntelliChem only reports a tiny amount of the configuration data and allows control of even less of it.  

I wonder if the probes would work on an Atlas circuit.  If they did you would used everything except the braindead IntelliChem controller.  Either way njsPC catches the doses at their transitions so at least we now know how much and how long it dosed.

tagyoureit, 2021-08-05:
Please open a new issue if there are any more questions about this.  We've done a lot of refactoring on this code base.

closed 2021-08-05 by tagyoureit