tagyoureit/nodejs-poolController #318: Heat Pump type heater does not turn on.

sonapsent, 2021-06-09:
**Describe the bug**
I have only have a heat pump heater, so I selected that type during setup. It refused to turn on depict meeting the following requirements: "For a heater to be on the water temperature must be below the setpoint, the pool circuit must be on, and the heat mode must be set on the body."

So what happens if I add more heaters? The gas heater worked fine, but the heat pump still did not. Deleting the heat pump and setting the relay to trigger as a gas heater works as expected.

**Pool Equipment**
 - Controller: [e.g. Nixie Single Body (nodejs-PC v7.1)]
 - 30,000 Gallon In Ground Pool
 - Pump(s) manufacturer and model: [e.g. V-Green 165 VS Motor / Stealth Pump]
 - Filter: [Pressure Gauge]
 - Chemical controller: [e.g. REM homegrown w/Atlas]
 - Heater(s): [e.g.  Heatpump]
 - Chlorinator: [e.g. Aquarite 900]

rstrouse, 2021-06-09:
Is your heatpump connected to RS485.  If so which manufacturer and model do you have?  There may be differences protocol differences here.

sonapsent, 2021-06-09:
I have a dumb Hayward Heatpro Heatpump. It includes a pair of contacts that if open is "pool" mode and if shorted goes into "spa" mode each of which have their own temp set point. I will keep the pool set at 70 on the unit and the spa set at 90, but without a second body and valves it just gives me a min and max. If I choose a setpoint of 80 on pool controller, the relay should switch into the spa (90) mode and will turn it back to pool (70) mode when it reached the 80 setpoint measured at the thermistor. So no RS485, just a relay through REM.

tagyoureit, 2021-06-10:
In dashPanel, you can go to settings→controller→model and change the body type to a shared body.  Then you can set setpoints for pool and spa separately.

sonapsent, 2021-06-10:
Sorry to get off topic, I was just trying to explain my hack with a relay to control my heater. I only have an inground pool, no spa. Simply put if I choose a gas heater with a relay, it works as expected. If I choose a heat pump heater with a relay it refuses to turn on.

rstrouse, 2021-06-10:
Oh you are using the fireman’s switch on the heater.  There are very few heat pumps that communicate on rs485.  And those that do, often don’t provide much control.  Three wire is also dumb because that makes you have to go to the heater to change setpoints.  The way you are doing it is the best way.

tagyoureit, 2021-06-10:
If you want to be clever, you could wire up an Aux Circuit to a relay through REM and call it "high temp" and it would close the spa relay to the heater.  You would just flip this circuit to turn on the spa mode on your heater or keep it off for the pool mode.

closed 2021-06-26 by tagyoureit