tagyoureit/nodejs-poolController #319: Issues with Serialport install and RS-485 not showing up in Dash Panel

psmith3, 2021-06-09:
I have been spinning my wheels for days trying to get my install to recognize my Aquarite Chlorinator and am having issues with the Serialport portion of the install. Does this look normal? I have tried several versions of Node using NVM and am currently using Node 12.18.3. I have fewer warnings and errors on this Node version. I have also tried running this command to fix the Serialport install, but it does not help. 
`sudo npm install serialport --unsafe-perm --build-from-source`

```
pi@raspberrypi:~/nodejs-poolController $ npm i
 
 @serialport/bindings@9.0.7 install /home/pi/nodejs-poolController/node_modules/@serialport/bindings
 prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
 
 prebuild-install WARN install No prebuilt binaries found (target=12.22.1 runtime=node arch=arm libc= platform=linux)
 make: Entering directory '/home/pi/nodejs-poolController/node_modules/@serialport/bindings/build'
   CXX(target) Release/obj.target/bindings/src/serialport.o
 ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)’:
 ../src/serialport.cpp:78:69: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
    uv_queue_work(uv_default_loop(), req, EIO_Open, (uv_after_work_cb)EIO_AfterOpen);
                                                                      ^~~~~~~~~~~~~
 ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)’:
 ../src/serialport.cpp:135:71: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
    uv_queue_work(uv_default_loop(), req, EIO_Update, (uv_after_work_cb)EIO_AfterUpdate);
                                                                        ^~~~~~~~~~~~~~~
 ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Close(Nan::NAN_METHOD_ARGS_TYPE)’:
 ../src/serialport.cpp:175:70: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
    uv_queue_work(uv_default_loop(), req, EIO_Close, (uv_after_work_cb)EIO_AfterClose);
                                                                       ^~~~~~~~~~~~~~
 ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Flush(Nan::NAN_METHOD_ARGS_TYPE)’:
 ../src/serialport.cpp:215:70: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
    uv_queue_work(uv_default_loop(), req, EIO_Flush, (uv_after_work_cb)EIO_AfterFlush);
                                                                       ^~~~~~~~~~~~~~
 ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Set(Nan::NAN_METHOD_ARGS_TYPE)’:
 ../src/serialport.cpp:271:68: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
    uv_queue_work(uv_default_loop(), req, EIO_Set, (uv_after_work_cb)EIO_AfterSet);
                                                                     ^~~~~~~~~~~~
 ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Get(Nan::NAN_METHOD_ARGS_TYPE)’:
 ../src/serialport.cpp:316:68: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
    uv_queue_work(uv_default_loop(), req, EIO_Get, (uv_after_work_cb)EIO_AfterGet);
                                                                     ^~~~~~~~~~~~
 ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE GetBaudRate(Nan::NAN_METHOD_ARGS_TYPE)’:
 ../src/serialport.cpp:366:76: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
    uv_queue_work(uv_default_loop(), req, EIO_GetBaudRate, (uv_after_work_cb)EIO_AfterGetBaudRate);
                                                                             ^~~~~~~~~~~~~~~~~~~~
 ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Drain(Nan::NAN_METHOD_ARGS_TYPE)’:
 ../src/serialport.cpp:412:70: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
    uv_queue_work(uv_default_loop(), req, EIO_Drain, (uv_after_work_cb)EIO_AfterDrain);
                                                                       ^~~~~~~~~~~~~~
 ../src/serialport.cpp: At global scope:
 ../src/serialport.cpp:433:28: warning: unnecessary parentheses in declaration of ‘ToParityEnum’ [-Wparentheses]
  SerialPortParity NAN_INLINE(ToParityEnum(const v8::Local<v8::String>& v8str)) {
                             ^
 ../src/serialport.cpp:452:30: warning: unnecessary parentheses in declaration of ‘ToStopBitEnum’ [-Wparentheses]
  SerialPortStopBits NAN_INLINE(ToStopBitEnum(double stopBits)) {
                               ^
 In file included from ../../../nan/nan.h:56,
                  from ../src/./serialport.h:6,
                  from ../src/serialport.cpp:1:
 /home/pi/.cache/node-gyp/12.22.1/include/node/node.h:736:43: warning: cast between incompatible function types from ‘void (*)(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’ {aka ‘void (*)(v8::Local<v8::Object>)’} to ‘node::addon_register_func’ {aka ‘void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
        (node::addon_register_func) (regfunc),                          \
                                            ^
 /home/pi/.cache/node-gyp/12.22.1/include/node/node.h:770:3: note: in expansion of macro ‘NODE_MODULE_X’
    NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
    ^~~~~~~~~~~~~
../src/serialport.cpp:486:1: note: in expansion of macro ‘NODE_MODULE’
  NODE_MODULE(serialport, init);
  ^~~~~~~~~~~
   CXX(target) Release/obj.target/bindings/src/serialport_unix.o
 ../src/serialport_unix.cpp: In function ‘int setup(int, OpenBaton*)’:
 ../src/serialport_unix.cpp:176:60: warning: ‘%s’ directive output may be truncated writing up to 1023 bytes into a region of size 1005 [-Wformat-truncation=]
      snprintf(data->errorString, sizeof(data->errorString), "Error %s Cannot open %s", strerror(errno), data->path);
                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~
 ../src/serialport_unix.cpp:176:13: note: ‘snprintf’ output 20 or more bytes (assuming 1043) into a destination of size 1024
      snprintf(data->errorString, sizeof(data->errorString), "Error %s Cannot open %s", strerror(errno), data->path);
[... 46 lines cut ...]
[6/9/2021, 15:57:12] info: Initializing Nixie Controller
[6/9/2021, 15:57:12] info: Initializing body Pool
[6/9/2021, 15:57:12] info: Initializing Filter Filter
[6/9/2021, 15:57:12] info: Initializing Nixie circuit Pool
[6/9/2021, 15:57:12] info: Nixie Controller Initialized
[6/9/2021, 15:57:12] info: Nixie Single Body control board initialized
`

When I run Dash Panel, I get this...

```
pi@raspberrypi:~/nodejs-poolController-dashPanel $ npm start
> nodejs-poolcontroller-dashpanel@7.0.0 start /home/pi/nodejs-poolController-dashPanel
> npm run build && node dist/app.js
> nodejs-poolcontroller-dashpanel@7.0.0 build /home/pi/nodejs-poolController-dashPanel
> tsc
info: Server is now listening on 172.30.17.100:5150
```

I am able to find the server in Dash Panel
[image]

Ultimately, this is where I cannot see the RS485 Port and I cannot seem to enable it manually. Am I missing a step or is this something to do with the Serialport not installing correctly? I am using the Industrial RS485 adapter from Amazon.

[image]

rstrouse, 2021-06-10:
Post a replay here but when you tell it to do it do not Capture the configuration reload.  I don’t think we are capturing the temp from the chlorinator.  This isn’t emitted on IntelliChlor.

tagyoureit, 2021-06-10:
I've only ever seen the iChlor report their temp but the other chlorinators do not.  

We need to finish wiring up the RS485 port in the config; it isn't finished yet.

The Super Chlor settings you see in the config/chemistry pages are how long it would run but that doesn't enable it.  To enable it, click on your AquaRite on the main page and it will bring up a box with start superchlor.  Having said that, I think someone else reported an issue with it not working.  One more thing to look into but do give it a try.

psmith3, 2021-06-10:
Thanks for the clarifications. Yes, I see the Super Chlorinate button, but it does now work. Sounds like I have the rest configured properly and can ignore the RS485 drop down not populating.

closed 2021-06-10 by psmith3