tagyoureit/nodejs-poolController #320: Socat ECONNRESET errors after upgrade from 6.1.0 to 7.1.0

bobhavlin, 2021-06-11:
I recently upgraded to from v 6.1.0 (which was very solid and had few problems) to v 7.1.0 which seems to daily present the following error messages and then fail to function as expected.

> 1|njsPC      | [11/6/2021, 3:47:36 pm] debug: MQTT: Inbound easytouch2-8/state/circuits/toggleState: {"id": 6}
> 1|njsPC      | [11/6/2021, 3:47:36 pm] verbose: Wrote packet [255,0,255,165,44,16,33,134,2,6,1,1,145].  Retries remaining: 3
> 1|njsPC      | [11/6/2021, 3:47:36 pm] **error: Connection: Error: read ECONNRESET. Retry in 20 seconds**
> 1|njsPC      | [11/6/2021, 3:47:36 pm] info: Net connect (socat) closed due to error: hf-ew11:9801
> 1|njsPC      | [11/6/2021, 3:47:56 pm] info: Net connect (socat) connected to: hf-ew11:9801
> 1|njsPC      | [11/6/2021, 3:47:56 pm] info: Net connect (socat) ready and communicating: hf-ew11:9801
> 1|njsPC      | [11/6/2021, 3:48:53 pm] debug: MQTT: Inbound easytouch2-8/state/circuits/toggleState: {"id": 6}
> 1|njsPC      | [11/6/2021, 3:49:10 pm] info: Requesting easytouch configuration
> 1|njsPC      | [11/6/2021, 3:49:10 pm] verbose: **Config Queue Completed... 0% (57 remaining)**

and then it never reconnects until I restart njsPC.  Note that I am using socat to an rs485 wifi adapter (which as I said worked great until the update and nothing else changed) and njsPC is running on the same linux VM.

rstrouse, 2021-06-12:
Run npm i.  This looks like you do not have the updated serial port drivers.

rstrouse, 2021-06-13:
I too use SOCAT to connect to my remote njsPC.  I do not see the messages you are seeing with the ECONNRESET.  After some research I have determined that this has something to do with SOCAT forcibly closing the socket between the serial port and njsPC.  The error message then appears the next time there is an attempt to write to the port.  I have found that using PM2 to initialize the port there are some anomalies.  There are times when I restart the pi and SOCAT will initialize the serial port with an incorrect baud rate, parity, and flow control setting.  Perhaps the startup command for SOCAT needs to have all of the port settings included but I cannot for the world of me decipher what the difference is between ospeed, ispeed, and b9600.

bobhavlin, 2021-06-14:
Thanks for the suggestions.  I pulled a fresh copy of njsPC v7.1 and  did npm -i and the problem unfortunately persists.  Not sure how to get more details from logging, but happy to try anything that might help.

As I am using a VM, I easily switched back to my earlier state with v 6.1 and the problem goes away...  So maybe some differences in Socat between the two versions that might point in the right direction.

tagyoureit, 2021-06-14:
Have you tried to `ping hf-ew11` to see if the underlying network layer is stable?  I've had issues with socat when my router was acting up.

And did the v6.1 show them same errors and then recover or never show the same errors?

bobhavlin, 2021-06-15:
Yes, I have a ping monitor setup for the hf-ew11, given that it is wifi, I see few missed packets - < 1%.  So I think that is pretty stable.  

I tried doing debugging in both v7.1 and v6.1 by starting with `NODE_DEBUG=net npm start` and debug logging set in dashpanel.  As you can see I get the ECONNRESET message in both versions, but v.6.1 recovers.

Maybe the following helps, but I do not see anything obvious other than between the error and reconnect v6.1 does a `_final: not ended, call shutdown()` while v7.1 does `pipe false undefined`.

**in v7.1**
Rebooting rs485 device and trying to change setting via dashpanel:

> NET 411831: _read
> NET 411831: Socket._read readStart
> [15/6/2021, 8:41:48 am] info: [8:41:48 am] 192.168.2.151 PUT /state/circuit/setState {"id":6,"state":true}
> [15/6/2021, 8:41:48 am] verbose: Wrote packet [255,0,255,165,44,16,33,134,2,6,1,1,145].  Retries remaining: 3
> NET 411831: destroy
> NET 411831: close
> NET 411831: close handle
> [15/6/2021, 8:41:48 am] **error: Connection: Error: read ECONNRESET. Retry in 20 seconds**
> NET 411831: emit close
> [15/6/2021, 8:41:48 am] **info: Net connect (socat) closed due to error: hf-ew11:9801**
> NET 411831: _read
> NET 411831: _read
> NET 411831: **pipe false undefined**
> NET 411831: connect: find host hf-ew11
> NET 411831: connect: dns options { family: undefined, hints: 32 }
> NET 411831: afterConnect
> [15/6/2021, 8:42:09 am] info: Net connect (socat) connected to: hf-ew11:9801
> [15/6/2021, 8:42:09 am] **info: Net connect (socat) ready and communicating: hf-ew11:9801**
> NET 411831: _read
> NET 411831: Socket._read readStart
> NET 411831: _read
> 

Then, reload config via dashpanel:

> NET 411831: _read
> NET 411831: _read
> [15/6/2021, 8:43:52 am] info: [8:43:52 am] 192.168.2.151 PUT /app/config/reload {}
> NET 411831: onconnection
> NET 411831: _read
> NET 411831: Socket._read readStart
> NET 411831: _read
> NET 411831: _read
> Pentair EasyTouch System Detected!
> [15/6/2021, 8:43:54 am] info: Found Controller Board EasyTouch2 8
> [15/6/2021, 8:43:54 am] info: Requesting easytouch configuration
> [15/6/2021, 8:43:54 am] verbose: Config Queue Completed... 0% (57 remaining)
> NET 411831: _read
> NET 411831: _read
> NET 411831: _read
> NET 411831: _read
> NET 411831: _onTimeout
> NET 411831: destroy
> NET 411831: close
> NET 411831: close handle
> NET 411831: has server
> NET 411831: SERVER _emitCloseIfDrained
> NET 411831: SERVER handle? true   connections? 3
> NET 411831: emit close
> NET 411831: _read
> NET 411831: _read
> ...
> NET 411831: _read
> NET 411831: _read
> [15/6/2021, 8:44:06 am] **verbose: Config Queue Completed... 2% (56 remaining)**
> NET 411831: _read
> NET 411831: _read
> 

And it is hung here at 2% "Config Queue" no matter how long I wait.

**now in v 6.1:**

Rebooting rs485 device and trying to change setting via dashpanel:

> NET 42708: _read
> NET 42708: Socket._read readStart
> NET 42708: onconnection
> NET 42708: _read
> NET 42708: Socket._read readStart
> [15/6/2021, 9:05:01 am] info: [9:05:01 am] 192.168.2.151 PUT /state/circuit/setState {"id":6,"state":true}
> [15/6/2021, 9:05:01 am] verbose: Wrote packet [255,0,255,165,44,16,33,134,2,6,1,1,145].  Retries remaining: 3
> NET 42708: destroy
> NET 42708: close
> NET 42708: close handle
> [15/6/2021, 9:05:01 am] **error: Connection: Error: read ECONNRESET. Retry in 20 seconds**
> NET 42708: emit close
> [15/6/2021, 9:05:01 am] **info: Net connect (socat) closed due to error: hf-ew11:9801**
> NET 42708: **_final: not ended, call shutdown()**
> NET 42708: _read
> 
> ...
> 
> NET 42708: Socket._read readStart
> NET 42708: pipe false undefined
> NET 42708: connect: find host hf-ew11
> NET 42708: connect: dns options { family: undefined, hints: 32 }
> NET 42708: afterConnect
> [15/6/2021, 9:05:12 am] info: Net connect (socat) connected to: hf-ew11:9801
> [15/6/2021, 9:05:12 am] **info: Net connect (socat) ready and communicating: hf-ew11:9801**
> NET 42708: _read
> NET 42708: Socket._read readStart
> [15/6/2021, 9:05:12 am] verbose: Wrote packet [255,0,255,165,44,16,33,134,2,6,1,1,145].  Retries remaining: 2
> NET 42708: _read
> NET 42708: _read
> NET 42708: _onTimeout
> NET 42708: destroy
> NET 42708: close
> 

Now, reload dashpanel:

> NET 42708: Socket._read readStart
> NET 42708: _read
> [15/6/2021, 9:16:19 am] verbose: Config Queue Completed... 98% (1 remaining)
> [15/6/2021, 9:16:19 am] verbose: Wrote packet [255,0,255,165,44,16,33,216,1,2,1,221].  Retries remaining: 3
> NET 42708: _read
> [15/6/2021, 9:16:19 am] **verbose: Config Queue Completed... 100% (0 remaining)**
> [15/6/2021, 9:16:19 am] **info: EasyTouch system config complete.**
> 

**And v6.1 is working again!**

bobhavlin, 2021-06-15:
Apologies, I just realized that I was on "next" branch with my "v6.1" ...
Not sure exactly how that changes things, but I have made a diff among the two Comms.ts here (First is v6.1(next) Second is v7.1(master)):
https://gist.github.com/9636b85605dfce2250c7660f3f7d5452

Rethinking about the above error makes me think that the order of `_final: not ended, call shutdown()` is after reconnecting in v7.1 while it is before in v6.1.  I'm guessing that is source of the problem.

tagyoureit, 2021-06-16:
Can you pull this commit?  (https://github.com/tagyoureit/nodejs-poolController/commit/5df40ac7b908d69921c9c7a37219b801b0d419b1) I was working on the comms anyway and now you can help out.  Forgot to tag it with this issue when I committed it.

1. `git checkout ResponseRefactor`
2. `git pull`
3. Restart app.

Wait the allotted 20s that the port says it will try to open and see what happens.  If you can do a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution#-dashpanel) if it doesn't work that would be helpful.

bobhavlin, 2021-06-16:
Thanks for the reply!

I pulled that commit, restarted the app, verified it worked, then rebooted the rs485 and got the following, which looks like it will eventually fully reload successfully, but is very slow (I'll let it run overnight and check tomorrow):

> 1|njsPC      | [15/6/2021, 11:26:32 pm] info: EasyTouch system config complete.
> 1|njsPC      | [15/6/2021, 11:26:32 pm] info: Initializing Nixie Controller
> 1|njsPC      | [15/6/2021, 11:26:32 pm] info: Nixie Controller Initialized
> 1|njsPC      | [15/6/2021, 11:26:47 pm] info: [11:26:47 pm] 192.168.2.151 PUT /state/circuit/setState {"id":6,"state":true}
> 1|njsPC      | [15/6/2021, 11:26:47 pm] verbose: Wrote packet[255,0,255,165,44,16,33,134,2,6,1,1,145].Retries remaining: 3 
> 1|njsPC      | [15/6/2021, 11:28:11 pm] info: [11:28:11 pm] 192.168.2.151 PUT /state/circuit/setState {"id":6,"state":false}
> 1|njsPC      | [15/6/2021, 11:28:11 pm] verbose: Wrote packet[255,0,255,165,44,16,33,134,2,6,0,1,144].Retries remaining: 3 
> 1|njsPC      | [15/6/2021, 11:29:24 pm] info: [11:29:24 pm] 192.168.2.151 PUT /state/circuit/setState {"id":6,"state":true}
> 1|njsPC      | [15/6/2021, 11:29:24 pm] verbose: Wrote packet[255,0,255,165,44,16,33,134,2,6,1,1,145].Retries remaining: 3 
> 1|njsPC      | [15/6/2021, 11:29:24 pm] error: Net connect (socat) Connection: Error: read ECONNRESET. Retry in 20 seconds
> 1|njsPC      | resetting connection timer
> 1|njsPC      | [15/6/2021, 11:29:24 pm] info: Net connect (socat) closed due to error: hf-ew11:9801
> 1|njsPC      | [15/6/2021, 11:29:44 pm] info: Net connect (socat) connected to: hf-ew11:9801
> 1|njsPC      | [15/6/2021, 11:29:44 pm] info: Net connect (socat) ready and communicating: hf-ew11:9801
> 1|njsPC      | [15/6/2021, 11:30:03 pm] info: [11:30:03 pm] 192.168.2.151 PUT /state/circuit/setState {"id":6,"state":true}
> 0|dashPanel  | 192.168.2.151 GET /options?null
> 1|njsPC      | [15/6/2021, 11:30:25 pm] info: [11:30:25 pm] 192.168.2.151 GET /app/config/log {}
> 1|njsPC      | [15/6/2021, 11:30:25 pm] info: [11:30:25 pm] 192.168.2.151 GET /config/all?null {}
> 1|njsPC      | [15/6/2021, 11:30:38 pm] info: [11:30:38 pm] 192.168.2.151 PUT /app/config/reload {}
> 1|njsPC      | Pentair EasyTouch System Detected!
> 1|njsPC      | [15/6/2021, 11:30:38 pm] info: Found Controller Board EasyTouch2 8
> 1|njsPC      | [15/6/2021, 11:30:38 pm] info: Requesting easytouch configuration
> 1|njsPC      | [15/6/2021, 11:30:38 pm] verbose: Config Queue Completed... 0% (57 remaining)
> 1|njsPC      | [15/6/2021, 11:31:17 pm] info: [11:31:17 pm] 192.168.2.151 PUT /app/logger/setOptions {"packet":{"enabled":true}}
> 1|njsPC      | [15/6/2021, 11:31:24 pm] info: [11:31:24 pm] 192.168.2.151 PUT /state/circuit/setState {"id":6,"state":true}
> 1|njsPC      | [15/6/2021, 11:31:36 pm] info: [11:31:36 pm] 192.168.2.151 PUT /app/logger/setOptions {"packet":{"logToConsole":true}}
> 1|njsPC      | [15/6/2021, 11:31:37 pm] info: {"id":590,"valid":true,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255],[255,0,255],[165,44,15,16,2,29],[23,31,0,0,0,0,0,0,0,0,0,0,32,4,82,82,0,0,74,0,0,0,1,0,0,112,137,0,13],[3,94]],"ts": "2021-06-15T23:31:37.717-0500"}
> 1|njsPC      | [15/6/2021, 11:31:39 pm] info: {"id":592,"valid":true,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255],[255,0,255],[165,44,15,16,2,29],[23,31,0,0,0,0,0,0,0,0,0,0,32,4,82,82,0,0,74,0,0,0,1,0,0,112,137,0,13],[3,94]],"ts": "2021-06-15T23:31:39.735-0500"}
> 1|njsPC      | [15/6/2021, 11:31:40 pm] info: {"id":593,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,96,16,7,0],[],[1,28]],"ts": "2021-06-15T23:31:40.049-0500"}
> 1|njsPC      | [15/6/2021, 11:31:40 pm] info: {"id":594,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,16,96,7,15],[4,0,0,0,0,0,0,0,0,0,0,0,0,23,31],[1,101]],"ts": "2021-06-15T23:31:40.050-0500"}
> 1|njsPC      | [15/6/2021, 11:31:41 pm] info: {"id":596,"valid":true,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255],[255,0,255],[165,44,15,16,2,29],[23,31,0,0,0,0,0,0,0,0,0,0,32,4,82,82,0,0,74,0,0,0,1,0,0,112,137,0,13],[3,94]],"ts": "2021-06-15T23:31:41.747-0500"}
> 1|njsPC      | [15/6/2021, 11:31:43 pm] info: {"id":598,"valid":true,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255],[255,0,255],[165,44,15,16,2,29],[23,31,0,0,0,0,0,0,0,0,0,0,32,4,82,82,0,0,74,0,0,0,1,0,0,112,137,0,13],[3,94]],"ts": "2021-06-15T23:31:43.766-0500"}
> 1|njsPC      | [15/6/2021, 11:31:46 pm] info: {"id":601,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,96,16,4,1],[255],[2,25]],"ts": "2021-06-15T23:31:46.138-0500"}
> 1|njsPC      | [15/6/2021, 11:31:46 pm] info: {"id":602,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,16,96,4,1],[255],[2,25]],"ts": "2021-06-15T23:31:46.140-0500"}
> 1|njsPC      | [15/6/2021, 11:31:46 pm] info: {"id":603,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,96,16,6,1],[4],[1,32]],"ts": "2021-06-15T23:31:46.142-0500"}
> 1|njsPC      | [15/6/2021, 11:31:46 pm] info: {"id":604,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,16,96,6,1],[4],[1,32]],"ts": "2021-06-15T23:31:46.142-0500"}
> 1|njsPC      | [15/6/2021, 11:31:47 pm] info: {"id":606,"valid":true,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255],[255,0,255],[165,44,15,16,2,29],[23,31,0,0,0,0,0,0,0,0,0,0,32,4,82,82,0,0,74,0,0,0,1,0,0,112,137,0,13],[3,94]],"ts": "2021-06-15T23:31:47.799-0 [... cut ...]
> 1|njsPC      | [15/6/2021, 11:40:33 pm] info: {"id":1411,"valid":true,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255],[255,0,255],[165,44,15,16,2,29],[23,40,0,0,0,0,0,0,0,0,0,0,32,4,81,81,0,0,74,0,0,0,1,0,0,112,137,0,13],[3,101]],"ts": "2021-06-15T23:40:33.459-0500"}
> 1|njsPC      | [15/6/2021, 11:40:33 pm] info: {"id":1412,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,96,16,7,0],[],[1,28]],"ts": "2021-06-15T23:40:33.791-0500"}
> 1|njsPC      | [15/6/2021, 11:40:33 pm] info: {"id":1413,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,16,96,7,15],[4,0,0,0,0,0,0,0,0,0,0,0,0,23,40],[1,110]],"ts": "2021-06-15T23:40:33.792-0500"}
>

tagyoureit, 2021-06-16:
Interesting that there was only 1 disconnect.  But that's good.  This whole thing wreaks of network communication errors because it shouldn't be that slow.  Typical communications are measured in 10s of ms and you are at ~2000ms per packet.  I get <100ms on Socat and it's even quicker.

You might also want to lower the `inactivityRetry` in config.json.  This is what is the source of the 20s delay.  It won't make a huge difference overall but when it does disconnect this will make it try again sooner.  

I'll close this for now but let us know if you figure anything else out.

tagyoureit, 2021-06-24:
I merged these changes back to the `master` branch.  Please switch back to that and you'll be in sync with the latest.