tagyoureit/nodejs-poolController #322: New Request: Pentair SolarTouch Controller support

enjos318, 2021-06-13:
**Describe the bug**
This is more of a new request then a bug...
The nodejs-poolController does not recognize the SolarTouch controller, and does not discover my IntelliFlo VS or IntelliChlor.

**To Reproduce**
Steps to reproduce the behavior:
1. Start nodejs-poolController
2. SolarTouch,IntelliFlo, and IntelliChlor are not discovered with 7.1.0 but briefly recognized with 6.5.1.

**Expected behavior**
I checked the RS485 messages on USB0 and saw the repeating ff00 00a5  and expected to see the pump or chlorinator on the dashpanel app. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Packet Capture**
[replay (5).zip](https://github.com/tagyoureit/nodejs-poolController/files/6643122/replay.5.zip)

**Pool Equipment**
-Solar Touch, firmware 2.0
-IntelliFLO VS
-IntelliChlor IC40
-Raspberry Pi 4
-Industrial USB to RS485 converter

node - v12.22.1
npm - 7.17.0

**Additional context**

The SolarTouch panel that I am connected to does communicate with the pump just fine. I tried nodejs-poolController 6.5.1 and it said it found both the chlorinator and the pump but would not show up in the dash panel.
[image]

I initially posted this issue on troublefreepool.
https://www.troublefreepool.com/threads/nodejs-poolcontroller-doesnt-communicate-with-pentair-pool-equipment.233722/#post-2050405

tagyoureit, 2021-06-13:
Unfortunately, the Solar Touch (and IntelliConnect) don't talk RS485 to other controllers.  They are essentially black boxes.  

With 7.0 we removed the autodiscovery which is why we are not searching for your pump/chlorinator.  However, you need to either use Solar Touch or njsPC since they would conflict with being the master and may give your equipment competing sets of directions.

closed 2021-06-13 by tagyoureit

adymanis, 2021-06-15:
I also have a SolarTouch and would like to just pull temperature and status data over RS485. When plugged all I am seeing on the wire is A5 00 60 10 04 01 ff 02 19 which seems to indicate the SolarTouch trying to control the pump. I do not have any other accessories or devices other than the SolarTouch. I attempted to respond a confirmation to the message to see if SolarTouch would begin sending any status but it gave me a "Lost Connection to Pump".  Is there a way to get this data directly off of SolarTouch over RS485?

tagyoureit, 2021-06-15:
That is indeed a pump message.  The SolarTouch does talk rs485 to the pump to control it.  But Solartouch won't send this information out in any way except for the LCD display on the screen.

They best you could hope for is to listen to the pump messages and you could know if it is running, get the rpm (or gpm), watts, etc.  Those would be in the 7 status packet.  You wouldn't be able to control the pump without conflicting with the SolarTouch.

There is no way to retrieve the temperatures from SolarTouch.