tagyoureit/nodejs-poolController #323: Super Chlorinate not working

Tegguy, 2021-06-13:
The super chlorinate button is not working. It will attempt to enable superchlorinate mode but doesn't fully do so. The pentair app has the super chlorinate turned on but the ichlor doesn't light up as if it is and the NJSPC doesn't show the button as active. Here is a log from trying to use the NJSPC and also the intellicenter (hope I did it right)
[Nodejs Super Chlor.log](https://github.com/tagyoureit/nodejs-poolController/files/6644406/Nodejs.Super.Chlor.log)
[Super Chlor Intellicenter.log](https://github.com/tagyoureit/nodejs-poolController/files/6644407/Super.Chlor.Intellicenter.log)

rstrouse, 2021-06-14:
You appear to have a syntax error in your config.json file.  Did you make edits?  Copy the contents of the file to this online verifier to find the error.  You can just copy and paste it then click the validate button and it will point out the error.

https://jsonlint.com/

Tegguy, 2021-06-14:
I was attempting to figure out how to setup the MQTT integration to work with google home and stuff but thought I set everything back. I ran my config.json through the validator and it came back valid. I rebooted the raspberrypi and the superchlorinate seems to work now however, I have it set for 24 hours but the counter starts at around 5:45. I think there might be another issue open for this so please feel free to close my issue if it's a duplicate.

tagyoureit, 2021-06-15:
If you want to tag onto the other issue, please upload a set of logs.  I haven't been able replicate the issue again.
https://github.com/tagyoureit/nodejs-poolController/issues/312

closed 2021-06-15 by tagyoureit