tagyoureit/nodejs-poolController #324: UltraTemp Heat Pump error "SERIAL COMM LOST"

ibehren1, 2021-06-13:
**Describe the bug**
Connected Pentair Ultratemp Heat Pump via RS-485.  When nsjPC sends command (i.e. I try to set the temp) the heat pump flips on the service light with an error "SERIAL COMM LOST" on the screen.  Not seeing any packets to or from the heat pump in the message center.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect heat pump via RS-485 and set INTELL ADDRESS to known value in heat pump menu.
2. Add heat pump as a heater in dashPanel.
3. Turn heater mode to Ultra Temp and set a desired temp.
4. See error and no packets to heater in message center.

**Expected behavior**
Expect dashPanel to see the water temp from the heat pump and set the desired temp without the heat pump throwing COMM error.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]
[image]

**Packet Capture**
No packets captured
[heat_pump_replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/6644468/heat_pump_replay.zip)

**Pool Equipment**
 - Controller: njsPC
 - Controller Model: Nixie Single Body
 - Controller Firmware Version: n/a
 - Pump(s) manufacturer and model: IntelliFlo VSF (working)
 - Chlorinator: n/a
 - Heater(s): Pentair Ultratemp
 - Chemical controller:  IntelliChem (not working https://github.com/tagyoureit/nodejs-poolController/issues/315)
 - Valves: n/a
 - Any other relevant equipment: n/a

**Desktop (please complete the following information):**
 - OS: MacOS
 - Browser: Brave 
 - Version: Version 1.25.70 Chromium: 91.0.4472.77 (Official Build) (x86_64)

**Additional context**
Add any other context about the problem here.

More than happy to help do capture or other work to help get this supported.  Same with the IntelliChem on https://github.com/tagyoureit/nodejs-poolController/issues/315.

rstrouse, 2021-06-14:
We have seen our first ultratemp a couple of weeks ago and have just now started working with the protocol.  We have figured out how to turn on the heatpump to heat and cooling but not much beyond that.  In the next couple of weeks we will push this code so that it should engage the heater.  As for the IntelliChem we are also working through that protocol as our original guidance on that equipment was not correct.  The IntelliChem code will likely come first though as we have pretty much rooted all that out.

ibehren1, 2021-06-14:
Thanks again for the update and if there is some way I can help, please let me know.

rstrouse, 2021-07-03:
If you pull njsPC the support for UltraTemp has been enabled.  Let me know how it goes but if you turn on Heater messages on the log it should be showing the 114 and 115 message communication between Nixie and the heater.

ibehren1, 2021-07-03:
See similar issue to the IntelliChem with all outbound packets no inbound.  Tried swapping the wires, no change.  Still flips on the Service light with the comms message.  Think this needs a set source as well?

[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/6758956/replay.zip)

I notice the source for these are OCP instead of plugin.  What is the difference there?
[image]

ibehren1, 2021-07-03:
Just saw your mention on the other thread of water and air temp sensors.  I don't have those.  The UntraTemp does sense water temp internally to cycle on and off.

Is there a recommended air temp sensor?

rstrouse, 2021-07-03:
For you to get any automation out of ultra temp it will need a water temp sensor and an air sensor.   The least expensive way to go is to use an ads1x and REM.  Sensors are typically 10k sensors and you can even buy the Pentair ones that only require drilling a small hole in the pipe after the pump.  You have a ton of options.  If you implement in REM you can also hook up some relays for cheap as well.

rstrouse, 2021-07-04:
I looked into why you were not getting responses from the heater and found an issue with the checksum calculation for outbound heater messages.  If you do a pull the heater should now at least respond with an action 115 for all 114s sent to it.

I guess I glossed over the UltraTemp water temp sensor.  It really doesn't have utility outside of the heater box.  If it did Pentair would be using it as the means to display and operate.  Here is a list of 10k thermistors that will get the job done for both air and water.
https://www.amazon.com/s?k=pentair+temperature+sensor&gclid=EAIaIQobChMI4tPbqObJ8QIVrRmtBh2B8QiYEAMYASAAEgKIRvD_BwE&hvadid=176956807337&hvdev=c&hvlocphy=9032516&hvnetw=g&hvqmt=e&hvrand=16750041925627105145&hvtargid=kwd-14130263861&hydadcr=13743_9595221&tag=googhydr-20&ref=pd_sl_7hv1ejejwg_e

These are then connected via an analog to digital converter (ADC) using REM.  Depending on your tinker level you can choose to communicate with it via a scratch build ADS1x15 or MCP spi chip on the bust out a soldering iron side.  Or you can choose one of the myriad of inexpensive hats that support the ADS1x15 chips.   We have even built support for the Sequent MEGA-IND card that has 4 0-10v inputs suitable for this purpose as well as a myriad of other I/O.

ibehren1, 2021-07-04:
Thanks for another update.  I can confirm that does enable the 2-way communication.  Shame that Pentair does not deliver the water temp back.  I will have to look at the details of adding the sensors but I may just leave that to the UltraTemp since is can fully automate itself heat/cool to maintain the temp, though we really only run it early and late in the season.

I think we can probably close out both #315 and #324. Thanks again for all the work and let me know if I can ever help out with info gathering or anything.

rstrouse, 2021-07-04:
I think that you will find the ability to see the pool temp and not visit the pad to change it is a major part of the automation.  At that point it is like you spent a bunch of money on an automation controller.  If you have lights you can hook them up as circuits and even control other things related to the pool.

The heater interfaces are very rudimentary for all of the manufacturers out there.  All of them are just poor.

ibehren1, 2021-07-04:
For the sensors, it would take 2 right?  1 for air and 1 for water (the sensor does not have 2 sensors in 1 physical piece, right)?

Do you use the water temp in the LSI calc?  or is that just coming from what the Intellichem is reporting?  Also, does the code push values to the IntelliChem for the ALK and CH or just reading what is set there?  Seems like if I try to change ALK in the interface it is not holding the value.  CH does seem to hold it.  (May also be a rounding thing... i.e. was adjusting ALK to 34 today but keeps going back to 30.)

I do also have Pentair LED lights, but I dont think I have any way to control them at the pad.  Power is run from there to the control switch/color selector 50ft away which to my knowledge, all the "smarts" for the lights are in that switch/color selector.  https://smile.amazon.com/Pentair-600054-IntelliBrite-Waterproof-Controller/dp/B008H9EI9U/ref=sr_1_3?dchild=1&keywords=pentair+color+light+switch&qid=1625423954&sr=8-3

rstrouse, 2021-07-04:
The LSI calc is actually done inside the IntelliChem.  The temperature that it uses is reported on byte(31) of the 18 message so theorhetically, one could steal that temp from there.  

I'll have a look at the alkalinity setting as there shouldn't be any rounding issues with that and it does send all the values with the exception of Borates to the chem controller.

If you replace the controller with a $20 relay.  njsPC and REM will allow you to change the colors and control as many lights as you have relays individually.  The only thing that controller is doing is turning the circuit on and off to change the controller on the fixture.
[image]

rstrouse, 2021-07-04:
Let's see what the data from the IntelliChem temperature probe is like for you.  Pull njsPC.  Before you start it back up again go into the data/poolConfig.json file and add or change the attribute for feedBodyTemp under the ph.probe section so the ph.probe should lool like below.  The key is to add `"feedBodyTemp": true` as an attribute of the ph probe.
```
...
"ph": {
  "chemType": "ph",
   ...  Other attributes
   "probe": { "feedBodyTemp": true }
   ... Other attributes
}
```

ibehren1, 2021-07-04:
make sure I am getting this right...

existing
```
      "ph": {
        "chemType": "ph",
        "setpoint": 7.6,
        "phSupply": 1,
        "probe": {
          "enabled": true,
          "master": 1
        },
        "acidType": 0,
```

updated
```
      "ph": {
        "chemType": "ph",
        "setpoint": 7.6,
        "phSupply": 1,
        "probe": {
          "feedBodyTemp": true
          "enabled": true,
          "master": 1
        },
        "acidType": 0,
```

ibehren1, 2021-07-04:
It did grab the temp...
[image]

Kinda think this is hard-coded/manual entry from the LSI menu in the IntelliChem though and not actually from a sensor.  I was able to update the temp that way from 87 to 88.

rstrouse, 2021-07-05:
Wow I am in shocked that it does not compensate for temperature.  So much for getting the temperature from there.  Heck the message that we send to change the settings on IntelliChem contains no ability to set the temperature either.

rstrouse, 2021-07-05:
Ok so temperature is a major factor in all of this and it really was bugging me that they could/would get away without temperature compensation.   So I did a little reading.  Let me just say boo to this as I would have expected temperature at the flow cell but it appears it is expecting a temperature reading from an EasyTouch, IntelliTouch, or IntelliChem controller. I am pretty sure I know where it gets it from and if you do set up a temp sensor for Nixie we can feed IntelliChem with it.

[image]

ibehren1, 2021-07-05:
Good choices?

https://smile.amazon.com/gp/product/B07Q8RHK7D/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

https://smile.amazon.com/gp/product/B07MH2Y8NW/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

rstrouse, 2021-07-06:
Those should do the trick.  It amazes me that they sell that unit standalone without a temp feed.  pH probes are temperature sensitive.  ORP notsomuch but the potential increases with temperature.

rstrouse, 2021-07-06:
I am going to close this for now and when you get your temp sensors hooked up or need any help open another issue.  Thanks for the help getting at least some of the UltraTemp protocol figured out.

closed 2021-07-06 by rstrouse

ibehren1, 2021-07-08:
Got the equipment...

OS sees the ADS hat
```
user@host:~$ lsmod | grep ads
ti_ads1015             24576  0
industrialio_triggered_buffer    16384  1 ti_ads1015
industrialio           98304  3 industrialio_triggered_buffer,kfifo_buf,ti_ads1015
```

I have installed REM and have it running.

Also, I have connected a temp sensor to the 3V3 and AIN3 contacts.

I am not clear on how to click through all the setup items.  Is there a write somewhere that I can reference?

Thanks

rstrouse, 2021-07-08:
Open another issue as I think walking you through this one might be helpful to others.  In the mean time get 1 10k resistor for each temp sensor.

ibehren1, 2021-07-08:
Sure thing.  Are there other items that I am going to need.  Probably a breadboard to wire stuff onto, I supporse.

rstrouse, 2021-07-08:
Nothing that dramatic.  We will need a voltage divider circuit so that we are measuring the resistance value from the Thermistor.  Aside from the parts you already have this simply involves a 10k resistor.

rstrouse, 2021-07-09:
I also decided to create a writeup on temperature sensors which may help understand it all.
https://github.com/rstrouse/relayEquipmentManager/wiki/Temperature-Sensors

ibehren1, 2021-07-10:
Excellent write up!!!  Just waiting on my resistors now...