tagyoureit/nodejs-poolController #325: Easytouch + dashpanel : Cannot turn off "pool"

emlowe, 2021-06-14:
Having trouble turning off the pool. I can turn on/off the spa, and turn on/off the spillway, and can switch between spa, pool, spill way, but cannot turn off pool

In Dashpanel, the light alternates between yellow, off and then back to green rapidly

[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/6645285/replay.zip)

rstrouse, 2021-06-14:
The EasyTouch controller is what is turning this back on.  This is true even after it has confirmed the request to turn off the pool circuit.  Here is what is going on.

The first line in the display is njsPC receiving your request to turn circuit 6 off.  The next line where the action is 134 is njsPC asking EasyTouch to turn off the pool circuit.  To which it responds with a 1 indicating that it will do the 134 command as requested.  The next message however is a 2 message which indicates the current status of the circuits among other things.  The first two bytes 19, 34 represent the current time on the panel but the next byte [32] is the list of current circuits.  This is telling us that regardless of the request to turn it off the pool circuit remains on.  This is because the 6th bit is 1 on that byte.
[image]

[image]

So now down to what could possibly be making the EasyTouch do this.  Well the next message right under that is a 210 message.  This is a request from EasyTouch attempting to contact IntelliChem.  In your log file EasyTouch has been trying to get in contact with it but I don't see any responses.  This could be because you have those filtered out but that is not what your log indicates.

Do you have an IntelliChem?  If you do not then it is partially configured in the EasyTouch control panel?  Maybe this behavior is because EasyTouch cannot determine whether IntelliChem is dosing or not.  

There are some other well known conditions where Pentair controllers will force on a circuit.  First, Freeze Protection comes to mind but that is probably not the case given your water temp is 88.  Next there is Super Chlorination which is probably not the case given the there is no chlorinator communication going on.  Finally there is Heater Cooldown but I do not see a delay triggered for the heater.  This leaves the mysterious communication with IntelliChem.  If you do not have one disable it on the the EasyTouch control panel.

emlowe, 2021-06-14:
I do not have IntelliChem - that interesting. Thanks for the diagnosis, I'll see what I can do on the panel

rstrouse, 2021-06-16:
Did you get this issue resolved?

emlowe, 2021-06-16:
The intellichem did seem to be enabled on the panel so I turned it off (not even sure why it appears if you don't have one). I haven't had a chance to test it out though

rstrouse, 2021-06-16:
It is not normally enabled and I have not seen another case where the IntelliChem function was mysteriously enabled.  Perhaps you were just fiddling and accidentally enabled it.  njsPC won't pick up that there is an IntelliChem until the OCP says it is actually there.

emlowe, 2021-06-17:
Seems to have fixed it - however, when I turn off the pool, it goes yellow, blank, green again - and then takes some time (dunno how long - maybe it's waiting for some cool-down delay) - but it does turn off.

tagyoureit, 2021-06-18:
Feel free to submit another packet capture.  Doesn't sound 100% fixed yet...

closed 2021-08-05 by tagyoureit