tagyoureit/nodejs-poolController #326: Can only start chlorinator on one circuit

colin-young, 2021-06-15:
**Pool Equipment**
 - Controller: Nixie single body
 - Controller Model: 
 - Controller Firmware Version: 
 - Pump(s) manufacturer and model: Pentair
 - Chlorinator: AquaRite
 - Heater(s): none
 - Chemical controller: none
 - Valves: none
 - Any other relevant equipment:

I have 2 circuits set up: 1 for regular operation (low RPM), the other for cleaning (high RPM). On the first I am able to set the output level of my chlorinator. On the second I cannot. It stays off no matter what I set for the pool setpoint.

This is the relevant configuration, extracted from `/state/all/`. I see where the chlorinator is associated to the body, but I do not see the circuits associated to a body.

What am I missing?

```
"chlorinators": [
    {
      "id": 1,
      "disabled": false,
      "poolSetpoint": 30,
      "spaSetpoint": 0,
      "superChlorHours": 8,
      "type": {
        "val": 0,
        "name": "pentair",
        "desc": "Pentair"
      },
      "name": "AquaRite",
      "isActive": true,
      "status": {
        "val": 0,
        "name": "ok",
        "desc": "Ok"
      },
      "targetOutput": 0,
      "currentOutput": 0,
      "lastComm": 1623716586386,
      "body": {
        "val": 0,
        "name": "pool",
        "desc": "Pool"
      },
      "saltLevel": 3400,
      "saltRequired": 0,
      "superChlorRemaining": 0,
      "superChlor": false,
      "equipmentType": "chlorinator"
    }
  ],
  "circuits": [
    {
      "id": 6,
      "showInFeatures": true,
      "type": {
        "val": 12,
        "name": "pool",
        "desc": "Pool",
        "hasHeatSource": true
      },
      "name": "Default Pool",
      "isOn": false,
      "equipmentType": "circuit"
    },
    {
      "id": 2,
      "isOn": true,
      "name": "Cleaner",
      "type": {
        "val": 12,
        "name": "pool",
        "desc": "Pool",
        "hasHeatSource": true
      },
      "showInFeatures": true,
      "equipmentType": "circuit"
    }
  ],
```

rstrouse, 2021-06-15:
Chlorinators are related to the specific body.  In this case you only have one but the relationship is important since the body controls the valve positions related to the pool circuit.  

However, I don't see the definition of your pump.  You should have this set up so that the cleaner circuit and the pool circuit are on at the same time.  This will select the highest speed for your pump.  If you have only the cleaner on and the pool off the Chlorinator will not come on by design.  But it will operate the pump that is specified for the cleaner at the RPM of the cleaner.

Think about it like this.  You have a pool schedule that runs for 8 hours but during 3 hours of that you can run the cleaner which may turn on a booster pump or even increase the RPMs on your filter pump to service both items.  When the cleaner schedule kicks on it does whatever it needs to do to support the cleaner but when that schedule expires it continues with the pool schedule.

colin-young, 2021-06-15:
Thanks, that's kind of what I was thinking. As noted I only have 1 body defined, and in the circuit definitions both are type 12. This behavior is not exactly intuitive, at least not from an API perspective. I can't remember exactly what time I was testing things, but I do know it extended beyond the time my default pool circuit was set to turn off (that's when I captured the configuration), so it was probably a combination of bad logic originally and then the default circuit not being on later. I'll verify tomorrow.

I gather from your comments that the effective pump speed is the highest speed defined for all the circuits that are turned on? And is the chlorinator guaranteed to be off if the pump is not running? I only ask because I had a generator cell burn out on me because the flow sensor got jammed and left the chlorinator running while the pump was off, and that is really what I am trying to avoid. Maybe I'm overcomplicating things and can just turn the main circuit on and off?

For the record, my pump definition is:

```
[
  {
    "id": 50,
    "master": 1,
    "type": 4,
    "circuits": [
      {
        "circuit": 6,
        "speed": 1500,
        "units": 0,
        "master": 1
      },
      {
        "circuit": 2,
        "speed": 2800,
        "units": 0,
        "master": 1
      }
    ],
    "isActive": true,
    "name": "Pump 1",
    "address": 96,
    "minSpeed": 450,
    "maxSpeed": 3450,
    "primingSpeed": 0,
    "minFlow": 15,
    "maxFlow": 130,
    "primingTime": 0,
    "model": null,
    "relays": []
  }
]
```

rstrouse, 2021-06-15:
Yes the body setting ensures the chlorinator is tuned off if the pool is not on.  And the speed setting is the max speed for all circuits assigned to the pump that are on.  On another note the schedules in njsPC are now operational.

rstrouse, 2021-06-15:
One other thing is that if you are using REM you can hook the Chlorinator up to the filter relay and it will actually kill power to the Chlorinator if the latch does not get a watchdog call.

tagyoureit, 2021-06-15:
Another thing is that we are somewhat limited by the original structures of the Pentair OCP's.  We are slowly working through extending/complementing/fixing that where we can but there are some constructs where we just need to learn to live with the bad decisions they made.

Pentair OCP's always use circuits 6 and 1 as the first two bodies.  On EasyTouch/Intellitouch a single body model uses circuit 6 as the pool; shared/dual models have circuit 1 hardcoded as the 2nd body.  Intellicenter is a little more forgiving in that it allows you to set circuits 1 and 6 as either pool or spa.  There was some conjecture that this probably related to early hardware where the pins on some IC's were physically grouped together and it was easy for the engineers to hardcode these two circuits as the bodies, but who knows.

All this to say that you can only have a single circuit defined as the pool body and it must be circuit 6.  Setting the function of circuit 2 to pool doesn't do what you think.  This might be an area that we can extend/modify the existing paradigm (or at least put in some error checking) but with the OCP's you can set a different circuit to either a pool or spa function.  What does it do?  Dunno.   

Setting circuit 6 on/off turns on the pool body.  Turning on circuit 2 (even with the pool function) while circuit 6 is off will only set the pump to run at that specified rpm.  That's why you see the chlorinator work with your pool but not you cleaner.  

The right setup is to have your low speed on circuit 6 and then _also_ have your circuit 2 running when you want the cleaner to kick in.  This is exactly what @rstrouse was suggesting above but hopefully the additional details help explain what's going on.

colin-young, 2021-06-15:
Thanks, that will allow me to simplify some logic. I did see a note about the schedules being enabled in the nixie controller, but, as I've noted elsewhere, I've got my own scheduling code running and I'm really just using this project as an interface to the pump and chlorinator. Not that there is anything wrong with it, I just have different goals than what this is primarily built to support -- i.e. most people just want a trouble-free run it and forget it system and I want to, well, tinker. I'm still thinking about adding a current sensor on the pump to kill power to the chlorinator if there is no current. I may be a little paranoid about the cell getting burnt out again, but I really don't want to give Hayward any more money.

@tagyoureit yes, I understood what @rstrouse was getting at, and that is indeed how it runs, at least when my cleaner timer is running at the same time as my sunrise/sunset schedule (offset 2 hours inside). Unfortunately the time I have to debug happens to fall outside of the time when the pool circuit is running by default. The background and speculation is appreciated and will make it easier as I add new features to my setup. While I would not have designed things the way Pentair did, in their defense I'm not building hardware that any dummy can bolt to the side of their house and run trouble-free for years, and I'm not trying to lock people in to my proprietary solutions so that they are forced to buy over-priced hardware to do simple things like change the colors of a light. I really appreciate the time that has been put into this project so that I can continue to not have to learn the gory details of writing code for RS-485 communications.

rstrouse, 2021-06-15:
Oh one final point on the body management.  If you had more than one Pool/Spa, Pool/Pool. or Spa/Spa the primary function of these bodies are to maintain that shared equipment.  While water flow Suction/Return is typically controlled by automatic valve rotation, it also comes into play with heaters, pumps, chlorinators, and chemical controllers.   These bits of equipment typically operate differently depending on the selected body in a shared body system.

The relays on REM do a pretty good job at disengaging should a failure occur.  While you can never hit all the failure modes, the biggies are handled.   Think of REM as a hardware interface that can get you out of the business (gory details) related to I2c, SPI, GPIO, and ADC transformations.  This too can be accessed by your HA.

colin-young, 2021-06-15:
I'm in no danger of needing to deal with multiple-body systems any time soon (a decision my wife still regrets from when we were building the pool, but who needs a hot tub in Florida?)

I've looked at REM and decided that since I had already decided to go my own route it was just as easy for me to handle interfacing with sensors and relays myself through Circuit Python (to handle the low-level comm protocols, a.k.a. gory details). Everything is published as a message on an MQTT channel and Node Red reacts to those of interest, in addition to being able to log data into Mongo. It's an architecture I'm comfortable with and have been working with for years (albeit on a much, much larger scale). Also, I have a degree in mechanical engineering (work in software now), and sometimes miss the days of building industrial water treatment systems, so this gives me a chance to play with that stuff again.

Once I "finish" the initial build, I'll probably take REM for a spin just to see what it offers.

rstrouse, 2021-06-15:
I don't know all of the equipment you have but it does things like sensor reading smoothing, MQTT, Sockets, Http RPC, and rest.  The other thing it does is expose the configuration interface for quite a few I2c, ADC, and IO boards and sensors.  Here are a couple of screens shots.
[image]
[image]
[image]

colin-young, 2021-06-15:
Yeah, I've looked at the repo and it looks really slick. I just want to do things "the hard way" first to learn the ins and outs of the devices. It's my first real foray into Raspberry Pi and hardware sensors and so far most of my issues have been hardware/wiring (it's been >20 years since I was wiring up PLCs), not software, and those are much easier to debug if I'm dealing directly with the code.

I'll probably take a deeper look at it around the time the Atlas sensors make if off my wish list and onto my shopping list.

rstrouse, 2021-06-15:
The code there will learn you pretty good regarding the protocol references.

closed 2021-06-19 by tagyoureit