tagyoureit/nodejs-poolController #327: Intellitouch i7+3 missing circuits

dkroesche, 2021-06-23:
**Describe the bug**
Hi - I am having problems with not all of my circuits being identified. I was previously using an older version of poolController that had all my circuits. I wanted to update to get some bug fixes and now I am missing two circuits. I've tried stopping the controller and manually adding them, but when I start again they get removed. I have a Pentair Intellitouch i7+3 system.  I've also tried removing the config files in the data directory and restarting, but still get the same results. 

Here are the missing circuits from my previous config file:

    {
      "id": 7,
      "name": "Pool Light",
      "nameId": 63,
      "type": 7,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 0
    },
    {
      "id": 8,
      "name": "Low Speed",
      "nameId": 48,
      "type": 0,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 0
    }

Circuits 1-6 are the same in the new and old config.

**To Reproduce**
This happens every time.  I've removed the config file and restarted and get the same results.  I've tried to manually add the missing circuits while the controller is not running, but they get removed when I re-start.

**Expected behavior**
Along with the 6 circuits being identified (see capture), the circuits mentioned in my description above should also be identified.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Packet Capture**
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/6698118/replay.zip)

**Pool Equipment**
 - Controller: IntelliTouch
 - Controller Model: IntelliTouch i7+3
 - Controller Firmware Version: undefined
 - Pump(s) manufacturer and model: IntelliFlow VSF 
 - Chlorinator: N/A
 - Heater(s): gas
 - Chemical controller: N/A
 - Valves: N/A
 - Any other relevant equipment:

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

dkroesche, 2021-06-23:
I saw something similar in issue #281 so I made a similar change to the fix there so that circuit 7 & 8 would not be marked as invalid in  my system.   After doing that, my pool light (circuit #7) shows up and works.  My "Low Speed" circuit still didn't show up.

rstrouse, 2021-06-23:
Yeah that was for the first 6 circuits.  For some odd reason I though this all operated differently than IntelliCenter but it is exactly the same.  This has to do with how shared vs single body OCPs allocate their Ids.
For instance i7+3 actually has 8 total ids.  However, circuit 1 and 6 are shared.  If this were an i7+3S then circuit 1 is simply disabled and 2-8 represent the relays.  This is precisely how IntelliCenter operates.  The only thing that gets a little weird is that Touch systems reuse the ids that are not assigned to circuits as features.

I have a change in the works to rectify this but it we are working some changes to the inbound responses at the moment.

dkroesche, 2021-06-23:
As I mentioned, circuit 8 didn't get automatically added but I was able to add it manually and it now "sticks" and is working.  I'm not sure if that is expected, or if additional code changes are needed to make circuit 8 also be discovered.

So, I have things working pretty well with a local change to allow these circuits.   However, I did notice one more oddity.  My SPA type is getting set to 0 (Generic) instead of 1 (SPA).  If I try to change it manually, it goes back to generic as soon as a restart poolController.   Also, probably related, in Dashpanel, I am unable to change the type for my SPA circuit.  I can change the type for all other circuits in Dashpanel.

Thanks for your help!
David

[image]

rstrouse, 2021-06-23:
Thanks for letting me know about the spa circuit.  This is an njsPC thing.  Oddly enough Pentair allows you to assign pool/spa to random aux circuits but they do not do anything special.  However, when they report the circuit function for either pool or spa in circuit 1 and 6 it is always 0.  Somewhere we are not checking if the incoming function is for circuit 1 and redirecting it appropriately to either pool or spa.

rstrouse, 2021-06-23:
To answer your other questions you should not be able to change the circuit function for pool or spa.  This is correct in dashPanel.  The other change was made in another repo that still has some work as we weed out some changes to improve response message processing.  I will fix the circuit functions in there as well after some collaboration with @tagyoureit.

dkroesche, 2021-06-23:
The SPA change to 'GENERIC' also shows up on my pentair control panel.   Similar to what you described in Dashpanel, the Pentair panel won't let me change the type for pool or spa.   When you fix the code to change the type for SPA, will that also update the programming for my controller?   The reason I ask is because I thought njsPC was mostly reading configuration data and not actually making changes.

Thanks,
David

rstrouse, 2021-06-23:
I'll have a look at the code but we will force circuit 1 & 6 to spa and pool respectively.  However, ET always sends back 0 (generic) for these two circuits when you ask for the circuit function I believe.

dashPanel/njsPC does allow you to make configuration changes.  However, the ability to change the circuit function byte for circuit 1 or 6 is not one of its abilities.  The dropdown should be disabled for those circuits.

tagyoureit, 2021-06-24:
Can you go out to your controller and hit the reset button?  Let's see if that clears it up first.

dkroesche, 2021-06-24:
That didn't make a difference.  I also restarted njsPC.  SPA is still type GENERIC.

Thanks,
David

rstrouse, 2021-06-24:
Since resetting the panel didn't do anything and dashPanel doesn't allow you to change the type.  Clearly this byte is not set correctly in the OCP.  So we have a couple of options here.  The easiest way would be to send an http or mqtt messsage to njsPC.  Are you running an MQTT broker with njsPC or do you know how to send an http request to njsPC?

If not I can walk you through sending an RS485 message through messageManager that will tell IntelliTouch that it is wrong regarding the type for circuit 1.

rstrouse, 2021-06-24:
I have updated the signatures for the i7+3 that should fix your missing circuits.

The following curl command from the raspberry pi command line should fix your type issue.  Replace the `<ip address of njsPC>` with the actual address of the raspberry pi where njsPC is running and this should set the type correctly for you.

```curl -X PUT --header 'Content-Type: application/json' --header 'Accept: application/json' -d '{"id":1, "type":1}' 'http://<ip address of njsPC>:4200/config/circuit'```

dkroesche, 2021-06-24:
Hi,

The command to fix my SPA configuration worked.  Thanks!

I will update to the latest code later today to confirm that fix as well.

Thanks,
David

rstrouse, 2021-06-24:
I thought that might be clever enough to work.

dkroesche, 2021-06-25:
Hi,

I've tested the new code and it seems to be working as far as my circuits go.  However, I am getting a steady stream of this message from njsPC:

2|njsPC          | resetting connection timer

I pulled the latest code.  Is this a known issue?

Thanks,
David

tagyoureit, 2021-06-25:
You shouldn't see that unless your port is timing out or no packets are being sent from Intellitouch.  How often does it happen?  And what is the value of your `inactivityRetry` in config.json?

dkroesche, 2021-06-25:
It's kind of bursty, but I would say it happens about 3 times per second.

inactivityRetry is 10

rstrouse, 2021-06-25:
Actually you can ignore that message.  That is a diagnostic message to troubleshoot adapter issues.

dkroesche, 2021-06-25:
In that case, I will probably comment it out in my local code so it doesn't fill up the log and make it more difficult to spot other useful info.

Thanks again for your help.  Should I click the close button, or do you do that?

rstrouse, 2021-06-25:
I already commented it out and checked it in.  Line 39 of Comms.ts

rstrouse, 2021-06-26:
Either one of us can do it

closed 2021-06-26 by rstrouse

dkroesche, 2021-06-26:
I've pulled the latest code and everything is working well.  Thanks again for your help!