tagyoureit/nodejs-poolController #329: Influx2.0 not supported

zombielinux, 2021-07-13:
**Describe the bug**
Influx has recently released version 2, the built in InfluxDB component no longer functions and instead seems to return "Unauthorized access"

**To Reproduce**
1. Set up InfluxDB in docker container as described on their github page (https://github.com/docker-library/docs/blob/master/influxdb/README.md)
2. Create bucket for nodejs-poolController
3. Follow instructions here https://docs.influxdata.com/influxdb/v2.0/reference/cli/influx/v1/auth/ to create credentials for v1 clients to access the bucket. 

**Expected behavior**
I expected the data to be written. I suspect a similar issue as Telegraf has been having. 

**Pool Equipment**
 - Chlorinator: IntelliChlor-40]

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser: Firefox
 - Version Latest

**Additional context**
Influx Error messages: 

```[13/07/2021, 21:26:57] error: InfluxDB tag binding failure on body:heatMode/@bind=data.heatMode.desc; --> @bind=data.heatMode.desc;  {"id":1,"heaterOptions":{"total":0,"gas":0,"solar":0,"heatpump":0,"ultratemp":0,"hybrid":0,"maxetherm":0,"mastertemp":0},"name":"Pool","setPoint":0,"circuit":6,"type":{"val":0,"name":"pool","desc":"Pool"},"heatStatus":{"val":0,"name":"off","desc":"Off"},"isOn":false}

```
ERROR: Write to InfluxDB failed. d [HttpError]: unauthorized access
    at IncomingMessage.<anonymous> (/home/pi/nodejs-poolController/node_modules/@influxdata/influxdb-client/src/impl/node/NodeHttpTransport.ts:255:13)
    at IncomingMessage.emit (node:events:406:35)
    at endReadableNT (node:internal/streams/readable:1329:12)
    at processTicksAndRejections (node:internal/process/task_queues:83:21) {
  statusCode: 401,
  statusMessage: 'Unauthorized',
  body: '{"code":"unauthorized","message":"unauthorized access"}',
  contentType: 'application/json; charset=utf-8',
  json: { code: 'unauthorized', message: 'unauthorized access' },
  code: 'unauthorized',
  _retryAfter: 0
}
```

```

rstrouse, 2021-07-13:
Have you tried adding the `u:p` to the url or the `Authorization: Token` header in the config.json file.

zombielinux, 2021-07-13:
Where would that be documented? I’m not seeing any relevant information in
the wiki.

rstrouse, 2021-07-13:
Here

https://docs.influxdata.com/influxdb/v2.0/reference/api/influxdb-1x/#1x-compatible-authorizations

The options are defined like all other interfaces in the config.json and passed into the interface.

zombielinux, 2021-07-14:
> Here
> 
> https://docs.influxdata.com/influxdb/v2.0/reference/api/influxdb-1x/#1x-compatible-authorizations
> 
> The options are defined like all other interfaces in the config.json and passed into the interface.

I meant the config.json documentation for nodejs-poolController, do i just add "Authorization": "Token", under the "options" header itself?

rstrouse, 2021-07-14:
Actually, try the username/password first.  Then set the compatibility on Influx.

zombielinux, 2021-07-15:
Yep. I’ve tried that as well. Also attempted the dbrp mapping. I was able
to get it working with some basic python I wrote for a different project,
but not this one.

rstrouse, 2021-07-16:
Well it looks like they put us between a rock and a hard place.  The 2.x client does not support 1.x servers and 1.x clients cannot connect to 2.x servers.  I guess they think that somehow software is just static and everyone is ready to toss their 1.x servers, spend time to convert the data, and run the 64-bit pi OS.  Sounds to me like the sun is orbiting the earth at this moment in time.

zombielinux, 2021-07-16:
I don’t think that’s the case. My python code is using the 1.x
authentication scheme (user, password, database name) vs the 2.x scheme
(bucket, token) and it’s working just fine.

I think there is something else in the nodejs code that is stalling things
out.

johnny2678, 2021-07-16:
> I guess they think that somehow software is just static and everyone is ready to toss their 1.x servers

I realize this isn't for everybody, but I just froze my influx docker on `influxdb:1.8.4-alpine`.  I'm not running enterprise IT for an IOT shop.  It's my house and I just want to store/visualize the data the same way I have been doing it for the past 3-5 years.  No time for major version/schema changes now that I have it the way I like it.

zombielinux, 2021-07-16:
I understand that. I made the leap to 2.0 because it touted backwards
compatibility as well as the ability to timeshift data series relative to
themselves (I.e compare this year with last, without year before)

I’ll try to do some more digging myself to see where the problem might lie.

johnny2678, 2021-07-16:
> I made the leap to 2.0 because it touted backwards compatibility

I haven't looked at it recently, but when I did look into 2.0 it looked like a whole new query language. Everything would have to be rewritten... Grafana queries, nodered flows, python scripts, etc... it's exhausting just thinking about it.

Would be happy to be wrong about this.

rstrouse, 2021-07-16:
> I don’t think that’s the case. My python code is using the 1.x
authentication scheme (user, password, database name) vs the 2.x scheme
(bucket, token) and it’s working just fine.

I'll bet your code is running a 2.x client though.  Try connecting that to a 1.x server.  It appears the 2.x client requires 64-bit pi OS as well, although I cannot seem to pin down anything definitive saying one way or another.

zombielinux, 2021-07-17:
> I'll bet your code is running a 2.x client though. Try connecting that to a 1.x server. It appears the 2.x client requires 64-bit pi OS as well, although I cannot seem to pin down anything [...]

Nope, changing nothing but the URL and the port number (spun up a legacy influx DB container) allows my code to write to the legacy 1.8.6 database. 

So that tells me I've got the legacy v1 bridge set up properly in my 2.0 influx.

emes, 2021-07-18:
> I haven't looked at it recently, but when I did look into 2.0 it looked like a whole new query language. Everything would have to be rewritten... Grafana queries, nodered flows, python scripts, [...]

You're (mostly) not wrong.

zombielinux, 2021-07-18:
The whole point of the backwards compatibility is not having to change
anything and just providing the credentials and legacy bridging on the
database side.

tagyoureit, 2021-07-18:
I'll take a look this week. Knee deep in another part of the code now.

zombielinux, 2021-08-10:
> I'll take a look this week. Knee deep in another part of the code now.

Has this been investigated?

tagyoureit, 2021-08-12:
No, unfortunately not yet.  Been tied up with a few other things.  Still on my radar.  Thanks for the patience.

tagyoureit, 2021-08-24:
I pushed up 7.3.1 with some changes for supporting Influx 2.0.  The API's internally are the same but...
1. user/password is now a singular token
2. org is required
3. bucket is now an id that represents the (formerly known as) database/retention policy

Further, the API is a bit whacky in that it caches writes for performance.  This is normally a good thing, but it provides no way to trap errors.  We should have been seeing something like:

```js
[HttpError]: unauthorized access
    at IncomingMessage.<anonymous> (D:\Programming\nodejs-poolController\node_modules\@influxdata\influxdb-client\dist\index.js:16:10167)
    at IncomingMessage.emit (events.js:326:22)
    at endReadableNT (_stream_readable.js:1241:12)
    at processTicksAndRejections (internal/process/task_queues.js:84:21) {
  statusCode: 401,
  statusMessage: 'Unauthorized',
  body: '{"code":"unauthorized","message":"unauthorized access"}',
  contentType: 'application/json; charset=utf-8',
  json: { code: 'unauthorized', message: 'unauthorized access' },
  code: 'unauthorized',
  _retryAfter: 0
}
```

but they were just swallowed up.  I'm now forcing an explicit write so we can trap any of these errors if they should surface with config or server side errors.

Please do the following:
1. Pull the latest code
2. Run `npm i` in the repo.  I updated to the latest version of the Influx client.  Most likely it's just bug fixes and the existing one will work but I didn't test it out.
3. Start/stop the app and a new InfluxDBv2 binding section will appear.  See the [wiki - InfluxDBv2](https://github.com/tagyoureit/nodejs-poolController/wiki/Bindings-Integrations-in-2.0#influxdb-v20) for the parameters.  Edit this with the right values and then restart the app and measurements should show up within a few mins (depending on processing time on the Influx side).

Let me know if it all works.  It took quite a while to go through all of the docs and figure out how to translate from 1.8 to 2.0 but I think we are there.

If you have any tips/tricks/suggestions/advice about using Telegraph, scrapers, dashboards, etc I'd be happy to hear.  Everyone here will now want to recreate/upgrade the existing dashboards and take advantage of all new functionality and we haven't scratched the surface yet.

johnny2678, 2021-08-24:
> Everyone here will now want to recreate/upgrade the existing dashboards and take advantage of all new functionality

Welllll.... not everyone 🙃

Does this update maintain compatibility with Influx 1.8.6?

tagyoureit, 2021-08-24:
Yes, still compatible with v1.8.

closed 2021-09-03 by tagyoureit