tagyoureit/nodejs-poolController #330: Add Home Assistant Setup Instructions to Wiki

sonapsent, 2021-07-17:
As requested here is a writeup for the wiki to setup with Home Assistant. I am new to this, so look it over and let me know if you would like to see any changes.

[nodePC - HA.md](https://github.com/tagyoureit/nodejs-poolController/files/6835416/nodePC.-.HA.md)

truvec, 2021-07-19:
Looks good. Your post inspired me set this up too. Here are two automations that could be used instead of NodeRed to keep the input selects in sync.

```
automation:
  - id: '808456'
    alias: "Set SWG slider"
    trigger:
      platform: mqtt
      topic: "pool/state/chlorinators/1/aquarite/poolSetpoint"
    action:
      service: input_number.set_value
      target:
        entity_id: input_number.pool_chlorine_set_point
      data:
        value: "{{ trigger.payload_json.poolSetpoint }}"

  - id: '808457'
    alias: "SWG slider moved"
    trigger:
      platform: state
      entity_id: input_number.pool_chlorine_set_point
    action:
      service: mqtt.publish
      data:
        topic: "pool/state/chlorinator"
        retain: true
        payload: "{\"id\":1,\"poolSetpoint\":{{ states('input_number.pool_chlorine_set_point')|round(0)}}}"
```

556duckvader, 2021-07-21:
@sonapsent  One thing I noticed in your lovelace card you have a switch next to the SWG stuff to enable it and I see its pointing to the entity input_boolean.make_chlorine.  I am assuming this is used to turn the SWG on and off.  What is the code for the input_boolean.make_chlorine?

556duckvader, 2021-07-21:
Got mine all setup thanks again for your write-up!!
[image]

sonapsent, 2021-07-21:
Great, glad to it going to good use a few days after putting it together. I have only had my system hooked up for a month or two and am really impressed and thankful with how well it works. As for my "make / enable chlorine" button I am still trying to figure out how to best use it. I do not yet have a bunch of confidence with ORP sensors from what I have read from troublefreepool.com so I am waiting to see what level of automation makes sense.

greenb8, 2021-07-29:
Any idea how i would go about changing the colors on my intellibrite with this?

rstrouse, 2021-07-29:
If you have a circuit with a circuit type of IntelliBrite it simply needs to set the theme.  Either the OCP or Nixie will take care of the rest.

sonapsent, 2021-08-02:
Use an application like MQTT Explorer to monitor the messages live.

In home assistant you can test your messages by sending them in developer-tools / service / MQTT: Publish.

I would imagine the result would look similar to this example circuit:

~~~
switch:
  - platform: mqtt
    name: Pool Pump Speed 1
    icon: mdi:pump
    state_topic: "pool/state/circuits/2/pumpsp1"
    command_topic: "pool/state/circuits/setState"
    payload_on: '{"id":2,"isOn":"on"}'
    payload_off: '{"id":2,"isOn":"off"}'
    optimistic: false
    retain: false
    qos: 0
~~~

tagyoureit, 2021-08-05:
Added to wiki.  Thanks!
https://github.com/tagyoureit/nodejs-poolController/wiki/Home-Assistant-Setup-Instructions

closed 2021-08-05 by tagyoureit