tagyoureit/nodejs-poolController #334: influx Write Failing (Influx v1.8.6)

arrmo, 2021-08-11:
**Describe the bug**
As discussed in gitter.im, no output to influx database it seems.

**To Reproduce**
Start up njsPC, check traffic to influx, and as well data inside influx => confirm no data points written.

**Expected behavior**
Data stored to influx.

**Screenshots**
N/A

**Packet Capture**
Replay attached!
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/6969994/replay.zip)

rstrouse, 2021-08-11:
I am very confused by this output.  There are no RS485 messages or any socket responses.  Did you run `npm -i` after you did `git pull`?

We upgraded the socket io and serial adapter a couple of months ago.  Perhaps both of these interfaces are not connecting for output.  I do see that it opened the serial port but I don't see any traffic whatsoever.

rstrouse, 2021-08-11:
For the `npm -i` run it again.  This time use `sudo npm -i` and see if it completes pulling down the socket binaries.  Perhaps there were more errors to include the serial port as well.  Either way it must complete without error.

In the end, our issue with Influx is that there is no RS485 traffic coming from the RS485 bus since 08-08-2021 6:58pm.  We need to do some troubleshooting of the RS485 connection.  It is dead silent and there is no incoming traffic so njsPC will sit in a holding pattern until it confirms the existence of your IntelliTouch.  Double check your connections to the port and make sure IntelliTouch is up and running.

arrmo, 2021-08-13:
> For the `npm -i` run it again. This time use `sudo npm -i` and see if it completes pulling down the socket binaries. Perhaps there were more errors to include the serial port as well. Either way it must complete without error.

That was it, thanks! sudo was the trick - I admit, didn't think it was needed, but I was wrong. Thanks for the suggestion, back up and running. Appreciate it!

closed 2021-08-13 by arrmo