tagyoureit/nodejs-poolController #336: Chlorinator ORP Branch

DrJeff, 2021-08-16:
Ok the beginning of SWG ORP help
The IC60 model isn't sticking here is the replay [replay_IC60.zip](https://github.com/tagyoureit/nodejs-poolController/files/6995128/replay_IC60.zip)

tagyoureit, 2021-08-16:
It's giving an error like it doesn't have all of the right files.  

Do you have this chlorinatorModel section in SystemBoard.ts like it is here: https://github.com/tagyoureit/nodejs-poolController/blob/chlorOrp/controller/boards/SystemBoard.ts#L443?

DrJeff, 2021-08-16:
Yes I have that

tagyoureit, 2021-08-16:
did you run TSC standalone or as part of starting up njsPC?

DrJeff, 2021-08-16:
TSC?

tagyoureit, 2021-08-16:
typescript compile.  If you run `npm start` it will compile it.  Or you can run `npm run build` to build it standalone.

DrJeff, 2021-08-16:
I'm using pm2 let me see...
`npm start`

tagyoureit, 2021-08-16:
I just pushed up a change for the undefined map error I saw, but it wasn't related to the chlor model.  Pull again and see if it sticks now.

DrJeff, 2021-08-16:
IC60 still doesn't stick in GUI Mixing now will see if dose works in 10min

DrJeff, 2021-08-16:
Turned on capture and borked my install, I remember this happening in the past with capture let me load backup

tagyoureit, 2021-08-16:
I thought I changed the default to not reset the config.  👎

DrJeff, 2021-08-16:
Oh my, so do I just replace poolConfig and poolState?

DrJeff, 2021-08-16:
Ok got it back up! where is the capture if I set it inside of defaultConfig?

tagyoureit, 2021-08-17:
It will be in the logs directory and a sub-directory with the date/time.

DrJeff, 2021-08-17:
can I set the controller in the code or config somewhere the GUI just isn't letting me keep it

DrJeff, 2021-08-17:
Still not sticking I set it in the poolConfig.json type 4
```
  "chlorinators": [
    {
      "id": 1,
      "disabled": false,
      "ignoreSaltReading": false,
      "master": 1,
      "isActive": true,
      "name": "IntelliChlor",
      "body": 0,
      "poolSetpoint": 100,
      "spaSetpoint": 0,
      "superChlorHours": 1,
      "superChlor": false,
      "type": 4,
      "isDosing": true
    }
  ],
```
But I still get this error
```
1|poolController  | [8/16/2021, 6:44:07 PM] error: Error polling Chem Controller - EquipmentNotFound: Please specify a chlorinator model to allow Nixie to calculate chlorine demand
1|poolController  | [8/16/2021, 6:44:07 PM] verbose: End polling Chem Controller 50
1|poolController  | [8/16/2021, 6:44:07 PM] error: Please specify a chlorinator model to allow Nixie to calculate chlorine demand
1|poolController  | [8/16/2021, 6:44:08 PM] verbose: Chem dose ignore chlor because it doesn't have a dosing rating.
```

tagyoureit, 2021-08-17:
I'm seeing dashPanel send `[8/16/2021, 7:31:47 PM] info: [7:31:47 PM] 127.0.0.1 PUT /config/chlorinator {"id":1,"name":"IntelliChlor","body":0,"model":4,"poolSetpoint":100,"spaSetpoint":0,"superChlorHours":1,"ignoreSaltReading":false}` and the type is correctly stored.

If you made the changes manually, did you stop njsPC, change the poolConfig.json, and then restart njsPC?

DrJeff, 2021-08-17:
> did you stop njsPC, change the poolConfig.json, and then restart njsPC

Yes, I remember being told to always stop before making any changes

tagyoureit, 2021-08-19:
Closing this until you are ready to pick it back up.

closed 2021-08-19 by tagyoureit