tagyoureit/nodejs-poolController #338: Chlorinator goes offline when setting to 16%

truvec, 2021-08-24:
Chlorinator goes offline when setting it to 16%. On the Aquarrite the no flow led blinks and the controlled remotely led goes off. This happens when set via Pool Control dashboard or via MQTT.

I have a Nixie single body with intelliflo vs and Hayward aquarite.

tagyoureit, 2021-09-03:
[image]

I had a chance to look into this for a few minutes, and, well color me silly... this is something new.  The packets for setting the chlor to 15% and 17% are the same, but the packet for 16% has an extra 0 in the payload.  I don't think I've ever seen anything like that before.  

Pull the latest.
I'll push up a fix for this in a bit.

truvec, 2021-09-03:
OK- Yes it does now set to 16%. Do you need to see the output? 

Thanks!

tagyoureit, 2021-09-03:
Nope, all good.  Thanks for confirming.  Sweet 16!!!

closed 2021-09-03 by tagyoureit