tagyoureit/nodejs-poolController #340: Commands fail using Network connection

jwtaylor310, 2021-08-31:
When njsPC is connected to the controller through a network connection commands to turn on/off functions fail after the program has been running a while - typically an hour or more.  However, the DashPanel continues to display correct system status.

**To Reproduce**
Steps to reproduce the behavior:
1. Start njsPC Controller and DashPanel on a Raspberry Pi
2. Open a web browser to the DashPanel.  Wait 1+ hours
3. Click on a function ("Lights" for example)
4. Lights do not go on
5. On the Pi, restart the Controller
6. Click on a function and it will turn on/off as expected.

In some cases when the DashPanel has been running for several hours it will display "Loading x%" and status "light" will turn yellow.  When this occurs the DashPanel continues to display current system status.

**Expected behavior**
Functions should reliably go on/off as commanded. 

**Packet Capture**
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/7086409/replay.zip)
Note: After starting the capture, and selecting "Lights On", the Settings menu would not display.  I had to restart the Controller and DashPanel to regain access to the Settings menu to turn off the capture.

**Pool Equipment**
 - Controller: EasyTouch
 - Controller Model: [EasyTouch 2.8]

*** Software Versions ***
njsPC 7.3.1 commit 03f6c4b
dashPanel 7.3.0  commit df0c37f
 
**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser FireFox 
 - Version 91.0.2

**Additional context**

Until recently this system was connected via an RS485/USB adapter and did not have this problem.  Last week I switched to the network connection.  At that time I reported a problem with the Send Stats showing 100% failure rate.  You provided an update which fixed that problem and it is that version of the software that is currently running on my system.

Attached is a copy of the text displayed by the Controller centered around the time of a failure:

Session 1 (1:07 PM): Function 2 (Lights) selected for "On", then "Off".  The DashPanel did not indicate that the lights went on but they did, in fact, do so.

Session 2 (1:10 PM): Function 7 selected for "On", then "Off".  Worked as expected.

Session 3 (2:56 PM): Function 7 selected for "On", then "Off".  Failed.  Lights did not go on.

[Console_Log.docx](https://github.com/tagyoureit/nodejs-poolController/files/7086288/Console_Log.docx)

rstrouse, 2021-09-02:
Hey that console_log.docx file is a screenshot of a usb adapter hooked up.  If you are using the wifi adapter that setting needs to be network.

Do you have a heat pump, gas heater, and solar?  Your configuration shows all 3 but that may be because ET sent some garbage related to the heaters.

jwtaylor310, 2021-09-02:
I uploaded the wrong file.  The system is definitely configured for network.  I've attached the correct  (hopefully) log file here:
[njsPool Console Log.txt](https://github.com/tagyoureit/nodejs-poolController/files/7100497/njsPool.Console.Log.txt)

My system has a Pentair "Ultratemp"  heat pump and solar.  njsPC is reporting a Gas Heater which doesn't exist.  When I try to delete the Gas Heater in njsPC the deletion appears to work but the Gas Heater later reappears.  I think this reflects a problem with the Pentair controller.  When the solar panels were installed (about 2 years ago) neither the solar installer nor the Pentair service person were able to get the controller to allow both heat pump and solar to be enabled at the same time.  To get the system to work they wired a connection from the Pentair controller to the heat pump water flow sense circuit so that when the Pentair controller isn't calling for heat the heat pump thinks it has low water flow and shuts down.  Supposedly that is a Pentair-approved "work around".   I'm guessing that the Pentair controller thinks the low water flow connection is a Gas Heater but also sees the heat pump through its RS485 connection.  

My on-going problem is with on/off commands from the DashPanel failing intermittently.  Typically if after several hours of no activity I try to turn on or off a feature through the DashPanel the feature's indicator "light" switches to the new state and a few seconds later switches back to the original state.  If I try again sometimes the on/off transition will work correctly and other times it won't.  When it does, I can then make other changes without a problem.  I'm guessing that the outgoing connection to the socket server is being closed and not reliably re-opened.    However, even when commands from the DashPanel are being ignored, the DashPanel continues to reflect the current status of the system including status changes commanded by other means.  If I ignore the problem for a few hours it sometimes heals itself.  Other times I have to restart the Controller program which reliably fixes the problem for the short term.

rstrouse, 2021-09-02:
Yeah the on/off problem may be a symptom of the heater issue believe it or not.  I am writing some code to clean up the configuration so that it doesn't assume you want to have 3 heaters on the OCP.

jwtaylor310, 2021-09-02:
Wow … that’s going above and beyond.  Thank you!  --John

rstrouse, 2021-09-02:
In your EasyTouch configuration I assume you have a Gas Heater defined (ET controllers always enable the heater relay) and a Solar defined.  However, do you have the Solar is Heatpump checked?  Really, it blows me away that they would create this as a workaround rather than updating the firmware.  If this were one of the UltraTemp Hybrids I wouldn't think their workaround would be save since it really is two heaters in one.

Interestingly, this would all work swimmingly on IntelliCenter.

rstrouse, 2021-09-02:
Booh!  Your Pentair rep is full of crap.  The ET controller never communicates with the Ultratemp.  I have an output of the messages sent by ET and it never once sends an message to it.  As a result it never responds.  There would be no difference between having the RS485 hooked up or not.  I was hoping to spoof the messages coming from the ET controller when it sent them off so I could twist njsPC into controlling it where the ET fell short.

jwtaylor310, 2021-09-02:
On my system, currently there is only one data cable running from the ET controller to the Ultratemp, and I'm reasonably confident that it is used for the 'low flow' kludge.  If that is right, there is no RS485 connection to the Ultratemp.  There may have been one before the reconfiguration for the solar panels, but not now.

I don't see any Gas Heater defined in the ET controller.  Under Settings there are options for:

UltraTemp ETi which is currently set to Enabled: Yes.  There are entries for Settings and Status which are not populated.

Heat Pump Com which is set to Enable: No, Heating : No and Cooling: No.  and Status: [OP NORMAL ]

Solar: which is set to Enable: Yes  Heat Pump: No  Freeze Enbl: No  Night Cool: Yes Temperature Diff Start: 3 Run 2
(If Heat Pump is set to Yes, the Freeze Enbl and Night Cool options disappear)

There is also a Settings entry for "Man Heat" which is set On

This controller board was replaced about a month ago and the tech had to manually enter all of the settings so it is possible that he didn't set up the heater correctly.  In any case, I don't see an entry for "Gas Heater" unless it is buried in some other part of the configuration.

BTW, I used the night cooling feature regularly this time of year and was pleasantly surprised to see that the DashPanel correctly displays a snowflake icon when that feature is active!

rstrouse, 2021-09-02:
In ET the gas heater is always there by default.  There is no way to disable it.  The only thing you can configure is solar but solar and UltraTemp ETi are mutually exclusive meaning you cannot have a solar system and a heatpump at the same time.

The Man Heat function isn't relevant to the issue it simply disables the feature that turns on the heater if you select spa.  I suspect you only get the option to select UltraTemp features and no ability to set solar features on the panel.  Is that correct?

jwtaylor310, 2021-09-02:
The panel offers the following Solar features:

Enable: Yes/No
Heat Pump: Yes/No
Freeze Enbl: Yes/No
Night Cool: Yes/No
Temperature Diff
   Start: 3 degrees
   Run 2 degrees

When Night Cool is enabled, "Max Temp" is added to the Pool temperature setting screen.  This setting can only be changed on the physical panel.  The Pentair wireless remote will allow the setting to be raised or lowered but won't actually record the change.

As I recall, enabling the Heat Pump will disable the Solar options.

rstrouse, 2021-09-02:
Yeah Pentair needs to be flogged for this implementation.  Unfortunately, they replace the solar heater (probably because they share a sensor) instead of the Gas Heater which cannot be uninstalled.  Heck why they didn't allow you to simply install another heater and disable the gas heater is beyond me.  Instead they chose to hack in a heatpump setting.

Go ahead and pull njsPC as well as dashPanel and lets see if it eliminates all the undefined heaters from the configuration.  Then if it does we can be assured the comms aren't getting hosed by the heater weirdness.

If we are still having comms issues then the port may be losing its settings.  What I mean by that is that at the buffering end it isn't keeping the 9600, 8, N, 1 settings for the port and renegotiating a faster speed.  Perhaps there is a negotiation setting somewhere that forces it to the lower speed.

jwtaylor310, 2021-09-02:
I pulled and reinstalled both.  This time there were no warning/error messages about heaters during the startup of nodeJS and the dashPanel reports only a Gas Heater and a Solar Heater.  However, the dashPanel throws "OutboundMessageErrors" when I try to start/stop any function.  I'm going to restart both and see if the system heals itself.  I have to leave for a couple of hours so I won't be able to report the results until later this evening, but I will definitely follow up then.  Thanks!

jwtaylor310, 2021-09-02:
I'm back.  The OutboundMessageErrors were caused by a network issue unrelated to the pool system.  I fixed that and then restarted njsPC and dashPanel and everything is working as expected.  There are no heater warnings or errors.  The dashPanel displays current status correctly and on/off commands work normally.

The only warning messages are repeated message aborts for 165,30,16,33,203,1,21,1,213 and 165,30,16,33,203,1,22,1,214.  The full startup log follows.  The last two lines in the log are the result of me turning the pool lights on and then off which worked as designed.

I'll monitor the system and let you know if the on/off command problem recurs.

------------------------------------------
njsPC console log:

Init state for Pool Controller
[9/2/2021, 6:33:50 PM] info: The current git branch output is master
[9/2/2021, 6:33:50 PM] info: The current git commit output is 1efff3c2818d33fd33eaa07af812282c3d97dea2
[9/2/2021, 6:33:50 PM] info: Starting up SSDP server
[9/2/2021, 6:33:50 PM] info: Checking njsPC versions...
[9/2/2021, 6:33:50 PM] info: Starting Pool System easytouch
[9/2/2021, 6:33:50 PM] info: Server is now listening on 0.0.0.0:4200
[9/2/2021, 6:33:50 PM] info: Net connect (socat) connected to: 172.16.31.241:8899
[9/2/2021, 6:33:50 PM] info: Net connect (socat) Connection connected
[9/2/2021, 6:33:50 PM] info: Net connect (socat) ready and communicating: 172.16.31.241:8899
Pentair EasyTouch System Detected!
[9/2/2021, 6:33:52 PM] info: Found Controller Board EasyTouch2 8
[9/2/2021, 6:33:52 PM] info: New socket client connected 6EsIWcedC7VnS0f2AAAB -- 192.168.80.55
[9/2/2021, 6:33:52 PM] info: Requesting easytouch configuration
[9/2/2021, 6:33:54 PM] info: New socket client connected GsHvW6r4c7ZmntCAAAAD -- 172.16.31.151
RS485 Stats:{ "bytesReceived": 1129 "success": 39, "failed": 1, "bytesSent": 408, "collisions": 0, "failureRate": 2.50% }
[9/2/2021, 6:34:04 PM] warn: Message aborted after 4 attempt(s): 165,30,16,33,203,1,21,1,213 
[9/2/2021, 6:34:09 PM] warn: Message aborted after 4 attempt(s): 165,30,16,33,203,1,22,1,214 
RS485 Stats:{ "bytesReceived": 1293 "success": 43, "failed": 2, "bytesSent": 480, "collisions": 0, "failureRate": 4.44% }
[9/2/2021, 6:34:18 PM] info: EasyTouch system config complete.
[9/2/2021, 6:34:18 PM] info: Initializing Nixie Controller
[9/2/2021, 6:34:18 PM] info: Nixie Controller Initialized
[9/2/2021, 6:56:02 PM] info: Requesting easytouch configuration
RS485 Stats:{ "bytesReceived": 42020 "success": 1188, "failed": 3, "bytesSent": 1034, "collisions": 1, "failureRate": 0.25% }
RS485 Stats:{ "bytesReceived": 42655 "success": 1211, "failed": 4, "bytesSent": 1226, "collisions": 1, "failureRate": 0.33% }
[9/2/2021, 6:56:17 PM] warn: Message aborted after 4 attempt(s): 165,30,16,33,203,1,21,1,213 
[9/2/2021, 6:56:21 PM] warn: Message aborted after 4 attempt(s): 165,30,16,33,203,1,22,1,214 
[9/2/2021, 6:56:28 PM] info: EasyTouch system config complete.
[9/2/2021, 6:56:28 PM] info: Initializing Nixie Controller
[9/2/2021, 6:56:28 PM] info: Nixie Controller Initialized
[9/2/2021, 6:57:01 PM] info: [6:57:01 PM] 172.16.31.151 PUT /state/circuit/setState {"id":2,"state":true}
[9/2/2021, 6:57:13 PM] info: [6:57:13 PM] 172.16.31.151 PUT /state/circuit/setState {"id":2,"state":false}

rstrouse, 2021-09-03:
Yeah the aborts can be normal when sending large numbers of messages to the controller.  It simply is getting stuck in traffic.  It will try again later.  I have asked @tagyoureit  what that ET message and whether it is important.

tagyoureit, 2021-09-03:
Those aren't a problem.  We were incorrectly asking the OCP for features 21 and 22 and it would just fail like you see above.  Funny no one noticed (or squawked) about it because it wouldn't really have any effect on finding the real circuits.   You can pull down the latest and it will fix those errors.

If you have other, unrelated network issues they could interfere with Socat.  You might want to have the output of a constant ping (from your njsPC to the remote pc) and see if there is any correlation between network blips and njsPC losing connectivity.

Also, it would be really interesting if you had any extra cables that could carry RS485 packets and hook up the ultatemp to the easytouch and see if they start talking to each other.

jwtaylor310, 2021-09-03:
The system ran overnight and was responding normally to on/off requests this morning, so the issue with on/off commands being ignored may have been resolved by the heater changes you made.  I installed the latest this morning and that fixed the aborted messages issue.   njsPC now starts with NO warnings or error messages!

I have configured a session on the NodeJS Pi that pings the socket server every 5 seconds and logs the output to a file.  The njsPC log shows it responding to an on/off request by doing a 'PUT' with the correct id and SetState information even when the system isn't responding, so it doesn't appear that the problem is between njsPC and the remote PC.  

I'll try to connect the Ultratemp to the RS485 bus this afternoon.

Thank you for your time and effort resolving these issues!

rstrouse, 2021-09-03:
Awesome.  Let us know if you have other issues.

jwtaylor310, 2021-09-03:
New error, "UnhandledPromiseRejection", this time in the dashPanel console:

Updated configuration file
info: Server is now listening on 192.168.80.85:5150
192.168.80.55 GET /
192.168.80.55 GET /socket.io-client/socket.io.js
192.168.80.55 GET /config/serviceUri?null
192.168.80.55 GET /favicon.ico
192.168.80.55 GET /config/serviceUri?null
192.168.80.55 GET /options?null
192.168.80.55 GET /config/appVersion?null
Updated configuration file
The current git branch output is master
The current git commit output is 8166137616747dfede885b0ac6723d0fa844c0fe
Updated configuration file
node:internal/process/promises:246
          triggerUncaughtException(err, true /* fromPromise */);
          ^

[UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "No data returned.".] {
  code: 'ERR_UNHANDLED_REJECTION'
}

Don't know if it is significant ...__

jwtaylor310, 2021-09-03:
I have connected the Ultratemp to the RS485 bus, enabled RS485 both on the Ultratemp and set "Heat Pump Com" to enabled on the OCP.  I also jumpered the water flow-to gas heater connection so the Ultratemp doesn't think it has a water flow issue.

The only indication that I can see that reflects the RS485 connection is that now when I switch from Pool to Spa mode the Ultratemp starts a timer while the water temperature is being sensed.  Presumably the Ultratemp is receiving a signal from the OCP telling it that Spa mode has been enabled.  However, the Ultratemp display shows 'N/A' for the Pool and Spa setpoints and does not turn on even when I used the OCTP to turn on Spa mode and set the Spa target temp well above the current water temp.

This is FYI only.  The system does what I need in the non-RS485 mode, so I don't need to get RS485 working to the Ultratemp.

Is there any testing or data capture that you would like me to try while the Ultratemp has RS485 connected and enabled?

tagyoureit, 2021-09-03:
Yes, thanks for offering!  If you can run a [replay capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) we'd love to see what the packets are...

Can you follow these steps?
1. Disable the ultratemp in screenlogic/OCP.
2. Start packet capture as above (do not capture configuration reset)
3. Enable the ultratemp; wait ~10-15s.  This will tell us if there any initialization packets going back/forth.
4. Change the heat setpoint so the heater kicks on; wait 10-15s for it to do it's thing
5. Set the setpoint so the heater turns off; again wait 10-15s.
6. End the replay and attach here

We reserve the right to ask for a couple more tests if we see something interesting.  ;)

jwtaylor310, 2021-09-04:
I ran the replay capture and it is attached.  The procedure was:

15:39:00 Start njsPC
15:36:40 Start Replay Capture
15:37:50 Change Ultratemp "Remote Mode" setting from "Off" to RS485"
15:38:20 On OCP, change Settings...Heater...Comm Enable from No to Yes, Heating from No to Yes, Cooling from No to Yes 
15:39:10 On WCP turns on Spa mode (calls for heat).  Ultatremp display shows temperature sense timer started.
15:40:17 Ultratemp display shows temperature sensor timeout ended.  (HP should start at this point but does not)
15:40:40 On WCP turn off Spa mode
15:41:30 End Replay Capture

The Spa was set for a temperature 10 degrees above the current water temperature so the HP should have turned on when Spa mode was activated.  The HP apparently recognized that Spa mode had been enabled because its display showed a temperature sense timer starting at the same time Spa mode was turned on.  Other than that, I saw no indication that the HP was being controlled by the OCP.  The HP did not switch between Pool and Spa modes and the set points for Pool and Heater showed "N/A" on the HP's display.

I was surprised that the Ultratemp never turned on when in RS485 mode. I don't know whether there is something wrong with this particular installation or if the Ultratemp RS485 isn't fully supported by the EasyTouch controller.  Possibly the firmware in my EasyTouch controller needs an update to support the Ultratemp.  It is currently at Ver 2.16.

To support both the Ultratemp and solar panels, my system has been configured with the Ultratemp's "Remote Mode" set Off and "Comm Enable" set to No on the OCP.  The "Gas Heater" connection on the OCP has been connected in series with the HP's water flow sensor.  The HP is set to operate manually in Pool mode with the temperature setting at its max.  With this configuration, the HP is off with a 'low flow' error until the OCP calls for heat and closes the Gas Heater circuit.

I hope the Replay helps you.  I will be interested to hear if you find any communication to/from the Ultratemp!

[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/7110453/replay.zip)

tagyoureit, 2021-09-06:
Very interesting.  So we do see some new packets...
[image]

They never change which is consistent with the heater never turning on.  At this point, we could give you some instructions on how to use `http://[njspc-dp]:5150/messageManager.html` to filter for these packets, but unless/until you can get it to respond we don't have any way of know which bytes in the packet do what.  Or like you said, it could be it would never work due to a FW version or the way your pool is wired up.  

Your FW version isn't too far off.  The latest is 2.18, but hard to tell the release dates or even patch notes.

If you want to keep playing with it when you have some free time, message manager will let you know when the packets change.

jwtaylor310, 2021-09-06:
Thanks for your continued interest in this.  I offer the following for whatever it may be worth.  njsPC and the dashPanel are working great for me so this information is being provided just as 'FYI' and not as a request for any sort of fix or upgrade.

I have a home-grown program that monitors all traffic on the RS485 buss and attempts to decode the Pentair and Intellichlor format messages.  This afternoon I started it and then enabled communications on the Ultratemp and the EasyTouch controller.  I then used the wireless remote to enable Spa mode which put the controller in heat mode.

All messages on the buss during this time were properly formatted Pentair messages with valid checksum bytes.  There were three sets of unknown messages that appear to be status/response pairs.  Two of those pairs appear to be directed to the pump.  The remaining pair is a likely candidate for communications with the Ultratemp.  Later, I disconnected the RS485 cable at the Ultratemp and ran the sequence again.  The suspect pair of messages did not appear during that run, which further supports the likelihood that the messages are associated with the Ultratemp.  Those messages are:

09/06/2021 16:38:15 Protocol:  0  Command: 114 - *** UNKNOWN ***                     Source: 16  Destination: 112 CRC: OK
 Header: b'\x00p\x10r\n'
Payload: b'\x90\x00\x00\x00\x00\x00\x00\x00\x00\x00'

09/06/2021 16:38:15 Protocol:  0  Command: 115 - *** UNKNOWN ***                     Source: 112  Destination: 16 CRC: OK
 Header: b'\x00\x10ps\n'
Payload: b'\xa0\x01\x00\x01\x00\x00\x00\x00\x00\x00'

Also, I previously reported that the Ultratemp starts a temperature sensing timer when the Spa is enabled.   It turns out that it will do that even with the RS485 line disconnected.  It looks like the Ultratemp is smart enough to start a new sensing timer cycle whenever the water flow rate changes significantly.  I confirmed this by increasing the pump speed while remaining in Pool mode which resulted in a new sensor cycle starting.  So the temperature sensing cycles appear to be unrelated to RS485 activity.  The only thing that I have found that suggests that the Ultratemp is actually aware of the RS485 connection is that when I disconnected the cable a communications error message appeared on the Ultratemp's display.

If you have a document available describing how to use the message manager I would be interested, though I don't think there is too much more that can be done with the Ultratemp communications, at least on my system. 

Here is the full log of the session with the Ultratemp connected to the RS485 buss:

[type1_summary.txt](https://github.com/tagyoureit/nodejs-poolController/files/7117894/type1_summary.txt)

rstrouse, 2021-09-07:
Actually, there are only 2 messages that are exchanged with UltraTemp.  These are action 114 which is a request from the OCP to the heater.  This message does a couple of things.  First, it disables the control panel on the heater.  Second, it tells the heater to go into one of 3 modes.  0 = Off, 1 = Heat, 2 = Cool.  This message is sent from the OCP to the heater every few seconds and has the following format.  Byte index 2-9 have never been witnessed as anything but 0.  We also have not witnessed anything but 144 for byte index 0.

In the end this acts the same way as a gas heater with the exception that the call for heat or cooling is sent through the RS485 bus rather than through the dry contact relay on the personality card.

Index | Value | Description
-- | -- | --
0 | 144 | Always appears to be 144   for this message.
1 | 0,1,2 | Sets heater mode to   0 = Off   1 = Heat   2 = Cool
2 | 0 |  
3 | 0 |  
4 | 0 |  
5 | 0 |  
6 | 0 |  
7 | 0 |  
8 | 0 |  
9 | 0 |  

The second message that is bandied about on the bus is the response to the previous message.  This message indicates back to the OCP what the current status of the heater happens to be.

Index | Value | Description
-- | -- | --
0 | 160 | Always appears to be 160 for this message.  We have not seen UltraTemp send back any other value.
1 | 1 | We have only ever seen a 1 in this position.
2 | 0,1,2 | Reports the current heater mode to   0 = Off   1 = Heat   2 = Cool.  In some cases the heater will report 0 when we request it to be on.  Perhaps this is a case where there is insufficient flow or the heater is busy doing other things.
3 | 0 |  
4 | 0 |  
5 | 0 |  
6 | 0 |  
7 | 0 |  
8 | 0 |  
9 | 0 |  

When the RS485 is hooked up and the heater settings are as such that the OCP thinks there is a heater on the bus this conversation between the OCP and the heater happens every few seconds.  We have tried not sending 114 messages for 10 seconds and the control is returned to the heater.

jwtaylor310, 2021-09-07:
Thanks, rstrouse.  That helps a lot!

The messages I captured are consistent with your documentation.  They were being sent several times each minute and indicated that the OCP was sending an 'Off' message and the Ultratemp responded with a status of 'Off' with Offset (?) of 1.  RS485 for the Heat Pump is currently disabled and, as expected, the messages are no longer appearing on the RS485 buss.

On my system when the Action 144 messages are being sent, the Ultratemp does  not disable its front panel or switch modes.  But that's probably an issue unique to this system since it was field modified by the installer to allow the OCP to control the heat pump without using RS485.   Accordingly, I am closing this issue with thanks to you and tagyoureit for all of your help.

closed 2021-09-07 by jwtaylor310

rstrouse, 2021-09-07:
I also created a quick start if you want to play around with Message Manager.  This works for every message exchanged between your EasyTouch and the equipment on the bus.

tagyoureit, 2021-09-07:
I must've overlooked this previously because I missed we already decoded these for the Intellicenter Ultratemp issues.  However, I cleaned up a few things in the code because 114 was pointing to the wrong code and 115 was being ignored.

jwtaylor310, 2021-09-07:
tagyoureit - thanks for the update.  Looks like at least some good came of the exercise!

rstrouse: Thanks for the link to the Message Manager.  It looks like a very slick tool ... particularly the 'show only changes' option which should help identify exactly what a particular action request does.  I'm working on a utility to download the current OCP configuration to a file so it can later be uploaded to reset the system to a previous configuration.  The Message Manager will be very helpful when testing time comes!

tagyoureit, 2021-09-07:
You do know we save all the config and state information in the data directory, right?  We have kicked around the idea of backing up/restoring from this data and we have all the hooks to write the information.  It probably wouldn't take very much for use to actually implement it.  Do you find the need for backup/restore often or is this just for a doomsday scenario?

jwtaylor310, 2021-09-07:
I did not know that you have everything stored in the data directory.  It would be great if you could add a 'save/restore' option and based on what I've seen in your programs your solution is likely to be a lot slicker than what I can manage.  I reconfigure my system twice a year to go from cooling mode to heating mode and it would be very helpful if I could just pull up a previous configuration and send it to the OCP.  Also, my controller got wacked by lightning earlier this summer and it was a pain to reenter all of the configuration information.  Neither the installer nor I could remember exactly what had been done to get the Ultratemp and solar to coexist and it took a couple of tries to get all of the settings right.  The ability to do a configuration restore from a file would have made that much easier.

tagyoureit, 2021-09-07:
Could you open a new issue/feature request for it and link here?  I don't want to belabor this issue but let's start anew.  Thx!