tagyoureit/nodejs-poolController #341: would like to be able to chose the units for temperature display

MichelMoriniaux, 2021-09-06:
**Describe the bug**
would like to be able to configure Centigrades as temperature displayed and used throughout.
Cannot find where to configure the units to use for temperature display and processing: F seems to be the norm though there are helper functions throughout the code for K C and F. 
using the API I tried PUT '{"tempUnits": 4}' to /config/general or to /config/tempSensors but it did not change anything.

**To Reproduce**
everywhere in dashboard temperatures are reported in F, no toggle to change in configuration pane
modify poolConfig.json with {"pool":{"options":{"units":4}}} has no effect
poolState.json always displays {"temps":{"units":{"val":0}}}

**Expected behavior**
either one of these:
- be able to configure the units for temperature in the GUI
- be able to configure the units for temperature via the API
- be able to configure the units for temperature via the config file

**Screenshots**
N/A

**Packet Capture**
N/A

**Pool Equipment**
 - Controller: Nixie
 - Controller Model: Shared Body
 - Controller Firmware Version: 7.4
 - Pump(s) manufacturer and model: IntelliFlow 2 VS
 - Chlorinator: Aquarite T15
 - Heater(s): Hayward H300
 - Chemical controller: N/A
 - Valves: H24
 - Any other relevant equipment: raspi, 8relay hat, industrial rs485

rstrouse, 2021-09-15:
Pull njsPC and dashPanel you will find another option under Timezone & Locality to set the system units.

[image]

MichelMoriniaux, 2021-09-28:
Thanks so much for this feature! it works perfectly!

closed 2021-09-28 by MichelMoriniaux