tagyoureit/nodejs-poolController #342: Enhancement Request - Add ability to save and restore Pentair controller configuration

jwtaylor310, 2021-09-08:
This is a request that you consider adding a feature to download and save the Pentair controller configuration to a local file and later be able to upload that file to restore the system to the previous configuration.  If possible, all configuration information would be stored in a single file which could then be moved or renamed to allow multiple configuration 'snapshots' to be managed.

This was originally discussed here: https://github.com/tagyoureit/nodejs-poolController/issues/340#issuecomment-914670829

Thank you!

rstrouse, 2021-09-28:
We are getting close.  The system is now able to store backups and restore them to the Pentair controller.

jwtaylor310, 2021-09-29:
That's great news.  I will download the new code and install it this afternoon.  Thanks for your work on this!

jwtaylor310, 2021-09-29:
I installed the new version and did a backup followed by a restore and it 'mostly' worked.  It looks like the core backup/restore function works  and the couple of bugs I ran into appear to be minor.  I had to restart the poolController program several times during the initial configuration and that appears to have 'healed' the problems.  I don't know if you want feedback yet or are still in the development mode.  If you are looking for feedback, here's my initial experience:

I had several problems during the first-time run of the backup program.  In particular, the first time I applied an automatic schedule the web interface locked up.  Restarting poolController fixed the problem.  This appears to have been caused by an undefined 'length' variable.

My first restore failed with a 'message aborted' error.  I restarted poolController and tried restoring the same file a second time and it worked.

I ran into additional instances of 'length' being undefined.  One example is a 'Stack Trace' error when entering the Settings window.  I haven't been able to determine a consistent cause of this.  In each case restarting poolController fixed the problem.

A copy of the console display of the poolController program showing the bugs follows:

==========================

First time configuration of automatic backups.  Schedule set to every 0 days, 1 hour.  Immediately after applying, the web display locked up and the following appeared in the poolController console:
[9/29/2021, 5:06:37 PM] info: [5:06:37 PM] 192.168.80.55 PUT /app/config/options/backup {"automatic":true,"interval":{"days":0,"hours":1},"keepCount":5,"njsPC":true}
[9/29/2021, 5:06:37 PM] info: Last auto-backup 2021-09-29T17:04:50.362-0400 Next auto - backup 2021-09-29T18:04:50.362-400
[9/29/2021, 5:06:43 PM] info: [5:06:43 PM] 192.168.80.55 GET /app/config/options/backup?null {}
[9/29/2021, 5:06:43 PM] error: Cannot read property 'length' of undefined
[9/29/2021, 5:06:54 PM] info: [5:06:54 PM] 192.168.80.55 GET /state/appVersion?null {}

First attempt to restore a backup.  Failed with the following.  Second attempt completed without error.  
[9/29/2021, 5:32:31 PM] info: [5:32:31 PM] 192.168.80.55 PUT /app/restore/file {"filename":"2021-09-29_17-25-52.zip","options":{"version":"1.00","name":"Automatic Backup","backupDate":"09/29/2021 05:25:52pm","njsPC":true},"filePath":"/home/pi/NodeJS/nodejs-poolController/backups/2021-09-29_17-25-52.zip"}
[9/29/2021, 5:32:31 PM] info: Begin Restore njsPC
[9/29/2021, 5:32:35 PM] warn: Message aborted after 4 attempt(s): 165,30,16,33,139,5,20,0,93,0,0,1,245 
[9/29/2021, 5:32:35 PM] error: Error restoring features: Message aborted after 4 attempt(s): 165,30,16,33,139,5,20,0,93,0,0,1,245 
[9/29/2021, 5:32:35 PM] info: End Restore njsPC

Intermittent "Stack Trace" error when entering Settings window.  Resolved by restarting poolController.
9/29/2021, 5:43:01 PM] info: [5:43:01 PM] 192.168.80.55 GET /app/config/log {}
[9/29/2021, 5:43:01 PM] info: [5:43:01 PM] 192.168.80.55 GET /config/all?null {}
[9/29/2021, 5:43:01 PM] info: [5:43:01 PM] 192.168.80.55 GET /app/config/options/backup?null {}
[9/29/2021, 5:43:01 PM] error: Cannot read property 'length' of undefined
[9/29/2021, 5:43:02 PM] info: [5:43:02 PM] 192.168.80.55 GET /state/appVersion?null {}

rstrouse, 2021-09-30:
Can I get you to attach your backup file.

jwtaylor310, 2021-09-30:
As requested:

[2021-09-29_17-25-52.zip](https://github.com/tagyoureit/nodejs-poolController/files/7261430/2021-09-29_17-25-52.zip)

rstrouse, 2021-09-30:
The message that is being rejected from EasyTouch is for the Aux Extra circuit.  Does that actually exist on your panel?  The .length error is because it is looking for external servers to back up and restore.  This means that it was waiting to timeout but eventually it would have given you back control. You have no external servers so I will make the code ignore them if they do not exist.

jwtaylor310, 2021-09-30:
Nope, I don't have an Aux Extra circuit:  [image]

I currently have the auto backup interval set for 12 hours and the program has successfully created the second automatic backup.  Looking good! 

Where is the description of the backup being stored (i.e, the information entered in the 'Name' field)?  The files in the backups folder are named using timestamps but I can't find the descriptive names.

rstrouse, 2021-09-30:
I believe that ends up on the features tab.

The name is stored in options.json.

rstrouse, 2021-09-30:
I just checked in some fixes for the restore and added logging so that I can see what errors are being output by the panel during the restore.  If you restore please upload the restoreLog(<current date>).log from the logs directory.

jwtaylor310, 2021-10-01:
I did a clean reinstall of both the Controller and the dashPanel.  This time no errors when I configured and ran a manual backup and then set up an automatic schedule.  

I then tried to restore the manual backup and got another 'length undefined' error:

Init state for Pool Controller
[10/1/2021, 11:31:42 AM] info: The current git branch output is master
[10/1/2021, 11:31:42 AM] info: The current git commit output is ab95b2243e75d84cf730af38419e75cae7668209
[10/1/2021, 11:31:42 AM] info: Starting up SSDP server
[10/1/2021, 11:31:42 AM] info: Checking njsPC versions...
[10/1/2021, 11:31:42 AM] info: Starting Pool System easytouch
[10/1/2021, 11:31:42 AM] info: Auto-backup initialized Last Backup: 1969-12-31T19:00:00.000-0500
[10/1/2021, 11:31:42 AM] info: Server is now listening on 0.0.0.0:4200
[10/1/2021, 11:31:42 AM] info: Net connect (socat) connected to: 172.16.31.241:8899
[10/1/2021, 11:31:42 AM] info: Net connect (socat) Connection connected
[10/1/2021, 11:31:42 AM] info: Net connect (socat) ready and communicating: 172.16.31.241:8899
[10/1/2021, 11:31:43 AM] info: New socket client connected mmMZADBhLWpi4uR9AAAB -- 172.16.31.151
[10/1/2021, 11:31:43 AM] info: [11:31:43 AM] 172.16.31.151 GET /state/all?null {}
Pentair EasyTouch System Detected!
[10/1/2021, 11:31:43 AM] info: Found Controller Board EasyTouch2 8
[10/1/2021, 11:31:43 AM] info: Requesting easytouch configuration
sendRS485PortStats set to false
[10/1/2021, 11:31:58 AM] info: EasyTouch system config complete.
[10/1/2021, 11:31:58 AM] info: Initializing Nixie Controller
[10/1/2021, 11:31:58 AM] info: Nixie Controller Initialized
[10/1/2021, 11:32:15 AM] info: [11:32:15 AM] 192.168.80.55 GET /state/all?null {}
[10/1/2021, 11:32:15 AM] info: New socket client connected 7H_CMxq7AwlLvuy-AAAD -- 192.168.80.55
[10/1/2021, 11:32:15 AM] info: [11:32:15 AM] 192.168.80.55 GET /state/all?null {}
[10/1/2021, 11:32:25 AM] info: [11:32:25 AM] 192.168.80.55 GET /app/config/log {}
[10/1/2021, 11:32:25 AM] info: [11:32:25 AM] 192.168.80.55 GET /state/appVersion?null {}
[10/1/2021, 11:32:25 AM] info: [11:32:25 AM] 192.168.80.55 GET /app/config/options/backup?null {}
[10/1/2021, 11:32:25 AM] info: [11:32:25 AM] 192.168.80.55 GET /config/all?null {}
[10/1/2021, 11:32:37 AM] info: [11:32:37 AM] 192.168.80.55 GET /state/all?null {}
[10/1/2021, 11:32:38 AM] info: New socket client connected 3m76eNIPU0U-mEnUAAAF -- 192.168.80.55
[10/1/2021, 11:32:38 AM] info: [11:32:38 AM] 192.168.80.55 GET /state/all?null {}
[10/1/2021, 11:32:39 AM] info: [11:32:39 AM] 192.168.80.55 GET /app/config/log {}
[10/1/2021, 11:32:39 AM] info: [11:32:39 AM] 192.168.80.55 GET /state/appVersion?null {}
[10/1/2021, 11:32:39 AM] info: [11:32:39 AM] 192.168.80.55 GET /config/all?null {}
[10/1/2021, 11:32:39 AM] info: [11:32:39 AM] 192.168.80.55 GET /app/config/options/backup?null {}
[10/1/2021, 11:32:44 AM] info: [11:32:44 AM] 192.168.80.55 GET /app/config/options/backup?null {}
[10/1/2021, 11:32:59 AM] info: [11:32:59 AM] 192.168.80.55 PUT /app/config/createBackup {"name":"Manual Backup #1","njsPC":true,"automatic":false}
[10/1/2021, 11:33:06 AM] info: [11:33:06 AM] 192.168.80.55 GET /app/config/options/restore?null {}
[10/1/2021, 11:33:52 AM] info: [11:33:52 AM] 192.168.80.55 PUT /app/config/options/backup {"automatic":true,"interval":{"days":0,"hours":1},"keepCount":5,"njsPC":true}
[10/1/2021, 11:33:52 AM] info: Last auto-backup 2021-10-01T11:32:59.456-0400 Next auto - backup 2021-10-01T12:32:59.456-0400
[10/1/2021, 11:34:04 AM] info: [11:34:04 AM] 192.168.80.55 PUT /app/config/options/backup {"automatic":true,"interval":{"days":0,"hours":12},"keepCount":5,"njsPC":true}
[10/1/2021, 11:34:04 AM] info: Last auto-backup 2021-10-01T11:32:59.456-0400 Next auto - backup 2021-10-01T23:32:59.456-0400
[10/1/2021, 11:34:10 AM] info: [11:34:10 AM] 192.168.80.55 GET /app/config/options/restore?null {}
[10/1/2021, 11:34:10 AM] error: Cannot read property 'length' of undefined

I restarted the program and was then able to do a restore without any problems.  Here's the log for the restore.  It doesn't show much because there were no errors! 

[restoreLog(2021-10-01_11-56-07).log](https://github.com/tagyoureit/nodejs-poolController/files/7268518/restoreLog.2021-10-01_11-56-07.log)

jwtaylor310, 2021-10-01:
Also...you have been doing a lot of work on this and I really appreciate it.  Is there a link somewhere that I can use to make a $ donation to support your efforts?

rstrouse, 2021-11-22:
I am going to close this.  If you have any issues with the functionality.  Please open a new issue.

closed 2021-11-22 by rstrouse