tagyoureit/nodejs-poolController #343: Potential Schedule Limit issue with EasyTouch 2 4?

johnny2678, 2021-09-16:
I noticed that my schedules in dash don't match my schedules in ScreenLogic.

[image]

Attaching 2 packet captures.

1st one captures the startup sequence:
[replay (24).zip](https://github.com/tagyoureit/nodejs-poolController/files/7179754/replay.24.zip)

2nd one captures a stack trace error I get when adding schedules through dash panel (settings -> Schedules tab -> Add Schedule).
[replay (25).zip](https://github.com/tagyoureit/nodejs-poolController/files/7179759/replay.25.zip)

As always, thx! and let me know if I can provide more information.

rstrouse, 2021-09-16:
Very interesting.  I see 12 active schedules if you include the egg timers and this is the max advertised schedules for an ET2-4.  An egg timer in ET occupies a schedule slot.

In addition to the 5 schedules that I see above there are the following egg timers.  This makes 12 schedule slots leaving out the Pool Lowest schedule.
2: Lights - 4 hours
3: Air Blower - 1hour
4: High Speed - 45min
1: Spa - 2 hours
6: Pool - 16 hours
15: Pool High - 45min
16: Pool Higher - 45min

Can you verify the egg timers that you have set up with EasyTouch?

johnny2678, 2021-09-16:
Seems like an egg timer is set up for each circuit / feature.  I can modify, but not delete.

[image]

rstrouse, 2021-09-16:
The default is 12 hours so Pool Low, Pool Lower, Pool Lowest, and AuxEx are not occupying a slot so in essence they are deleted.  It's a bit goofy but If it is default (12 hours) then it will not occupy a schedule slot.  

I did see something on TFP a while back that said it will start ignoring the schedules if you define more than 12 schedules by setting the egg timers and adding schedules.  There are only 12 slots for EasyTouch but the programming is the same between IntelliTouch and EasyTouch which has 99 slots.

It's crazy but it allows you to define them but doesn't use them.  So if you have 7 egg timers and 6 schedules it will ignore one of your schedules or it will ignore the egg timer.  It simply depends on which one was added first.

johnny2678, 2021-09-16:
so, a good test would be to set some egg timers back to 12 hours and see if the new schedules are picked up?

rstrouse, 2021-09-16:
Yessir but you may need to re-add the last schedule which at this point looks like the one for Pool Lowest.  You may want to delete it after you clear out a slot on the egg timers and re-add it.  I am pretty sure that while the panel will store it the schedule won't actually execute.  If it did then I would be able to expand your ET to 99 schedules.

johnny2678, 2021-09-16:
Bingo! 
[image]

Once again you've figured it out, even though njspc was working as intended.  Only suggestion would be to maybe add a catch before that stacktrace error... something like, "couldn't add schedules, perhaps the max # of schedules / egg timers for your OCP has been reached?  Try removing a schedule / egg timer and adding your schedule again."

rstrouse, 2021-09-16:
I'll have a look and see what we can do.  I don't think you can exceed the max schedules on a Pentair controller but you apparently can do it in ScreenLogic.   The worst part is that they simply do not work if you manage to add too many.  At some point we may allow this to be expanded but control them ourselves.  Currently, the Nixie controllers allow an unlimited number of schedules (actually it is limited but the number is 2,147,483,647).  Mathematically, you would have to have ~1.5 million features and circuits to schedule every minute.

johnny2678, 2021-09-16:
pffft, in software sales that counts as "unlimited" 😂

thanks again!

closed 2021-09-20 by johnny2678