tagyoureit/nodejs-poolController #344: Unable to see Pool as Main Body

vj950, 2021-09-18:
Main body shows up as and controls "Solar" instead of "Pool" (Please see the screenshot. I am also uploading my mobiletouch2 screen pictures for comparison). It seems that Low-Temp mode on mobiletouch2 is detected as "solar" on njPC. When Solar is on the pool pump shuts down, and when I turn Pool on from mibiletouch2, Solar shuts down. Therefore, I cannot control pool pump or heater. Using solarheater only will not work where I live. The i5 expansion board is also not detected and so are its aux circuits (except Aux 1 which is Fountain pump and Booster Pump).

**Pool Equipment**
 - Controller: IntelliTouch
 - Controller Model: i9+3S connections Intelliflow VS (Address 1) at filter pump, Gas Heater, Solar
         Aux circuits: 1.Cleaner (Polaris) 2.Pool Lights 3.Fountain Lights 4.Fountain Fiber-optic, 5.Pool Fiber-optic 6.Outlet 7.Empty 8.Empty
 - Expansion Module i5 connections: None on filter pump
        Aux Circuits: 1.Whisperflow pump with Intelliflow VS (Address 2) as Booster pump 2.Empty 3.Cabana light 4.Back Light
 - 
 - Controller Firmware Version: unknown
 - Pump(s) manufacturer and model: IntelliFlow VS
 - Chlorinator: Not connected
 - Heater(s): gas, solar (not physically connected)
 - Chemical controller: Not present
 - Valves: Intellivalve two are present. 
 - Any other relevant equipment:

Pi Zero W connected by RS485 to USB converter https://www.amazon.com/gp/product/B081MB6PN2/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
[image]
[image]
[image]
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/7190604/replay.zip)

tagyoureit, 2021-09-22:
Can you first confirm which expansion card you have?  I sent your status packet to my ScreenLogic and it reported back an i10x.  I wouldn't doubt if SL was wrong, but please do confirm.

[image]

vj950, 2021-09-24:
The main panel states i9+3S on it, while the second panel states i5x. i5x is connected to main by com wire. Attaching the pictures of the panels since I am confused by Pentair's naming (intellitouch vs easytouch, i5x vs i5s ect).
[image]
[image]

tagyoureit, 2021-09-27:
The first step here is done.  njsPC should (to the best I can tell) determine when expansion cards are present.  Unfortunately, ScreenLogic defaults to i10x/i5x/i5x and I can't see exactly how to infer if the boards are i5x/i10x.  Hence, I defaulted to starting with i5x and the boards can be changed if they are wrong.

[image]

Next step is to fix the low-temp/hi-temp modes...

vj950, 2021-09-28:
That seems to have done the trick. I can see all my i5x aux circuits now.

tagyoureit, 2021-09-28:
Can you clarify your two comments:

> Controller Model: i9+3S connections Intelliflow VS (Address 1) at filter pump, Gas Heater, Solar

and

> Heater(s): gas, solar (not physically connected)

It's not clear if your solar is controlled by Intellitouch or not.

And in your replay that you attached, did you by chance change the heat modes/set points in each mode?  If not, if you could redo the capture and document the changes for:
1. Lo-temp cycle through heat mode and set points (eg turn up set point so heater/solar turns on; then lower so it turns off) on Intellitouch controller
2. Change to hi-temp, cycle through heat modes and set points on intellitouch controller
3. Repeat 1 and 2 but from the njsPC dashPanel so I can see where the packets differ.
Please write down the approx times/actions that you took so I can correlate to the packets captured.

vj950, 2021-09-29:
Clarification 1. I was just describing the connections of my i9+3S; as seen on my panel (picture of which I uploaded in my later post)
Clarification 2. I think that Intellitouch is controlling Solar circuit electronically only. The wiring connection to the actuator for the valve is present. However, Solar piping is intentionally physically turned off due to leakage, so there is no waterflow in it. The actuator is not physically connected to the valve either. I am also not sure if actuator receives power supply or not.

The intelliflow VS (address 1) is the filter pump of the "Pool" circuit, while there does not appear to be any filter pump connected to "Solar" circuit. Therefore, when solar is turned on from controller or njpc, "Pool" turns off as seen on controller, the intelliflow VS turns off, water does not flow through the piping. When I turn "Solar" off from njpc or controller, the "Pool" does not automatically turn on. Since njpc does not see "Pool" circuit, only way to turn it on is through the Controller. On the other hand, when I turn on the "Pool" from controller, the "Solar" turns off automatically. 

The Low Heat mode is for "Solar" circuit. When I set Low Heat to Gas Heater, and turn the "Solar" circuit on (either controller or njPC), the High Heat mode temp automatically gets set at 100 degrees and its heat mode gets set to "off". Also, the heater ignites, and shuts off immediately for protection since there is no actual water flowing through the heater when "Solar" is used. 

High Heat mode is for the "Pool" circuit, and it is only accessible through the controller. Any function related to "Pool" circuit is only accessible through the controller and not njPC.

vj950, 2021-09-29:
attaching my capture log and timestamps. all times are PM
[image]
[replay(1).zip](https://github.com/tagyoureit/nodejs-poolController/files/7248856/replay.1.zip)

tagyoureit, 2021-10-06:
After some deliberation, we have decided that these are nearly identical to a shared body system sans the intake/return valves.  Pull the latest and remove your data/pool*.json files so the app can rebuild itself.  

Let us know if it works.

vj950, 2021-10-07:
This partly works. I can now see the "Pool" as aux circuit and can turn it on and off and set a schedule. However, the controller is still recognized as single body (I think). 

Home screen is still unchanged. I don't see the 2nd body to control. As the "Solar" is set as "Master Pool" and "Pool" is set as "Master Spa", I can only control heater for "Solar". These seem to be hard-coded or are set by "service-remote" that I don't have access to (I have not been able to figure out to flip these using the remote)

attaching pictures to clarify.

Please let me know if you need me to upload the log.

[image]
[image]
[image]
[image]

vj950, 2021-10-07:
That did the trick!!! Everything in my system works now. Thank you very much!!

[image]

tagyoureit, 2021-10-08:
You can rename "Pool" and/or "pool" (above the temps) in the settings menu, btw.  But glad it's all working now!

closed 2021-10-08 by tagyoureit