tagyoureit/nodejs-poolController #345: Issue with install of 7.4 as reported in poolController Gitter

sparment, 2021-09-19:
From the log:

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.15
3 info using node@v12.22.6
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle nodejs-poolcontroller@7.4.0~prestart: nodejs-poolcontroller@7.4.0
6 info lifecycle nodejs-poolcontroller@7.4.0~start: nodejs-poolcontroller@7.4.0
7 verbose lifecycle nodejs-poolcontroller@7.4.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle nodejs-poolcontroller@7.4.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/pool-controller/nodejs-poolController/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle nodejs-poolcontroller@7.4.0~start: CWD: /home/pi/pool-controller/nodejs-poolController
10 silly lifecycle nodejs-poolcontroller@7.4.0~start: Args: [ '-c', 'npm run build && node dist/app.js' ]
11 silly lifecycle nodejs-poolcontroller@7.4.0~start: Returned: code: 1  signal: null
12 info lifecycle nodejs-poolcontroller@7.4.0~start: Failed to exec start script
13 verbose stack Error: nodejs-poolcontroller@7.4.0 start: `npm run build && node dist/app.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:314:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:314:20)
13 verbose stack     at maybeClose (internal/child_process.js:1022:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)
14 verbose pkgid nodejs-poolcontroller@7.4.0
15 verbose cwd /home/pi/pool-controller/nodejs-poolController
16 verbose Linux 5.10.60-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v12.22.6
19 verbose npm  v6.14.15
20 error code ELIFECYCLE
21 error errno 1
22 error nodejs-poolcontroller@7.4.0 start: `npm run build && node dist/app.js`
22 error Exit status 1
23 error Failed at the nodejs-poolcontroller@7.4.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

tagyoureit, 2021-09-19:
Let's take it one step at a time...
1. From the pool controller directory, run `rm -rf node_modules`.  This will delete any existing modules.
2. Run `npm i` again to reinstall the modules
3. Run `npm run build` to compile typescript to javascript
4. Run `node dist/app.js` to launch the app.

It almost seems like npm is still running when the app is trying to start up.

sparment, 2021-09-19:
Okay, I have pool controller up and running with 7.4. I followed the steps above. I also found a node_modules directory in my home directory and removed that as well. Then followed your instructions above.

My issue now is with DashPanel! I'll keep trying to figure it out and if I can't I'll open an issue there. Thanks!

closed 2021-09-19 by sparment