tagyoureit/nodejs-poolController #346: Featrue request: report Pentair Pool cover expansion card - open/closed status

lockeandev, 2021-09-25:
Following up on Gitter chat request to report auto pool cover open closed status from pentair pool cover expansion card. Packet capture attached. for the attached capture I open/closed the cover twice.

[Cover open close packet capture replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/7228934/Cover.open.close.packet.capture.replay.zip)

rstrouse, 2021-09-25:
Ok so I think I have it but the output above is like drinking out of a fire hose.  Can I get you to create a file a different way?

I only want one specific message and this is the 204 message.  So first change the settings in the logging tab to the following.  To get to the Broadcast Actions dialog click the little filter icon on the broadcast button.
[image]

Once you have that done then click on the Log to "File" box under the packets group and cycle the cover a couple of times.  This will create a file in the logs directory for njsPC.  Upload that log here.

lockeandev, 2021-09-25:
Ok, I tried that. Take a look.
[packetLog(2021-09-25_02-01-39).log](https://github.com/tagyoureit/nodejs-poolController/files/7229989/packetLog.2021-09-25_02-01-39.log)

rstrouse, 2021-09-25:
That was helpful.  It appears you only have one cover active.  Is that correct?  Too bad you don't have 2 because somewhere there is a bit that flips for the second cover.  For now I will assume it to be the second bit on byte 30.  Please review my assessment and let me know if I got it right.  The key is that you started capturing with it closed.

I assume it started with the cover closed then at 9:44 and 55 seconds you opened it as indicated by byte(30)
[image]
Then at 9:45 and 10 seconds you closed it again as indicated by byte(30)
[image]
Then at 9:46 and 16 seconds you opened it again as indicated by byte(30)
[image]
Finally at 9:46 and 40 seconds you closed it again as indicated by byte(30)
[image]

lockeandev, 2021-09-25:
That appears correct. I did start with it closed. And I open and closed it
twice as you described. If you want, I can hook my cover up to the second
port and see if registers the second bit?

rstrouse, 2021-09-25:
Yes please hook it up to the second port and create the same output.   This would put it to bed for good.

rstrouse, 2021-09-26:
A couple more things that aren't really clear from the docs.  When you assign an effective circuit, do you get to tell it whether the circuit should be off or on when closed.  Or is it just any circuit you add is off when the cover is closed.  Finally, how many circuits can you add?  There appears to be enough space for 10 circuits on the message but that is a very weird number for Pentair.

lockeandev, 2021-10-04:
Sorry for the delayed reply. I haven't had a chance to Hook my cover up to the 2nd port. I will try to do so next weekend. But I did look at the cover configuration options you asked about. You elect which circuits you want off when the cover is closed. And the panel limits you to 10 circuits but doesn't tell you it's limiting you. It just won't let you select more than 10.  And you can set it to turn the intellichlor off when closed, or  a % when closed. The descriptions of the functions are confusing on the panel. Personally I don't mess with the intellichlor cover settings and just let schedules handle that. But turning circuits off when the cover is closed is invaluable as it keeps anyone from pumping water out of the pool and onto the cover.

rstrouse, 2021-10-04:
Thanks.  Just to be clear you cannot open or close the cover from IntelliCenter.  Is that correct?

lockeandev, 2021-10-04:
Correct. Intellicenter has no connection to the cover motor. There is a rotary encoder installed on the cover that reports to intellicenter the open or closed status. That's it. Here is a document from Pentair with more info about my particular cover and integrating it with the cover expansion card in case you are interested
[IC POOL COVER INST DRAFT 4-2-19.pdf](https://github.com/tagyoureit/nodejs-poolController/files/7279597/IC.POOL.COVER.INST.DRAFT.4-2-19.pdf)
.

lockeandev, 2021-10-11:
Ok, I finally got a chance to try hooking my cover up to the 2nd port on the expansion card. 

I started the logs and then cycled the cover closed then open twice.

Here is the packet log.

[packetLog(2021-10-10_02-01-41).log](https://github.com/tagyoureit/nodejs-poolController/files/7318850/packetLog.2021-10-10_02-01-41.log)

While cycling the cover, dash panel did not report the status of the cover while it was connected to the 2nd port.

rstrouse, 2021-10-19:
Thanks for your help I am going to close this for now.  If you need anything else open another issue.