tagyoureit/nodejs-poolController #348: Testing SWG auto-dosing

johnny2678, 2021-09-27:
@tagyoureit - I turned off my SWG dosing routine and turned on the SWG dosing settings in dash.  Will use this thread for questions - status updates

[image]

First off - pool FC as of yesterday afternoon was 5.5 ppm

CYA - 20
TA - 70
CH - 400
pH - 7.7 via Atlas Consumer probe (or 7.4 according to Taylor kit - still working that out)
Salt - 3600 via Taylor salt kit

First question:

ORP setpoint is 620 and it's currently reading 601.4.  Should it be doing something now?

[image]

johnny2678, 2021-09-27:
wait a minute... just saw that I didn't have the SWG model set in the Chlorinator settings.  Set that to IC40 and got this... looks like it's working!
[image]

Will track throughout the day

johnny2678, 2021-09-27:
Had a weird hiccup during the 5 minute cycle above where the dosing stopped counting up 20 seconds into the 5 minute cycle.  Checked the logs and it said 
`error: Error polling Chem Controller - EquipmentNotFound: Please specify a chlorinator model to allow Nixie to calculate chlorine demand`

..which was odd because I had just set it.  I set it again and it looks like it has stuck this time.  At least it finished the cycle.

johnny2678, 2021-09-27:
hmm, lost the SWG model again...

```
1|poolController  | [9/27/2021, 10:36:29 AM] info: [10:36:29 AM] 192.168.5.140 GET /state/all/ {}
1|poolController  | [9/27/2021, 10:36:31 AM] error: Error polling Chem Controller - EquipmentNotFound: Please specify a chlorinator model to allow Nixie to calculate chlorine demand
1|poolController  | [9/27/2021, 10:36:41 AM] error: Error polling Chem Controller - EquipmentNotFound: Please specify a chlorinator model to allow Nixie to calculate chlorine demand
1|poolController  | [9/27/2021, 10:36:44 AM] info: [10:36:44 AM] 192.168.5.140 GET /state/all/ {}
1|poolController  | [9/27/2021, 10:36:46 AM] info: [10:36:46 AM] 192.168.5.146 GET /config/all?null {}

```

willl try to restart njspc to see if it sticks.

johnny2678, 2021-09-27:
yea, might have to table this until I can get the SWG model to stick in settings.  Hope this capture helps.  Let me know if you need another one.
[replay (26).zip](https://github.com/tagyoureit/nodejs-poolController/files/7237312/replay.26.zip)

tagyoureit, 2021-09-27:
@drjeff said he lost the model, as well.  Need to see why that's happening and will dig into your logs.

The current calcs are:
1. -20 to 50 demand - chlor will run at a sliding scale between 0-15 mins.
2. <-20 demand (extra surplus) - chlor will just be in mixing mode
3. >50 demand - chlor will run at 100%

It will appear "paused" if...
4. it is supposed to be dosing but the chlor is reporting 0%
5. chlor is supposed to be mixing but it is reporting 100%
6. Nixie has finished a dose/mix and is waiting for the next time to process (<10s).

Scenarios 4 & 5 above may take up to 30s to resolve themselves because the OCP only sets the chlor every 30s.  

The 'no chlor response in 7s' just indicates a lost packet.  It shouldn't have any effect on overall performance.

rstrouse, 2021-09-27:
I think I know.  It has to do with the 3 message coming back from the Chlorinator.  I am running tests now but if you pull I added a check to make sure it doesn't overwrite the model.

tagyoureit, 2021-09-27:
Oh, nice.  🤞  

I also think we need to normalize the chlorinatorType with the chlorinatorModel maps.  We shouldn't need both and one can be set independent of the other but they should be related.  But that's a different (internal) issue.

DrJeff, 2021-09-27:
Looks like the chlorinator is keeping and it set the proper one off the back, for me so far.

johnny2678, 2021-09-28:
Yep, the SWG sticks in the settings now.

Day 1 wrap - partial day so just leaving here as a data point.

reminder, my ORP setpoint is 620 and my probe was reading between 590-600.

After enabling SWG dosing (and getting it to stick), the SWG alternated between setpoints of 0 and 100, ~5 minutes on / ~10 minutes off:
[image]

ORP reading fell slight to the ~580 range during this time (to the right of the red line when I turned on SWG dosing).
[image]

Note:  this was not unexpected.  SWG dosing has never had an immediate effect on ORP in my pool.  But usually increases within a day or two.  Will continue to keep an eye on it and take some FC readings in the morning.

johnny2678, 2021-09-28:
Ok, ORP is reading slightly lower today, ~ 565.  FC is reading slightly lower too at 5.5.  Let's see how nsjpc with a full day.
[image]

Maybe related, I keep getting SWG comms dropouts during dosing cycles.  I don't think this is caused by njspc, maybe flow sensor going bad?  or maybe just normal operation?  It always seems to resolve itself within a few seconds and then dosing resumes.

```
1|poolController  | [9/28/2021, 9:06:00 AM] info: Chem Controller orp chlorinated 0.010423lbs of 0.014583lbs - 4min 16sec remaining
1|poolController  | [9/28/2021, 9:06:01 AM] info: Chem Controller orp chlorinated 0.010441lbs of 0.014583lbs - 4min 15sec remaining
1|poolController  | [9/28/2021, 9:06:02 AM] info: Chem Controller orp chlorinated 0.010459lbs of 0.014583lbs - 4min 14sec remaining
1|poolController  | [9/28/2021, 9:06:04 AM] info: Chem Controller orp chlorinated 0.010477lbs of 0.014583lbs - 4min 13sec remaining
1|poolController  | [9/28/2021, 9:06:04 AM] info: Chem Controller orp chlorinated 0.010483lbs of 0.014583lbs - 4min 13sec remaining
1|poolController  | [9/28/2021, 9:06:05 AM] info: Chem Controller orp chlorinated 0.0105lbs of 0.014583lbs - 4min 11sec remaining
1|poolController  | [9/28/2021, 9:06:06 AM] info: Chem Controller orp chlorinated 0.010519lbs of 0.014583lbs - 4min 10sec remaining
1|poolController  | [9/28/2021, 9:06:07 AM] info: Chem Controller orp chlorinated 0.010537lbs of 0.014583lbs - 4min 9sec remaining
1|poolController  | [9/28/2021, 9:06:08 AM] info: Chem Controller orp chlorinated 0.010555lbs of 0.014583lbs - 4min 8sec remaining
1|poolController  | RS485 Stats:{ "bytesReceived": 2694453 "success": 90822, "failed": 2573, "bytesSent": 48334, "collisions": 15, "failureRate": 2.75% }
1|poolController  | RS485 Stats:{ "bytesReceived": 2694564 "success": 90824, "failed": 2574, "bytesSent": 48334, "collisions": 15, "failureRate": 2.76% }
1|poolController  | [9/28/2021, 9:06:09 AM] info: [9:06:09 AM] 192.168.5.140 GET /state/all/ {}
1|poolController  | RS485 Stats:{ "bytesReceived": 2694742 "success": 90829, "failed": 2575, "bytesSent": 48334, "collisions": 15, "failureRate": 2.76% }
1|poolController  | [9/28/2021, 9:06:10 AM] warn: Chlor dose not added because current output is not 100%
1|poolController  | [9/28/2021, 9:06:11 AM] warn: Chlor dose not added because current output is not 100%
1|poolController  | [9/28/2021, 9:06:13 AM] warn: Chlor dose not added because current output is not 100%
1|poolController  | [9/28/2021, 9:06:14 AM] warn: Chlor dose not added because current output is not 100%
1|poolController  | [9/28/2021, 9:06:15 AM] warn: Chlor dose not added because current output is not 100%
1|poolController  | [9/28/2021, 9:06:17 AM] warn: Chlor dose not added because current output is not 100%
1|poolController  | [9/28/2021, 9:06:19 AM] warn: Chlor dose not added because current output is not 100%
1|poolController  | RS485 Stats:{ "bytesReceived": 2695533 "success": 90856, "failed": 2576, "bytesSent": 48598, "collisions": 15, "failureRate": 2.76% }
1|poolController  | [9/28/2021, 9:06:21 AM] warn: Chlor dose not added because current output is not 100%
1|poolController  | [9/28/2021, 9:06:23 AM] warn: Chlor dose not added because current output is not 100%
1|poolController  | [9/28/2021, 9:06:24 AM] info: [9:06:24 AM] 192.168.5.140 GET /state/all/ {}
1|poolController  | [9/28/2021, 9:06:28 AM] warn: Chlor dose not added because current output is not 100%
1|poolController  | [9/28/2021, 9:06:28 AM] error: *Touch setChlorAsync Error: no chlor response in 7 seconds
1|poolController  | [9/28/2021, 9:06:28 AM] error: chemController.chlor.turnOn: no chlor response in 7 seconds
1|poolController  | [9/28/2021, 9:06:28 AM] error: Error starting chlorination: TimeoutError: no chlor response in 7 seconds.
1|poolController  | [9/28/2021, 9:06:29 AM] warn: Chlor dose not added because current output is not 100%
1|poolController  | [9/28/2021, 9:06:30 AM] info: Chem Controller orp chlorinated 0.010908lbs of 0.014583lbs - 3min 46sec remaining
1|poolController  | [9/28/2021, 9:06:31 AM] info: Chem Controller orp chlorinated 0.010926lbs of 0.014583lbs - 3min 45sec remaining
1|poolController  | [9/28/2021, 9:06:32 AM] info: Chem Controller orp chlorinated 0.010944lbs of 0.014583lbs - 3min 44sec remaining
1|poolController  | [9/28/2021, 9:06:34 AM] info: Chem Controller orp chlorinated 0.010962lbs of 0.014583lbs - 3min 43sec remaining
1|poolController  | [9/28/2021, 9:06:35 AM] info: Chem Controller orp chlorinated 0.01098lbs of 0.014583lbs - 3min 42sec remaining
1|poolController  | [9/28/2021, 9:06:36 AM] info: Chem Controller orp chlorinated 0.010999lbs of 0.014583lbs - 3min 41sec remaining
1|poolController  | [9/28/2021, 9:06:37 AM] info: Chem Controller orp chlorinated 0.011017lbs of 0.014583lbs - 3min 40sec remaining

```

johnny2678, 2021-09-29:
quick update:
ORP rose slowly throughout the day:
[image]

dosing was more frequent in the AM when ORP was >50 below the setpoint of 620
[image]

after ~1:30 dosing became less frequent, presumably because ORP rose to more than the threshold minus 50

looking good! will check FC and post back in the AM

johnny2678, 2021-09-30:
DAY 3 update:

FC dropped to 4 (still good)
ORP started around 60-70 from the setpoint of 620 and rose slowly throughout the day.  ended up right around the setpoint...
[image]

still looking good!

johnny2678, 2021-10-01:
something that may be of interest... I have pH dose priority turned on.  My understanding is that will pause ORP dosing (set to 0%), perform the acid dosing, and then resume the ORP dosing when the acid is done.

I was noticing some short ORP cycles and thought it might be related to pH dose priority so I followed along in the logs and ran across an unhandled rejection around the same time the acid dose was about to start:

```
1|poolController  | [10/1/2021, 8:57:13 AM] info: Chem Controller orp chlorinated 0.002765lbs of 0.005062lbs - 2min 21sec remaining
1|poolController  | [10/1/2021, 8:57:14 AM] info: Chem Controller orp chlorinated 0.002783lbs of 0.005062lbs - 2min 20sec remaining
1|poolController  | [10/1/2021, 8:57:15 AM] info: Chem Controller orp chlorinated 0.002801lbs of 0.005062lbs - 2min 19sec remaining
1|poolController  | [10/1/2021, 8:57:16 AM] info: Chem Controller orp chlorinated 0.002819lbs of 0.005062lbs - 2min 18sec remaining
1|poolController  | [10/1/2021, 8:57:17 AM] info: Chem Controller orp chlorinated 0.002838lbs of 0.005062lbs - 2min 17sec remaining
1|poolController  | [10/1/2021, 8:57:18 AM] info: Chem Controller orp chlorinated 0.002856lbs of 0.005062lbs - 2min 16sec remaining
1|poolController  | [10/1/2021, 8:57:19 AM] info: Chem acid demand calculated 4mL for 1sec Tank Level: 2.366800521770861
1|poolController  | [10/1/2021, 8:57:19 AM] info: Starting a new acid dose 4mL
1|poolController  | RS485 Stats:{ "bytesReceived": 6672268 "success": 226105, "failed": 7962, "bytesSent": 137315, "collisions": 43, "failureRate": 3.40% }
1|poolController  | [10/1/2021, 8:57:20 AM] info: Chem Controller dosed acid 0.00mL of 4mL 1sec remaining
1|poolController  | [10/1/2021, 8:57:21 AM] info: Clearing any possible orp dosing or existing mix for mixingTime: 723.759
1|poolController  | RS485 Stats:{ "bytesReceived": 6672453 "success": 226110, "failed": 7963, "bytesSent": 137393, "collisions": 43, "failureRate": 3.40% }
1|poolController  | [10/1/2021, 8:57:21 AM] info: Chem Controller dosed acid 4.53mL of 4mL 0sec remaining
1|poolController  | [10/1/2021, 8:57:22 AM] info: Dose completed 4.525mL 0 0
1|poolController  | [10/1/2021, 8:57:22 AM] info: Chem Controller begin mixing acid for 20min of 20min
1|poolController  | [10/1/2021, 8:57:23 AM] error: *Touch setChlorAsync Error: no chlor response in 7 seconds
1|poolController  | [10/1/2021, 8:57:23 AM] error: chemController.chlor.turnOn: no chlor response in 7 seconds
1|poolController  | [10/1/2021, 8:57:23 AM] error: Error starting chlorination: TimeoutError: no chlor response in 7 seconds.
1|poolController  | [10/1/2021, 8:57:24 AM] warn: Trying to stop dosing chlor but it has not yet responded.
1|poolController  | [10/1/2021, 8:57:24 AM] error: cancelDosing ORP: Already trying to stop chlor dosing.
1|poolController  | [10/1/2021, 8:57:24 AM] error: chemController.chlor dose: Already trying to stop chlor dosing.
1|poolController  | [10/1/2021, 8:57:24 AM] error: Already trying to stop chlor dosing.
1|poolController  | [10/1/2021, 8:57:25 AM] warn: Message aborted after 4 attempt(s): 165,56,16,33,217,1,0,1,232 
1|poolController  | [10/1/2021, 8:57:25 AM] error: Error requesting chlor status: Message aborted after 4 attempt(s): 165,56,16,33,217,1,0,1,232 
1|poolController  | [10/1/2021, 8:57:25 AM] error: *Touch setChlorAsync Error: Message aborted after 4 attempt(s): 165,56,16,33,217,1,0,1,232 
1|poolController  | [10/1/2021, 8:57:25 AM] error: chemController.chlor.turnOff: Message aborted after 4 attempt(s): 165,56,16,33,217,1,0,1,232 
1|poolController  | [10/1/2021, 8:57:25 AM] error: Error stopping orp dosing: Message aborted after 4 attempt(s): 165,56,16,33,217,1,0,1,232 
1|poolController  | [10/1/2021, 8:57:25 AM] error: Error initializing orp mix: Message aborted after 4 attempt(s): 165,56,16,33,217,1,0,1,232 
1|poolController  | [10/1/2021, 8:57:25 AM] error: Error mixing chemicals: Cannot read property 'timeMixed' of undefined
1|poolController  | [10/1/2021, 8:57:25 AM] info: Chem Controller begin mixing orp for 15min of 15min
1|poolController  | [10/1/2021, 8:57:26 AM] error: *Touch setChlorAsync Error: no chlor response in 7 seconds
1|poolController  | [10/1/2021, 8:57:26 AM] error: cancelDosing pH: no chlor response in 7 seconds
1|poolController  | [10/1/2021, 8:57:26 AM] error: chemController.pump dose: no chlor response in 7 seconds
1|poolController  | unhandled rejection
1|poolController  |   == Node detected an unhandled rejection! ==
1|poolController  |   no chlor response in 7 seconds
1|poolController  |   ApiError: no chlor response in 7 seconds
1|poolController  |       at Timeout._onTimeout (/home/pi/nodejs-poolController/controller/boards/EasyTouchBoard.ts:1737:28)
1|poolController  |       at listOnTimeout (internal/timers.js:554:17)
1|poolController  |       at processTimers (internal/timers.js:497:7)
1|poolController  | [10/1/2021, 8:57:26 AM] error: self.dose error in finally:
1|poolController  | [10/1/2021, 8:57:26 AM] error: no chlor response in 7 seconds
1|poolController  | (node:746) PromiseRejectionHandledWarning: Promise rejection was handled asynchronously (rejection id: 2044)
1|poolController  | [10/1/2021, 8:57:26 AM] info: [8:57:26 AM] 192.168.5.140 GET /state/all/ {}
1|poolController  | [10/1/2021, 8:57:28 AM] error: *Touch setChlorAsync Error: no chlor response in 7 seconds
1|poolController  | [10/1/2021, 8:57:28 AM] error: chemController.chlor.turnOff: no chlor response in 7 seconds
[... cut ...]
1|poolController  | RS485 Stats:{ "bytesReceived": 6674452 "success": 226180, "failed": 7967, "bytesSent": 137471, "collisions": 43, "failureRate": 3.40% }
1|poolController  | [10/1/2021, 8:58:11 AM] info: [8:58:11 AM] 192.168.5.140 GET /state/all/ {}

```

I didn't get a packet capture, unfortunately.  What ended up happening is the acid dosed as expected and the currently running ORP dosing cycle was cancel and ORP mixing started.  If this is expected, no worries.  I just thought it would resume the ORP cycle once acid dosing was complete.

@tagyoureit Let me know if I'm tracking this right...

johnny2678, 2021-10-01:
DAY 5 update:
still tracking well, the pool starts the day below the setpoint and creeps up all day until it catches up.  Was not expecting it to be so smooth.  Too lazy to take readings today but will update again with chem levels.
[image]

Coming up on the weekend so we'll see how it does when we swim and add chemicals this weekend ;)

tagyoureit, 2021-10-02:
> something that may be of interest... I have pH dose priority turned on. My understanding is that will pause ORP dosing (set to 0%), perform the acid dosing, and then resume the ORP dosing when the acid is done. This is how Pentair does their dosing if you use their dumb IntelliPH controller.
I didn't get a packet capture, unfortunately. What ended up happening is the acid dosed as expected and the currently running ORP dosing cycle was cancel and ORP mixing started. If this is expected, no worries. I just thought it would resume the ORP cycle once acid dosing was complete.
@tagyoureit Let me know if I'm tracking this right...

Yes... this is the expected behavior.  pH Dose Priority will mean the Chlor (or tank) will stop dosing so there isn't the introduction of new chlorine and acid at the same time.

Maybe @rstrouse can help me with the unhandled promise rejections?  I've tried 12 different ways from Sunday to capture those so they don't throw an error (hence the final catch-all is the text display of the rejection so the app doesn't actually crash when we move to Node v14/16 and you can't have an app with uncaught exceptions).  But it's still eluding me why they are appearing.  You can safely ignore them as I'm dealing with them and they won't affect any dosing.

I'm pleasantly surprised that the ORP is so accurate.  I really did just make a "best guess" and excited to see it working as expected!  Thanks for testing it out.

johnny2678, 2021-10-02:
> Yes... this is the expected behavior. pH Dose Priority will mean the Chlor (or tank) will stop dosing so there isn't the introduction of new chlorine and acid at the same time.

ok, thanks for confirming

DAY 6 update

[image]

Took an FC reading this AM and FC was boosted to 8.  ORP was ranging between -50/+20 of my setpoint so there was repeated dosings that averaged about 28% output for the last 2-3 days.  Today ORP is around 660 (setpoint of 620) so dosing has been off most of the afternoon.  Hopefully that will bring FC back down a bit.  Otherwise I might have to drop the ORP setpoint to 600-610.

It's going to be a long game to see how the ORP dosing algo does but if it can keep FC in the 3-6 range on average I would call that a win.  Will keep posting updates.

chem:
FC 8
CC 0
CYA 30
TA 60
CH 375
SALT 4000
TEMP 86

tagyoureit, 2021-10-02:
Remind me again what you had your chlor set at before allowing njsPC to control it.  I know you told me but too lazy to search... I mean it's too hard to find.  :)  And how did those graphs look compared to these?

johnny2678, 2021-10-02:
This graph sort of shows it.  I've been toying with the ORP setpoint (dashed red line) every 1-2 months to dial it in but I'm currently at 620 and ORP was ranging as low as 550 and as high as 700 using my dumb dosing method.  Subsequently, FC ranged from 3 to 9 in the same timeframe.

[image]

If your dosing algo can tighten up both the ORP and FC ranges, then it's a keeper.  Will take time to observe but I'm not going anywhere.

tagyoureit, 2021-10-03:
[image]
Just for comparison, here's my past 7 days.  I'll caveat this entire graph with the fact that I need to get a new ORP sensor, but even despite that demand hardly moves out of the -50 to +20 range.

johnny2678, 2021-10-03:
What field are you using for demand?  Didn't see one in the chemControllersOrp measurement.

[image]

tagyoureit, 2021-10-03:
[image]
It is a more recent addition.  If you haven't pulled in the last few weeks, that'd be why.

johnny2678, 2021-10-03:
grr... was using a custom influx binding so I'm not on the latest and now stuck in diff land.  Probably should write up a script that injects my bindings into the repo InfluxDB.json as it gets updated...

tagyoureit, 2021-10-03:
Better yet, your custom binding should be a different name that you specify in config.json. Then you won't get merge conflicts when you pull.

johnny2678, 2021-10-03:
right, I have that - no merge conflicts, just wasn't adding back in the new fields to my custom binding as you guys were adding them to influx.json.

all is right with the world now.  Will just take a bit for the graph to fill out

[image]

tagyoureit, 2021-10-04:
I just pushed up a change that will allow the ChlorORP to go through the full dosing/mixing/monitoring cycle.  Previously it was only doing dosing/mixing and if the demand was <-20 the mixing would be the full 15 mins.  This worked fine it just meant that there would be emits every single second of every single day no matter what the status.  Now there is a true monitoring cycle where Nixie will monitor the demand every 10s and only kick on the dosing/mixing cycles as needed.

And one more change, @rstrouse corrected my (yet again, I might add) that Nixie should never actually be setting the Chlor Setpoint.  Rather, we are using flags to override the setpoints to either dose at 100% or disable the chlor at 0%.  This is a nuanced change but it might affect what you see in the Grafana logs.  EG you should not be looking at 'chlor swg setpoint' anymore as it isn't what we are controlling.  You should be looking at targetOutput vs currentOutput if you want to see what we expect the chlor to be doing vs. what it is actually doing.  (I also added target output to the influx binding because it was missing.)

johnny2678, 2021-10-04:
looks like it is working:
```
1|poolController  | [10/4/2021, 4:08:33 PM] info: Chlor % of time should be 0%
1|poolController  | [10/4/2021, 4:08:33 PM] info: Chem chlor calculated dosing at 0% and will dose 0Lbs of chlorine over the next 0sec.
1|poolController  | [10/4/2021, 4:08:36 PM] info: [4:08:36 PM] 192.168.5.140 GET /state/all/ {}
1|poolController  | [10/4/2021, 4:08:43 PM] info: Chlor % of time should be 0%
```

re: targetOutput vs currentOutput - what scenario would these two values be different?

tagyoureit, 2021-10-04:
> re: targetOutput vs currentOutput - what scenario would these two values be different?

Here are my last few hours with the new changes (image below).  On the bottom the orange is the currentOutput and the blue is the targetOutput.  It shows the difference in what we are asking for vs. when the chlor actual is dosing the right amount.  It should be max ~30s difference if the OCP is slow on telling the chlor what to output.

Without Nixie ChlorORP the poolSetpoint would normally indicate the targetOutput but since we aren't controlling the values from the internally passed flags poolSetpoint is being overriden and will not equal targetOutput.

[image]

(I can probably downgrade some of those "info" level logging... will look at that.  They are just clogging up the logs.)

tagyoureit, 2021-10-11:
@johnny2678 I'm guessing your readings might be all messed up because of the probe recalibration, but I've tweaked my graph a bit and wanted to share.

[image]

I actually dropped the demand in favor of simplicity and knowing that I never change the setpoint.  And once I inverted the demand axis it mimicked the orp so wasn't necessary.  Any time the ORP falls below the red line (demand >20) you can see the Chlor stays at 100%.  

The overall average is only 19% of the time which sounds a bit low but that's over a full 24 hours/day, not the time the pool is on... but still interesting to see it's on quite a bit less then the 44% (over 24 hours) when I just had the chlor set at 100% all the time.

johnny2678, 2021-10-11:
thanks for posting @tagyoureit 

I can (mostly) match that:
[image]

But as you said, I haven't really given the autoORP algo a fair shake since turning it on.

Ex. #1 in the picture - I chickened out because FC had fallen to 2.5 and SWG dosing hadn't kicked in yet so I upped the setpoint.  ORP was falling that day and dosing would probably have kicked in if I had let it.  Obviously I dropped it back down to 620 after a day.  Also, that day was the day I updated njspc and my influx bindings to include Target Output.

#2 - The way I run my pool is 12 hours on a reasonable RPM (2200), and then another 4 hours on a very low RPM (1450), just to keep the water moving a bit longer.  The RPM is low enough to trigger the "low flow" state in the SWG, which can be observed in #2 where current output never matches target output.

#3 - as you already mentioned, I recalibrated my ORP probe.

Here it is with FC added in:
[image]

ORP is well above my setpoint for now (currently 718 with a 620 setpoint).  FC is at 3 - just took the reading.  Will try and be more hands-off from now on and let it do its thing ;)

tagyoureit, 2021-11-21:
@johnny2678 How has this been working?  Has it had a chance to settle in without your meddling?  :)

johnny2678, 2021-11-21:
woo boy, not even sure where to start.  

1st, I have not been able to give this much attention.  New baby, work, life, etc...

[image]

2nd, I've had a couple of wild swings in FC but I think it was self-inflicted.  My prior dosing routine worked with an ORP setpoint of around 610.  FC wasn't entirely stable using my dosing method, but it kept it above above the danger zone (2-3 FC). 

When I switched to REM dosing, FC tightened up a bit.  I think your dosing is more accurate.  But I never adjusted my 610 setpoint which probably needed to at least be 650-670.  Sure enough, FC dropped below 1 and I had a cloud pool for a week.  After a slam at 12FC, it returned to normal so now I'm just on the hunt for the right ORP setpoint.

[image]

Not helping matters is I've had a few hangs / drop-outs that either keep the system from dosing at all or leave it pegged at 100%.  Not putting this on njspc, I just haven't had time to troubleshoot.  Usually a Pi reboot gets everything squawking again.

Happy to keep posting back with observations but no reason to keep this issue open.  I can always ping you on gitter if I have questions.

bottom line, I'm not going back to my dosing method.  Just need to get SWG dosing via REM dialed in.

tagyoureit, 2021-11-21:
Well congrats on the new baby!  I get the busy part, so we'll close this for now.  Thanks for all the great feedback and testing!

closed 2021-11-21 by tagyoureit