tagyoureit/nodejs-poolController #349: Control EasyTouch and Aquarite independently.

jmillerhyetech, 2021-09-28:
**Describe the bug**
I have an EasyTouch that controls all of my Pentair equipment. I have the Pool Controller running this stuff perfectly. However, I also have a Hayward Aquarite Salt system. Is there a way to control both the EasyTouch and the Hayward simultaneously? The only Chemical controller option to add is an Intellichlor. The intellichlor option kinda works, but it cycles the Aquarite on/off every 15 seconds or so even when set to superchlorinate or 100%. I guess I'm needing some hybrid setup where njsPC can control both EasyTouch and Aquarite.
What happens currently is the Easytouch sends a remote control message to the Aquarite every 15 seconds or so. The remote control light on the Aquarite flashes for about 1 second every 15-ish seconds. When the remote light flashes, the Aquarite turns on for about 5 seconds then turns off and reverts back to the manual setting on the Hayward panel. So my salt system turns on for 5 seconds, then off for 10 seconds (approx).

**Pool Equipment**
 - Controller: EasyTouch, Aquarite
  - Chlorinator: Hayward Aquarite

**Additional context**
Need Nixie to control Aquarite independent of Easytouch on same rs485 bus

tagyoureit, 2021-09-28:
What's happening is that when you add Intellichlor we are telling OCP (outdoor control panel = EasyTouch) to enable control.  What we want to accomplish is to have Nixie (the internal controller in njsPC) to control the Aquarite without the OCP getting in the middle.  Nixie will ping the Aquarite every 3s.

Here are the steps you should take:
1.  Enable the Chlorinator in the Chemistry section.  (I already have one so it is greyed out.)
[image]
2.  On your host server, open up the ./data/poolConfig.json file and copy the chlorinators section into a separate text file.  It will look something like: 
```
  "chlorinators": [
    {
      "id": 1,
      "disabled": false,
      "ignoreSaltReading": false,
      "isActive": true,
      "spaSetpoint": 100,
      "poolSetpoint": 100,
      "address": 80,
      "body": 32,
      "superChlor": false,
      "name": "Intellichlor--40",
      "superChlorHours": 0,
      "type": 31,
      "model": 3,
      "isDosing": true
    }
  ],
``` 
3. Go back to dashPanel -> Settings -> Chemistry and delete the Chlorinator.
4. Stop njsPC.
5. Go back and edit the ./data/poolConfig.json.  The chlorinators section should now be empty:
```
  "chlorinators": [  ],
```
We want to add back the chlorinators data that you copied above with the addition of `"master": 1.  (Make sure you add a comma on the line before it.). It should look like:
```
  "chlorinators": [
    {
      "id": 1,
      "disabled": false,
      "ignoreSaltReading": false,
      "isActive": true,
      "spaSetpoint": 100,
      "poolSetpoint": 100,
      "address": 80,
      "body": 32,
      "superChlor": false,
      "name": "Intellichlor--40",
      "superChlorHours": 0,
      "type": 31,
      "model": 3,
      "isDosing": true,
      "master": 1
    }
  ],
```
 6.  Save the poolConfig.json and restart njsPC.  On the dashPanel homepage you should see the controller and Nixie should be controlling it.

Test this out and let me know how it works.

jmillerhyetech, 2021-09-28:
Works perfectly now.
Thank You!

closed 2021-09-28 by jmillerhyetech

tagyoureit, 2021-10-03:
For some dumb reason I linked to issue #139 but the fix was really for this issue.  This is now in the UI.  You can't change an existing chlorinator, but when you delete it and re-add it you'll be able to choose to have Intellitouch or Nixie control the chlor.