tagyoureit/nodejs-poolController #350: Outbound message error when changing pool temperature

antamy, 2021-09-30:
This started recently - it was there with 7.2 and I just upgraded to 7.4 and it still occurs.  

- click on Pool Setpoint to change the temperature
- raise or lower the temperature in the dialog box
- error dialog appears:
     Error: OutboundMessageError
     
     Message: Invalid payload detected: 165,1,16,33,136,4,NaN,97,7,100,NaN,NaN

     Stack trace:

ApiError: Invalid payload detected: 165,1,16,33,136,4,NaN,97,7,100,NaN,NaN
    at Function.create (/home/pi/nodejs-poolController-7.4.0/controller/comms/messages/Messages.ts:635:19)
    at /home/pi/nodejs-poolController-7.4.0/controller/boards/EasyTouchBoard.ts:1179:34
    at new Promise ()
    at TouchBodyCommands.setHeatSetpointAsync (/home/pi/nodejs-poolController-7.4.0/controller/boards/EasyTouchBoard.ts:1150:16)
    at /home/pi/nodejs-poolController-7.4.0/web/services/state/State.ts:232:44
    at Layer.handle [as handle_request] (/home/pi/nodejs-poolController-7.4.0/node_modules/express/lib/router/layer.js:95:5)
    at next (/home/pi/nodejs-poolController-7.4.0/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/home/pi/nodejs-poolController-7.4.0/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/home/pi/nodejs-poolController-7.4.0/node_modules/express/lib/router/layer.js:95:5)
    at /home/pi/nodejs-poolController-7.4.0/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/home/pi/nodejs-poolController-7.4.0/node_modules/express/lib/router/index.js:335:12)
    at next (/home/pi/nodejs-poolController-7.4.0/node_modules/express/lib/router/index.js:275:10)
    at /home/pi/nodejs-poolController-7.4.0/web/Server.ts:404:25
    at Layer.handle [as handle_request] (/home/pi/nodejs-poolController-7.4.0/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/home/pi/nodejs-poolController-7.4.0/node_modules/express/lib/router/index.js:317:13)
    at /home/pi/nodejs-poolController-7.4.0/node_modules/express/lib/router/index.js:284:7

[replay(1).zip](https://github.com/tagyoureit/nodejs-poolController/files/7260485/replay.1.zip)

rstrouse, 2021-10-01:
Please pull njsPC and dashPanel.  Please make sure you do a hard refresh,  CTRL+F5 or CMD+SHIFT+R depending on your browser platform.

antamy, 2021-10-01:
Ok, thanks.  That’s fixed it, I didn’t realize the client had changed.

rstrouse, 2021-10-02:
Yeah we added the ability to manage the cooling setpoints for Solar and UltraTemp.  This meant that the single body setpoint was no longer viable.

closed 2021-10-02 by rstrouse