tagyoureit/nodejs-poolController #352: Aquacal PoolSync Compatibility?

dfizzo, 2021-10-06:
Hi - is this platform compatible with the AquaCal PoolSync controller? I am looking for best way to to control my AquaCal heat pump (Sunblazer) and AquaCal chlorinators (AutoPilot Nano+). Trying to determine either I need the PoolSync controller or should just go with basic RS-485 approach. Really appreciate any guidance you can provide as I'm at an impasse on this project. Thanks!

@dfizzo￼￼

rstrouse, 2021-10-06:
Some of the AquaCal models support the same protocol as the Pentair UltraTemp and can be controlled over RS485.  Other AquaCal models can be controlled via their fireman's switch.   You will not have any luck with the PoolSync controller and this software.   

For the chlorinator if it uses the similar protocols as the AquaRite or IntelliChlor chlorinators then it should work but if it doesn't perhaps there is something we might be able to do to drive it.  I believe their Autopilot line used to be supported by Jandy using the Hayward protocol so perhaps there is a chance.

My recommendation is to install the software with an RS-485 connection to the Autopilot and the AquaCal using one of the Nixie controller configurations then we can see whether we can wake up the Autopilot to at least tell us what the status is.  If your heat pump model is not compatible with Pentair automation then it should be able to be controlled via the fireman's switch interface.

dfizzo, 2021-10-06:
Thanks so much for the quick response and detailed response! I will proceed on this basis and let you know once I get it all organized. Much appreciated!!

_________________________________________
Daniel M. Fein
Tel: 609-203-2665
Email: ***@***.***

rstrouse, 2021-11-22:
I am closing this for now.  Let us know if you need more help.

closed 2021-11-22 by rstrouse

dfiore1230, 2022-08-15:
I have a similar setup and i haven’t had any luck, i see 100% serial failure rate when trying to communicate with my poolsync / chlorsync30

Happy to debug as necessary, I am keen on getting this supported.

Dave

rstrouse, 2022-08-15:
Please start a new discussion or issue.  This issue has been closed.  It would be good to hook up njsPC to the RS485 bus and control the heater and chlorinator while njsPC is capturing a replay.

mguinness, 2022-10-20:
@dfiore1230 Which baud rate were you using 9600 or 19200?  You should open a new issue and the folks here can help you figure it out.

dfiore1230, 2022-10-20:
I put that project on the shelf for now. Pool is closed for the year so will pick it up in May

rb2k, 2022-11-02:
Just for future archeologists:

The Aquacal website has a section thatm entions RS485:

[Connecting External Controllers to Heat Pump](https://www.aquacal.com/wp-content/uploads/manuals/landingpage/great_big_bopper_installer_hp9/Content/Resources/Topics_Manuals_QS/Mini%20TOC%20Pages/External%20Controllers.htm)

rstrouse, 2022-11-03:
So did you get it to work?

rb2k, 2022-11-03:
I'm still waiting for my RS485 adapter. I mainly wanted to point out the official documentation about that specific aquacal heat pump hardware and the two ways they suggest to hook them up (RS-485 for Smart Bus Controllers or 2 wire).

rstrouse, 2022-11-03:
It will be interesting to see if it talks via the UltraTemp protocol.  njsPC supports it already so define the heater as UltraTemp.

mguinness, 2022-12-09:
Also see [Aquacal heater/chiller heat pump with Intellicenter](https://www.troublefreepool.com/threads/aquacal-heater-chiller-heat-pump-with-intellicenter.266373/) thread on TFP regarding RS-485 availability based on installed CPU.