tagyoureit/nodejs-poolController #356: Docker not working

fgonza2, 2021-10-17:
**Describe the bug**
Ueed docker compose in Ubuntu

**To Reproduce**
Steps to reproduce the behavior:
implemented the docker-compose examples in the wiki

This is what i get:
`Init state for Pool Controller
/bin/sh: git: not found
/bin/sh: git: not found
Pentair EasyTouch System Detected!
RS485 Stats:{ "bytesReceived": 527 "success": 11, "failed": 1, "bytesSent": 60, "collisions": 0, "failureRate": 8.33% }`

tagyoureit, 2021-10-18:
I'm really not a Docker expert.  Have you gone over the [docker wiki](https://github.com/tagyoureit/nodejs-poolController/wiki/Docker) closely?  

These were contributed by @emes & @wurmr.  Tagging them in case they can help.

rstrouse, 2021-10-18:
No but it shows the current version and indicates whether there is an update available.

rstrouse, 2021-10-18:
It is using it to get the latest version.

wurmr, 2021-10-18:
How does it do that? Is it looking at contents of a specific file or is it peeking at git tags?

*Typically* you wouldn't want full blown git client in a docker image as it is a dev tool.

Solutions that come to mind are:

1. Use a different mechnaism to determine latest version (wget is likely in the image).
2. Add git to the container image. This isn't something I'd do in a comercial production project but since this is a hobby project maybe that's ok.

fgonza2, 2021-10-19:
Repro steps are simple, run the docker-compose.yaml that is in the wiki.  Here is mine:

`version: '3.7'  
services:
  poolcontroller:
    image: msmi/nodejs-poolcontroller
    container_name: poolcontroller
    restart: always
    user: node:dialout
       ports:
      - "4200:4200"
    volumes:
      - /home/myuser/containers/pool/config.json:/app/config.json
      - /home/myuser/containers/pool/data:/app/data`

Also the dockerfile DOES include the following line: `RUN apk add git` which should take care of it. 

As @wurmr mentioned, is `msmi/nodejs-poolcontroller` is not really built using the Dockerfile, but some outdated version?

putnam, 2021-12-17:
Piggybacking here, but the sample Dockerfile is currently broken due to referring to the deprecated `python` package. That should be replaced with `python3`.

rstrouse, 2021-12-21:
> Piggybacking here, but the sample Dockerfile is currently broken due to referring to the deprecated python package. That should be replaced with python3.

This change was added to the repo for njsPC and dashPanel

closed 2021-12-21 by rstrouse