tagyoureit/nodejs-poolController #358: doseHistory restore from poolState.json

snagytx, 2021-10-20:
Hello,

I'm not a TS developer but I can know enough to debug the code. It seems that when the doseHistory is restored from the json file, something is incorrect - all the elements are restored with 0 value.

This is the troubleshooting that I did:

Added a console.log in the DataLoggerEntry constructor as seen in the screenshot:
[image]

This is the console.log output:
[image]

When it returns to ChemicalDoseState, which calls the above code in the super, the `this` object is still empty and the element values are not transferred from the extended class:
[image]

I'm not really sure why this happens. Maybe this is something basic that I'm missing.

The issue is really easy to reproduce.
1) Ensure that your njPC instance is running for a while and dosing was reported to it.
2) Call http://localhost:4200/state/chemController/1/doseHistory - ensure that you have a few elements in the dosingHistory , it doesn't matter if it's ph or orp.
3) Restart your njPC instance.
4) Call http://localhost:4200/state/chemController/1/doseHistory - you should see correct elements in dosingHistory array - but you don't/

I'm using an IntelliChem as a chemContoller and using the latest source code from git.

This is my ph doseHistory section:

```
        "doseHistory": [
          {
            "id": 1,
            "method": "auto",
            "start": "2021-10-20T07:56:20.849-0500",
            "end": "2021-10-20T07:57:21.011-0500",
            "chem": "acid",
            "setpoint": 7.5,
            "demand": 0,
            "level": 7.61,
            "volume": 0,
            "status": "completed",
            "volumeDosed": 63,
            "time": 0,
            "timeDosed": 29
          },
          {
            "id": 1,
            "method": "auto",
            "start": "2021-10-20T07:35:52.305-0500",
            "end": "2021-10-20T07:36:50.709-0500",
            "chem": "acid",
            "setpoint": 7.5,
            "demand": 0,
            "level": 7.63,
            "volume": 0,
            "status": "completed",
            "volumeDosed": 67,
            "time": 0,
            "timeDosed": 31
          },
          {
            "id": 1,
            "method": "auto",
            "start": "2021-10-20T07:15:14.146-0500",
            "end": "2021-10-20T07:16:18.650-0500",
            "chem": "acid",
            "setpoint": 7.5,
            "demand": 0,
            "level": 7.64,
            "volume": 0,
            "status": "completed",
            "volumeDosed": 72,
            "time": 0,
            "timeDosed": 33
          }
        ],

```

snagytx, 2021-10-20:
This is what the call returns:

[image]

And this is how it's in the dashboard:
[image]

Please see attached the backup as well:
[backup.zip](https://github.com/tagyoureit/nodejs-poolController/files/7383789/backup.zip)

And this is how the poolState.json looked before I restarted the njPC
[poolState.txt](https://github.com/tagyoureit/nodejs-poolController/files/7383799/poolState.txt)

As for the log files, I see that they are generated, but they are not read - I added breakpoints in the code and they don't seem to be read 
[chemDosage_acid.log](https://github.com/tagyoureit/nodejs-poolController/files/7383805/chemDosage_acid.log)
[chemDosage_orp.log](https://github.com/tagyoureit/nodejs-poolController/files/7383806/chemDosage_orp.log)

rstrouse, 2021-10-20:
ChemicalDoseState is a derived class of DataLoggerEntry.  On line 417 or so you will see a line like
```
let o = extend(true, this, obj);
```
Add a console.log(this); right after that statement.  Then lets see what it outputs.

Just to be sure what version of node are you running?

snagytx, 2021-10-20:
I noticed the same issue on `msmi/nodejs-poolcontroller` - which uses `v14.17.6`

snagytx, 2021-10-20:
This is from container:

Before `docker-compose restart poolcontroller`
[image]

After `docker-compose restart poolcontroller`
[image]

rstrouse, 2021-10-21:
Well I'm not a docker user but I do know that node 16 hasn't even started active LTS support until the 26th and I am currently running 12 as 14 had serial port issues as well.  Something may be up with the extend libraries and node still.

snagytx, 2021-10-21:
I just used your Dockerfile, updated the base image from `node:ltd-alpine` to `node:12-alpine`, checked the output of `node -v` and it's `v12.22.7` - tested the problem - the issue persists.

I'm 99% sure that you should reproduce the issue is you put my json file - if you need I can send you a more recent one.

rstrouse, 2021-10-21:
I copied your doseHistory into my poolState.json then shutdown and restarted a few times.  No matter what I get the following.
[image]

Am I missing a step?  The only difference is that I am not running in a docker container.

snagytx, 2021-10-21:
This is the poolState.txt file that I use. I just change the `start` and `end` dates to today. To make sure that they are in the last 24 yours.

[poolState.txt](https://github.com/tagyoureit/nodejs-poolController/files/7391958/poolState.txt)

I'm available most of the day, CST, ping me and we can schedule a remote session when it's convenient to both of us.