tagyoureit/nodejs-poolController #364: poolController does not reconnect to EW11

snagytx, 2021-11-15:
**Describe the bug**
Using the socat functionality, if for any reason there is a network issue and the connection to EW11 drops then poolController don't detect and doesn't try to reconnect.

EW11 is configured to be the server to accept clients and feed them the messages.

Running  poolController as a dockler container build using the Dockerfile provided in this project from the source code from https://github.com/snagytx/nodejs-poolController (This repo contains the same changes as in tagyoureit plus the changes in pull request #359 - this pull is request is important for IntelliChem)

**To Reproduce**
No known steps to reproduce.

**Expected behavior**
Upon connection drop, poolController should detect and try to reconnect to the server.

**Screenshots**
[image]
To show the the last message was received - timezone is CST

**Packet Capture**
Not applicable.

**Pool Equipment**
 - Controller:  IntelliCenter
 - Controller Model: IntelliCenter i5PS
 - Controller 1.047
 - Pump(s) manufacturer and model: IntelliFlow VST & WisperFlow
 - Chlorinator: IntelliChem
 - Heater(s): gas
 - Chemical controller: IntelliChem
 - Valves: 2 x Intellivalve

**Desktop (please complete the following information):**
 - OS: MacOS - not really relevant
 - Browser Chrome
 - Version latest

**Additional context**

The Console log in DEBUG mode.
[consoleLog(2021-11-14_09-36-39).log](https://github.com/tagyoureit/nodejs-poolController/files/7535523/consoleLog.2021-11-14_09-36-39.log)
The timezone is UTC

rstrouse, 2021-11-16:
Interesting.  I an not seeing the socket drop in the console.log file you included.  Here is a sequence when I disconnect a socket based RS485 adapter.  
```
[11/16/2021, 3:42:56 PM] info: Net connect (socat) end event was fired
[11/16/2021, 3:42:56 PM] info: Net connect (socat) closed : raspberrypi:9801
[11/16/2021, 3:43:06 PM] error: Net connect (socat) Connection: Error: connect ECONNREFUSED 192.168.1.222:9801. Retry in 10 seconds
[11/16/2021, 3:43:06 PM] info: Net connect (socat) closed due to error: raspberrypi:9801
[11/16/2021, 3:43:16 PM] info: [3:43:16 PM] 192.168.1.234 GET /app/config/log {}
[11/16/2021, 3:43:16 PM] info: [3:43:16 PM] 192.168.1.234 GET /config/all?null {}
[11/16/2021, 3:43:16 PM] info: [3:43:16 PM] 192.168.1.234 GET /state/appVersion?null {}
[11/16/2021, 3:43:16 PM] info: [3:43:16 PM] 192.168.1.234 GET /app/config/options/backup?null {}
[11/16/2021, 3:43:16 PM] info: Net connect (socat) connected to: raspberrypi:9801
[11/16/2021, 3:43:16 PM] info: Net connect (socat) Connection connected
[11/16/2021, 3:43:16 PM] info: Net connect (socat) ready and communicating: raspberrypi:9801
[11/16/2021, 3:43:22 PM] info: [3:43:22 PM] 192.168.1.234 PUT /app/logger/setOptions {"app":{"logToFile":false}}
```

However, the keepAlive functionality apparently does not work correctly for sockets in Node because there also was not a timeout idle message in your log.  So it appears I may need to wrap up my own no activity timeout.

snagytx, 2021-11-17:
Right, not sure what is happening. Once I restart the poolcontroller container everything works for a few hours.

If I point to a socat port I get:

```
11/17/2021, 12:02:34 AM] info: Net connect (socat) end event was fired
11/17/2021, 12:02:34 AM] info: Net connect (socat) closed : 10.10.10.79:9802
```

Once I stop the socat process.

rstrouse, 2021-11-17:
Do a pull.  I synthesized an idle timeout to destroy and reopen a port when it doesn't have activity.

snagytx, 2021-11-17:
Done - I'll monitor the changes.

snagytx, 2021-11-18:
The issue just detected again, please see attached the docker console log no error messages:

```
poolcontroller              | [11/17/2021, 11:25:47 PM] debug: Invalid enumeration: val = false map = {}
poolcontroller              | DOSING STATUS === 1
poolcontroller              | [11/18/2021, 1:18:45 AM] info: [1:18:45 AM] 10.10.10.139 GET /state/all?null {}
poolcontroller              | [11/18/2021, 1:18:45 AM] info: New socket client connected gMfXy_19C0wNzqAHAAAF -- 10.10.10.139
poolcontroller              | [11/18/2021, 1:18:46 AM] info: [1:18:46 AM] 10.10.10.139 GET /state/all?null {}
poolcontroller              | [11/18/2021, 2:08:07 PM] info: [2:08:07 PM] 10.10.10.21 GET /state/chemController/1 {}
poolcontroller              | [11/18/2021, 2:08:08 PM] info: [2:08:08 PM] 10.10.10.21 GET /state/chemController/1/doseHistory?null {}
poolcontroller              | [11/18/2021, 2:10:41 PM] info: [2:10:41 PM] 10.10.10.21 GET /state/all?null {}
poolcontroller              | [11/18/2021, 2:10:41 PM] info: New socket client connected 0DJ-o7vEXEnPT2eHAAAH -- 10.10.10.21
poolcontroller              | [11/18/2021, 2:10:41 PM] info: [2:10:41 PM] 10.10.10.21 GET /state/all?null {}
poolcontroller              | [11/18/2021, 2:11:22 PM] info: [2:11:22 PM] 10.10.10.21 GET /config/all?null {}
poolcontroller              | [11/18/2021, 2:11:22 PM] info: [2:11:22 PM] 10.10.10.21 GET /config/options/general?null {}
poolcontroller              | [11/18/2021, 2:11:23 PM] info: [2:11:23 PM] 10.10.10.21 GET /app/options/interfaces?null {}
poolcontroller              | [11/18/2021, 2:11:24 PM] info: [2:11:24 PM] 10.10.10.21 GET /config/options/rs485?null {}
poolcontroller              | sendRS485PortStats set to true
poolcontroller              | [11/18/2021, 2:11:29 PM] info: [2:11:29 PM] 10.10.10.21 PUT /app/rs485Port {"enabled":true,"netConnect":true,"rs485Port":"/dev/ttyUSB0","netPort":8899,"netHost":"10.10.10.64"}
poolcontroller              | [11/18/2021, 2:11:29 PM] info: Net connect (socat) successfully closed: 10.10.10.64:8899
poolcontroller              | [11/18/2021, 2:11:29 PM] info: Net connect (socat) connected to: 10.10.10.64:8899
poolcontroller              | [11/18/2021, 2:11:29 PM] info: Net connect (socat) Connection connected
poolcontroller              | [11/18/2021, 2:11:29 PM] info: Net connect (socat) ready and communicating: 10.10.10.64:8899
poolcontroller              | [11/18/2021, 2:11:29 PM] debug: Invalid enumeration: val = false map = {}
poolcontroller              | DOSING STATUS === 1
poolcontroller              | [11/18/2021, 2:11:30 PM] info: Checking njsPC versions...
poolcontroller              | [11/18/2021, 2:11:30 PM] info: Checking njsPC versions...
poolcontroller              | [11/18/2021, 2:11:31 PM] debug: Invalid enumeration: val = false map = {}
poolcontroller              | DOSING STATUS === 1
poolcontroller              | [11/18/2021, 2:11:31 PM] verbose: sunriseLocal:12.799489727044133 sunsetLocal:23.393355920361003 Calculating Heliotrope Valid
```

I then went in the dashboard "Settings"->"Controller"->"RS485 Port" and just clicked on "Save Port" - no changes on the page - everything started to work again.

rstrouse, 2021-11-18:
Interesting.  I don't see an attempt to close the connection until you do it from the PUT on 2:11;29 PM.  This means that the adapter is probably still sending bytes but they don't contain any signatures that we recognize.  If it were silent then we would see the restart sequence after 10 seconds of silence.  One way to check this is to look at the traffic on the port.  Do you have a way to see this outside of njsPC?  We only report bytes when we our message signatures.

snagytx, 2021-11-18:
OK, I'll start a tcpdump process with a command like:

`tcpdump -i eth0 'port 8899' -w port_8899.pcap` 

then I'll be able to open the pcap file in wireshark. EW11 listens on 8899 - hence the port filter in tcpdump.

snagytx, 2021-11-18:
OK, the issue just happened. Please see attached the tcpdump file. There was no network traffic when the njsPC was hanging. I stopped tcpdump and started it again - it captured nothing until I triggered manually a new connection to be rebuilt.

Rename the file to .pcat and open it with WireShark to see the netwok traffic.
[port_8899.txt](https://github.com/tagyoureit/nodejs-poolController/files/7565907/port_8899.txt)

I updated the node command to add `--inspect:0.0.0.0:9229`, to debug the app. But it seems that in docker the .map files are not generated. Can I get home help to update https://github.com/tagyoureit/nodejs-poolController/blob/master/Dockerfile and have it generate the .map files? I'm looking to see if the app is hanging, or get the heap dump, or the  threads and the call stacks. Hopefully the inspector would give me more details.

rstrouse, 2021-11-18:
The app isn't hanging.  It is simply not detecting the condition where there is no traffic on the socket.  As far as help with docker, that is not on my list of items I know anything about.  Unfortunately, my test case is not an EW11.  Maybe I need to try to maintain the socket with socat and disconnect RS485.  However, it seems to detect it when data is not coming in on a local comm port.

snagytx, 2021-11-18:
Thanks, I was looking for help on how to update the npm commands to generate the map files, but I think I figured it out - remove `production` from two places.

If you have any places that you want me to analyze while the issue is seen - using the inspector just let me know,

rstrouse, 2021-11-18:
I have seen many posts out there where socket drops don't appear to trigger the event on the socket.   However, could you pull the network connection from the pi to see if it starts an RS485 reconnect after 10 seconds?  It should keep trying while it hears silence.

snagytx, 2021-11-18:
I can't really unplug the network cable. I have a complex setup. On an ESXi server I have a virtual machine that runs many docker containers. njsPC is one of the docker containers.

I did one test - change the port number on the server side -  know it's not the same but it's something I can do:

```
[11/18/2021, 9:58:49 PM] info: Net connect (socat) successfully closed: 10.10.10.64:8899
[11/18/2021, 9:58:49 PM] error: Net connect (socat) Connection: Error: connect ECONNREFUSED 10.10.10.64:8899. Retry in 10 seconds
[11/18/2021, 9:58:49 PM] error: Net connect (socat) Connection: Error: connect ECONNREFUSED 10.10.10.64:8899. Retry in 10 seconds
[11/18/2021, 9:58:49 PM] info: Net connect (socat) closed due to error: 10.10.10.64:8899
[11/18/2021, 9:58:59 PM] error: Net connect (socat) Connection: Error: connect ECONNREFUSED 10.10.10.64:8899. Retry in 10 seconds
[11/18/2021, 9:58:59 PM] error: Net connect (socat) Connection: Error: connect ECONNREFUSED 10.10.10.64:8899. Retry in 10 seconds
[11/18/2021, 9:58:59 PM] info: Net connect (socat) closed due to error: 10.10.10.64:8899
[11/18/2021, 9:59:09 PM] info: Net connect (socat) connected to: 10.10.10.64:8899
[11/18/2021, 9:59:09 PM] info: Net connect (socat) Connection connected
[11/18/2021, 9:59:09 PM] info: Net connect (socat) ready and communicating: 10.10.10.64:8899
```

rstrouse, 2021-11-18:
That looks like what I expected.  The connection doesn't have any data coming through so it closes the connection and restarts it.

rstrouse, 2021-11-19:
I posted up a change that attempts to use the socket inactivity to determine whether it has traffic.  Give it a shot and lets see if it makes any difference.

snagytx, 2021-11-19:
I think these changed did the trick.

I can already see:

```
poolcontroller              | 2021-11-19T00:36:42.435192657Z DOSING STATUS === 1
poolcontroller              | 2021-11-19T00:36:53.460346150Z [11/19/2021, 12:36:53 AM] warn: Net connect (socat) connection idle: 10.10.10.64:8899 retrying connection.
poolcontroller              | 2021-11-19T00:36:53.468171003Z [11/19/2021, 12:36:53 AM] info: Net connect (socat) successfully closed: 10.10.10.64:8899
poolcontroller              | 2021-11-19T00:36:53.482142911Z [11/19/2021, 12:36:53 AM] info: Net connect (socat) connected to: 10.10.10.64:8899
poolcontroller              | 2021-11-19T00:36:53.486625734Z [11/19/2021, 12:36:53 AM] info: Net connect (socat) Connection connected
poolcontroller              | 2021-11-19T00:36:53.489455671Z [11/19/2021, 12:36:53 AM] info: Net connect (socat) ready and communicating: 10.10.10.64:8899
poolcontroller              | 2021-11-19T00:36:53.918297555Z DOSING STATUS === 1
```

```
poolcontroller              | 2021-11-19T00:37:28.916954736Z DOSING STATUS === 1
poolcontroller              | 2021-11-19T00:37:38.915500690Z [11/19/2021, 12:37:38 AM] warn: Net connect (socat) connection idle: 10.10.10.64:8899 retrying connection.
poolcontroller              | 2021-11-19T00:37:38.922484298Z [11/19/2021, 12:37:38 AM] info: Net connect (socat) successfully closed: 10.10.10.64:8899
poolcontroller              | 2021-11-19T00:37:38.981750300Z [11/19/2021, 12:37:38 AM] info: Net connect (socat) connected to: 10.10.10.64:8899
poolcontroller              | 2021-11-19T00:37:38.987874130Z [11/19/2021, 12:37:38 AM] info: Net connect (socat) Connection connected
poolcontroller              | 2021-11-19T00:37:38.990571400Z [11/19/2021, 12:37:38 AM] info: Net connect (socat) ready and communicating: 10.10.10.64:8899
poolcontroller              | 2021-11-19T00:37:39.977660181Z DOSING STATUS === 1
```

```
poolcontroller              | 2021-11-19T00:38:06.231776678Z DOSING STATUS === 1
poolcontroller              | 2021-11-19T00:38:17.356939422Z [11/19/2021, 12:38:17 AM] warn: Net connect (socat) connection idle: 10.10.10.64:8899 retrying connection.
poolcontroller              | 2021-11-19T00:38:23.528063989Z [11/19/2021, 12:38:23 AM] info: Net connect (socat) successfully closed: 10.10.10.64:8899
poolcontroller              | 2021-11-19T00:38:24.676995178Z [11/19/2021, 12:38:24 AM] info: Net connect (socat) connected to: 10.10.10.64:8899
poolcontroller              | 2021-11-19T00:38:24.681892451Z [11/19/2021, 12:38:24 AM] info: Net connect (socat) Connection connected
poolcontroller              | 2021-11-19T00:38:24.684402807Z [11/19/2021, 12:38:24 AM] info: Net connect (socat) ready and communicating: 10.10.10.64:8899
poolcontroller              | 2021-11-19T00:38:30.002025259Z DOSING STATUS === 1
```

Something that I noticed with the latest changes is that it seems to overload EW11 - it might have happened as well - it seems that it might give you too soon - I updated the timeout from 10 to 20 seconds and seems to be better. This might be related to the power of EW11 and my WiFi in that area. I'll keep an eye on everything.

snagytx, 2021-11-19:
My performance seems to have recovered. It was only something temporary and as suspected has nothing to do with your code cahnges.

rstrouse, 2021-11-19:
Excellent.  I also integrated your changes for IntelliChem.  I don't have one so much of that code was blind.

rstrouse, 2021-11-22:
I am going to close this for now.  If you have any more problems with this open another issue.

closed 2021-11-22 by rstrouse