tagyoureit/nodejs-poolController #365: Circuit does not turn off when egg timer expires.

jslabotsky, 2021-11-19:
**Describe the bug**
A circuit with a timer does not turn off when the time runs out.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Settings > Circuits > Aux Circuits
2. Create a Generic circuit with the Egg Timer set to 30 minutes, Save
3. Go to Pumps > Pump 1, add the new circuit and Save.
4. Return to home screen.
5. Click on Pump 1 to open pump settings
6. Click on the button for the circuit you created. The green light will illuminate and remaining time is displayed
7. Wait until time expires. Remaining time is no longer displayed, but the green light is still illuminated and the pump is still running.
8. **NOTE: For purposes of packet capture, I reduced the timer duration to 3 minutes.**

**Expected behavior**
The circuit should turn off, indicated by the green light turning off. If no other circuits or programs are running, the pump should turn off.

**Screenshots**
Circuit configuration.
[image]

Pump configuration.
[image]

Two minutes after timer expired:
[image]

**Packet Capture**
Follow the instructions to complete a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) and attach the resulting zip/log files.
[replay(4).zip](https://github.com/tagyoureit/nodejs-poolController/files/7567117/replay.4.zip)

**Pool Equipment**
 - Controller: None. 
 - Controller Model: n/a
 - Controller Firmware Version: n/a
 - Pump(s) manufacturer and model: Pentair Intelliflo VS
 - Chlorinator: Hayward AquaRite
 - Heater(s): None
 - Chemical controller: None
 - Valves: None
 - Any other relevant equipment:

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Firefox
 - Version 94.0.1

**Additional context**
Add any other context about the problem here.

rock-crusher, 2021-11-21:
I would like to know the answer to your question as well. I just recently got the Nixie set-up and am playing around with it and ran into this very issue. I can’t seem to figure out what needs to get done in order to initiate the egg timer and have it shut off after it times out.

tagyoureit, 2021-11-21:
Pull njsPC and dP.  Should be fixed.

jslabotsky, 2021-11-21:
All fixed. Thank you!

jslabotsky, 2021-11-21:
OK now when I turn on a circuit, my home automation system (Home Assistant using MQTT) is not receiving the circuit status of any circuit as turned on. Do I need to change something in my MQTT topic following this fix?

jslabotsky, 2021-11-21:
Alright I spoke a bit too soon. If the circuit does not have an egg timer defined, it works fine. If an egg timer is defined, my HA system sends the MQTT command and the circuit turns on, but a few seconds later HA shows the switch turned off, as if it has received state data saying the circuit is off.  So the egg timer is working now, but it seems like maybe something has happened to the state reporting. I attached a capture.
[replay(6).zip](https://github.com/tagyoureit/nodejs-poolController/files/7577408/replay.6.zip)

rstrouse, 2021-11-22:
If you use something like MQTT explorer you can see what the state are being sent.  That being said and not knowing what the payload_on/off are used for you spelled endTime entTime on the off.

jslabotsky, 2021-11-22:
Yeah, I actually did catch and correct that typo. The  new capture I uploaded has the typo corrected. I'm still learning how to use MQTT. Will give MQTT Explorer a go and see if I can figure it out.

jslabotsky, 2021-11-22:
OK, I got it working. Verified njsPC/Dashpanel was sending correct stuff. For reasons unknown, with my original approach in HA, the presence of a value in the endTime field made it think the state was "off". I changed up my approach based on something another HA user posted and now it's all good.

From Home Assistant configuration.yaml for the benefit of any other HA users who see this:
  - platform: mqtt
    name: Pool Cleaning Cycle
    icon: mdi:pump
    state_topic: "nixie-singlebody/state/circuits/7/cleancycle"
    value_template: "{{ value_json.isOn }}"
    command_topic: "nixie-singlebody/state/circuits/setState"
    payload_on: '{"id": 7,"isOn": "on"}'
    payload_off: '{"id": 7, "isOff": "off"}'
    state_on: "on"
    state_off: "off"
    optimistic: false
    retain: false
    qos: 0

closed 2021-11-22 by jslabotsky