tagyoureit/nodejs-poolController #366: Help with Configuring the Building Automation Card using REM

rock-crusher, 2021-11-20:
Ok, so I believe I finally have all the requisite software installed and running when the Raspberry Pi starts up. I enabled REM in njsPC as per outlined in the “Configuring the Software” procedure. I am now at the procedure of “Configure the Building Automation Card”. The procedure says to: “from a browser window type http://<ip address of the pi>:8080. A webpage will appear to allow configuration of the devices on the Pi.” This “webpage to allow configuration of the devices on the Pi” does not appear at all, at least not for me. However, as per rstrouse (thanks again sir) http://raspberrypi:8080 will give me access to allow configuration of the REM. Unfortunately, I can’t seem to manipulate any of the fields, but again I have no experience with this and could sure use some help getting this completed. The attached image is what I get, but I can’t seem to do much with it?

[image]

rstrouse, 2021-11-20:
First click the + on the i2c buses list and click create bus.  Do not change the bus number as the default I2c bus is 1.  This will add a tab across the top.  After you do this select that tab and follow these instructions.

https://github.com/rstrouse/relayEquipmentManager/wiki/Sequent-Microsystems-Boards#sequent-building-automation-hat-mega-bas

rstrouse, 2021-12-13:
I am closing this for now.  If you have any other questions go ahead and submit another issue.

closed 2021-12-13 by rstrouse