tagyoureit/nodejs-poolController #367: Nixie “Schedules” start/end timer settings not responding as expected

rock-crusher, 2021-11-21:
Issue:  In the nixie scheduler when setting the on/off times the minutes are not changeable while in the 12:00pm hour. When you try, it automatically reverts back to 12:00pm. For instance, if you set the time as 12:37pm the system automatically resets the time back to12:00pm for some reason. All the other hours 1-11 am or pm you can set any minute you desire.  
I’m sure it’s just something I’m doing wrong.
Any Ideas?

[image]

rstrouse, 2021-11-21:
I’ll have a look.  Perhaps we have a validation issue with the field.

I assume this means you have figured out all the hardware setup.

rock-crusher, 2021-11-21:
Hi again rstrouse,
When I type it in, the system just automatically defaults it back to 12:00pm.
As far as getting it all figured out… lol, far from it. I’m just playing around with the relays right now on my work bench getting familiar with the circuits and scheduling in Nixie. I’m able to schedule the relays and get them to fire when I want, but then something glitches and things start going wonky. I’ll lose control of a relay and can’t get it to work again. I’ll refresh everything, reboot, delete the circuits/schedules start over and add them new again and nothing, I can’t get even one now to engage. Excuse me; I can get them all to work manually in REM by clicking on the relay buttons, but it’s like Nixie somehow starts acting erratic and disassociates itself. I can’t figure out the egg timer. Not sure quite how to initiate it, and when I think I do, it never shuts off no matter how the timing parameters are set. Another thing I haven’t figured out is the buttons in Nixie’s settings/appearance tab. I can click on them, but they do nothing I’m aware of.
Once again I must be doing something wrong, maybe you can throw me a bone or two.
Thank you rstrouse  

[image]

[image]

[image]

[image]

rstrouse, 2021-11-21:
**Issue 1**
I pushed up a fix for the time selection weirdness.  There was a bug in the handling for the 12 hour clock were it was not honoring the minutes.

**Issue 2**
For your relay, lets make sure we are looking at the same thing.   Are you saying that when you click on circuit 3 in dashPanel it does not change the little indicator to green?  If it does then it is sending the signal to REM and REM is responding that it processed the request.

If you are looking at the little LEDs on the Sequent board, the Sequent 8 relay board has the screen printing backwards on the indicator LEDs on the board.  At the time of printing the numbering got printed on the other side of the board.  What this did is make the relay 1 light the relay 8 led and relay 7 light the relay 2 led.

For troubleshooting you can open up the `pm2 monit` then select the njsPC item on the monitor.  Then watch the console to see if there are any errors when you press circuit 3 on the dashPanel.

**Issue 3**
@tagyoureit pushed some changes to fix the egg timer.  If you want to know more about what an egg timer does it is in this section of the wiki
https://github.com/tagyoureit/nodejs-poolController/wiki/Nixie-Controller-Setup#circuits----aux-circuits

rstrouse, 2021-11-21:
The buttons in the last screenshot are to change the way dashPanel displays the sections of the display.  Simply drag them to the location and order where you want them to display.  For instance, in your case you do not have a filter pressure sensor so simply drag the filters bubble to the hidden items list.  I also always drag the virtual circuits bubble there as well.

rock-crusher, 2021-11-21:
Hi again,
Well, I tried everything you suggested to no avail; the only thing that it seemed to fix is that I now have control of circuit #3 again. Problem still exists for the egg timer, it never turns off. The 12:00pm issue still exists too.
Funny thing with the egg timer is when the button is pushed it engages and the green light turns on firing the correct relay, but the countdown on the button starts out at 3hrs 50min even though I have it set for only 1min.
By the way, I have tried the above suggestions you have outlined twice with no success and yes I rebooted along with the hard refresh of the browser.

[image]

[image]

[image]

[image]

rstrouse, 2021-11-21:
That is the correct behavior for the timer because you started it in the middle of the schedule so it will work off the schedule time.

rock-crusher, 2021-11-21:
rstrouse, Lol… makes perfect sense! I feel like such a dumb ass. Also, the 12:00 issue is now working great!! (wonder why it took so long for it to take effect on my end?).  
Anyway, I’ll continue on with my quest. As you know this is all extremely new for me, but let me tell you a great learning experience and although frustrating at times (my dumb ass) it is super fun too!
Thanks again for taking the time to help me.

rstrouse, 2021-11-22:
No worries happy to help.  Some of this stuff ends being obvious when you have the goggles that were required to write the functionality then when somebody not wearing them looks at it you realize where the gaps are in function and documentation.  

The reason the 12:00 issue finally fixed itself is likely because you didn't get a hard refresh on the browser and it was using the cached script.  These have an expiration date set on them so eventually the browser asked for the new files with the new script in it.  dashPanel is a dynamic set of pages that are built on the fly so the cache mechanisms on the express web server are not able to appropriately look for cached files.

I suspect that most folks would have difficulty with the wiring but muddle their way through the configuration.   I spent more time documenting (how to) on the wiring part but less time on configuration.  This is partially, because the concepts are similar to commercial offerings out there.  

Bear in mind there is definitely the possibility for software errors since the source of all errors is code and Nixie is made up mostly of code.  Add to that neither I or @tagyoureit have all the equipment we rely a lot on folks providing feedback on their equipment and take some educated guesses when we need to.

rock-crusher, 2021-11-22:
Back again, as you knew I would be… lol.
Maybe a little help with getting my VSF pump setup? I’ve connected the RS485 cable to the pump and Building Automation Card, Green on A & yellow on B terminal. The jumpers on the card are in place, the RS485 pass through is in place. I keep getting the: Unable to open RS485 port message?
I know there is some procedure for this, so if you could just send me in the right direction.
Thank you

rstrouse, 2021-11-22:
Did you change the settings on the pi as indicated here.
https://github.com/rstrouse/relayEquipmentManager/wiki/Sequent-Microsystems-Boards#rs-485-tab

If you did then navigate to the Controller -> RS485 tab in the dashPanel configuration screens.  Enter ‘/dev/ttyS0’ as the port name.

rock-crusher, 2021-11-22:
Yes the pi has already been changed, and I keep getting the Unable to open RS485 port message.

[image]

[image]

rstrouse, 2021-11-22:
First lets verify the following
1. The jumpers are installed on the MEGA-BAS
[image]
2. The MEGA-BAS is configured for RS485 Pass thru
[image]
3. After you reboot the pi run the following command.  It should report that this device exists on the pi
`ls /dev/ttyS0`
[image]

rock-crusher, 2021-11-22:
Yes, it is set up exactly as you show it above.

rock-crusher, 2021-11-22:
I have been configuring all of this via wifi. Could this have anything to do with why I keep receiving the Unable to open RS485 port /dev/ttyS0 ?

rock-crusher, 2021-11-22:
Also, I don’t have the disc battery installed in the MEGA-BAS, is this a problem? Is it necessary?
The kit I bought didn’t have one included. What kind of battery is it when I go to purchase one?

rstrouse, 2021-11-22:
First some answers.
WiFi should not matter.
The battery is a CR2450 but it is for the RTC on the board so it should not matter.

I have a couple of questions

Did you set up an RDP or VNC connection to access the pi?  
Is this a Pi3 or Pi4?
Just to be sure when you perform the command `ls /dev/ttyS0` it returns the part in yellow like above?

Things to try
Turn Bluetooth off on the pi (it will be the little Bluetooth icon on the pi taskbar)?

Run the command `ls -l /dev/ttyS0` and paste the results here.

Run the command `groups pi` and paste the results here.

If the port exists lets try to see if there is any traffic on the port.  Hit ctrl + c to exit.
`od -x < /dev/ttyS0`

rock-crusher, 2021-11-22:
- _Did you set up an RDP or VNC connection to access the pi?_

Yes, RDP, but I’m not doing the set-up using that. I have a keyboard, mouse & monitor directly connected to the PI4.

- Is this a Pi3 or Pi4?

PI4

- _Just to be sure when you perform the command ls /dev/ttyS0 it returns the part in yellow like above?_

Yes it does, attached image of result.

- _Things to try
Turn Bluetooth off on the pi (it will be the little Bluetooth icon on the pi taskbar)?_

It has always been off, attached image.

- _Run the command ls -l /dev/ttyS0 and paste the results here._

Attached image

- _Run the command groups pi and paste the results here._

- _If the port exists lets try to see if there is any traffic on the port. Hit ctrl + c to exit._
_od -x < /dev/ttyS0_

Attached image

[image]

[image]

[image]

[image]

rock-crusher, 2021-11-22:
Does the pump (VSF) need to be configured in some special way from its menu, or is it just plug-n-play? 
I couldn’t find anything scrolling through its menu regarding this.

rstrouse, 2021-11-22:
It will be just plug and play.  Nixie will send the appropriate commands via RS485.  Under the pumps section in the configuration you add the pump as a VSF and tell it which circuits/features should make the pump run at which speed.

rock-crusher, 2021-11-22:
Yes, I’ve done all that and set-up a schedule for it as well. When the schedule dictates for it to turn on “Spa” as I have it set-up for, the spa button green light turns on in the "features" window and it also shows the amount of time left for it to run. Problem is, the actual pump does nothing.
Any other thoughts, suggestions?

rstrouse, 2021-11-22:
This is because we have not been able to get your RS485 port working.  I am researching now why you might be getting a permissions issue opening the port.

When Nixie starts communicating with the pump the VSF display will change to "Display Not Active" and you will be controlling the pump through Nixie over RS485.

rock-crusher, 2021-11-22:
Understood, 
Thank you for all your efforts!

rock-crusher, 2021-11-23:
Funny thing now, when I check the RS485 port on the controller tab I now get this: attached image
As opposed to yesterday when I got this: attached image
It still doesn’t work, but can what is circled in red be changed to get it working? 

[image]

[image]

rstrouse, 2021-11-23:
It cannot change it if it cannot access it.

rock-crusher, 2021-11-23:
My bad, just realized the port name was changed from Local to Network. I must have done it when screwing around.

rstrouse, 2021-11-23:
No worries, I am trying to figure out how to change the port permissions so that they stick.  They really made this overly complicated when they added bluetooth to the pi.

rstrouse, 2021-11-23:
Can I get you to enter the following command then try to save the port

`sudo chmod 0666 /dev/ttyS0`

If this works it will only work until the next reboot.  Not sure why your port would not have all the permissions.

rock-crusher, 2021-11-23:
Just tried it, unfortunately it did not work 
[image]

[image]

rstrouse, 2021-11-23:
Just for shucks can I get you to change the port name to `/dev/serial0` then try to save it.  It is so very odd that the permissions are not correct for this.  I have tried repeatedly to make a pi do what yours is doing but I cannot seem to get it to fail using `/dev/ttyS0` when bluetooth is disabled and the MEGA-BAS is set to pass thru on the RS485 tab.

rock-crusher, 2021-11-23:
yes, it saved, now what?

rstrouse, 2021-11-23:
Are you seeing traffic now?  Do you have the VSF pump defined in the configuration?

rock-crusher, 2021-11-23:
No traffic, nothing has changed.
Yes, VSF pump is defined in the configuration.

rstrouse, 2021-11-23:
Run `ls -l /dev/serial*` and post the results here.

rock-crusher, 2021-11-23:
[image]

rstrouse, 2021-11-23:
Run the following commands
`cd ~/nodejs-poolController`
`git pull`

Wait until dashPanel reconnects then select the hamburger menu.  

Then navigate to the Logging tab and press Capture Replay then Begin Capture on the dialog.

After you do that go to the RS485 Port and save the port.

Go back to the Logging tab and press Cancel Capture.  It will download a file to your browser.  Upload that file here.

rock-crusher, 2021-11-23:
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/7591164/replay.zip)

rstrouse, 2021-11-23:
Ok so it is trying to send messages to the pump but the pump is not responding.  Flip the green and yellow wires going to the pump.

rock-crusher, 2021-11-23:
OK, did that and still nothing has changed.

tagyoureit, 2021-11-23:
Make sure the pump isn't running a schedule; try to put it in Filter 1 and try again.

rock-crusher, 2021-11-23:
"try to put it in Filter 1 and try again."
Going to need help with that

rock-crusher, 2021-11-23:
should I put the wires back, green on A and yellow B?

tagyoureit, 2021-11-23:
On your pump, how has it been running in standalone mode?  On a schedule?

rock-crusher, 2021-11-23:
no schedule

rstrouse, 2021-11-23:
A should be yellow and B should be green.  

Lets try something different.  I am surprised that the pump does not say "Display Not Acitve"

`cd ~`
`git clone https://github.com/SequentMicrosystems/megabas-rpi.git`
`cd megabas-rpi`
`sudo make install`

Then after all this has finished run the following command and shoot  a screenshot.
`megabas 0 rs485rd`

Then issue the following command
`megabas 0 rs485wr 0 9600 1 0 1`

rock-crusher, 2021-11-23:
[image]

rstrouse, 2021-11-23:
Hmmm.... that is exactly what I would have expected to see if it was all configured properly so nothing there.

I have installed this board 3 times and all 3 times the RS485 connection was on `/dev/ttyS0`.   There were not any issues with permissions.   Lets me look inside your `/boot/config.txt`.  You upload that file here from your raspberry pi?

rock-crusher, 2021-11-23:
[config.txt](https://github.com/tagyoureit/nodejs-poolController/files/7591598/config.txt)

rock-crusher, 2021-11-24:
I’m a little confused about the conflicting RS485 color code.
In the below image it says green to A, yellow to B, but you have me the opposite?

[image]

rstrouse, 2021-11-24:
Yes the original documentation that we got this information from was backwards.  We need to update the wiki.

rock-crusher, 2021-11-24:
> Ok so it is trying to send messages to the pump but the pump is not responding. Flip the green and yellow wires going to the pump.

If it is sending messages to the pump, then could it be something wrong with my pump? 
It does work when controlled directly from the panel screen mounted on top of the motor itself.

rstrouse, 2021-11-24:
Alright this is so confounding.  Something about the pi OS is not right and all the goofy recommendations on the web are brutally complicated.  We may be better off re-imaging the OS and starting over.

I do not think there is anything wrong with your pump.  While it is writing to the port I am not so sure that the message is actually making it to the pump.  All of the gymnastics that we have been performing wrt the port permissions makes me think that the messages make it to the transmit buffer they end up blowing out into the ether.

How do you feel about re-imaging the SD card and starting over?

rock-crusher, 2021-11-24:
I'm ok with what ever you recommend, but will need your help getting it done.

rstrouse, 2021-11-24:
So I assume you bought a pi kit with a pre-loaded SD card.  It should also have an adapter to put the SD card in with a USB port.  Pull the SD card out of the pi and insert it into this adapter.

So are you a Mac or a Windows user?

rock-crusher, 2021-11-24:
Windows

rstrouse, 2021-11-24:
Ok so follow these instructions to get the image installed on the Raspberry Pi.  Once you do that we need to hook the pi up to a mouse, keyboard, and monitor to install XRDP.

https://projects.raspberrypi.org/en/projects/raspberry-pi-setting-up/2

rstrouse, 2021-11-24:
Once you get the pi back up and running with a keyboard, mouse, and monitor.  

Run the following commands so you can get access to it from your pc.  Set the basic settings for the pi and hook up a network connection to it.  Then run the following commands.

`sudo apt-get update`
`sudo apt-get upgrade`
`sudo apt-get install xrdp`

After you do this you should be able to access your pi using Remote Desktop Connection (rdp).  Open rdp and enter raspberrypi into the computer field and press connect.  Answer yes to the prompts and the login screen for the pi should appear.  You are now operating the pi headless so you can stop using the monitor, keyboard and mouse.
[image]

Next you will want to perform the install of the software starting with the pre-requisites.
https://github.com/tagyoureit/nodejs-poolController/wiki/DIY-Standalone-Nixie-Pool-Controller#install-software-prerequisites

When you get to the part where you are selecting the interfaces.  Disable the Serial Console and enable the Serial Port.

Once you get to that point go ahead and configure the MEGA-BAS and your relay card and let me know when you get to that point.

rock-crusher, 2021-11-24:
Ok, did all that, now just start everything all over again in DIY Standalone Nixie Pool Controller?

rock-crusher, 2021-11-24:
sorry the page didn't refresh, disregard my above comment

rock-crusher, 2021-11-24:
> Once you get the pi back up and running with a keyboard, mouse, and monitor.
> 
> Run the following commands so you can get access to it from your pc. Set the basic settings for the pi and hook up a network connection to it. Then run the following commands.
> 
> `sudo apt-get update` `sudo apt-get upgrade` `sudo apt-get install xrdp`
> 
> After you do this you should be able to access your pi using Remote Desktop Connection (rdp). Open rdp and enter raspberrypi into the computer field and press connect. Answer yes to the prompts and the login screen for the pi should appear. You are now operating the pi headless so you can stop using the monitor, keyboard and mouse. [image]
> 
> Next you will want to perform the install of the software starting with the pre-requisites. https://github.com/tagyoureit/nodejs-poolController/wiki/DIY-Standalone-Nixie-Pool-Controller#install-software-prerequisites
> 
> When you get to the part where you are selecting the interfaces. Disable the Serial Console and enable the Serial Port.
> 
> Once you get to that point go ahead and configure the MEGA-BAS and your relay card and let me know when you finish that configuration.

Ok, I'll get started on this, thanks.

tagyoureit, 2021-11-24:
I have the Mega-BAS card and have not yet enabled RS-485 pass thru.  I tried it and ran into the same issue.  Ironically, there seem to be thousands of similar issues out there but I figured out a quick and easy way to do this that I didn't see mentioned anywhere.  Give this a try and let me know if this works for you.

1. Run `sudo raspi-config`.
2. Select 3 `interface options`
3. Select P6 `serial port`
4. **Important one** - Would you like a login shell to be accessible over serial?  NO.
[image]
5. Would you like the serial port hardware to be enabled?  Yes.
[image]

Final settings should look like:
[image]

When I answered yes to the login shell (easy to do!) that is where this problem is caused.  I tried it about 3 times and was able to replicate the `Unable to open port` each time I answered yes to the login shell question.

[150 more replies]

closed 2021-12-14 by rstrouse