tagyoureit/nodejs-poolController #368: IntelliCenter / MasterTemp RS485 replay capture

bad350trk, 2021-11-23:
MasterTemp w/ RS-485 comms connected to Intellicenter.  Sending in a capture and information to see the conversation between MT and IC.

Here's the replay.
[replay 1916_22NOV2021.zip](https://github.com/tagyoureit/nodejs-poolController/files/7585539/replay.1916_22NOV2021.zip)

And here's the description of what I did.
[Heater comms test 1.docx](https://github.com/tagyoureit/nodejs-poolController/files/7585549/Heater.comms.test.1.docx)

Thanks!

--Jeff

rstrouse, 2021-11-23:
So I have done a quick breeze through of the messages and have witnessed a couple of things.

First this is similar to the UltraTemp protocol.  There are only 2 messages involved where the OCP sends a request to the heater and the heater responds with its status.

The first message is a message sent from the OCP to the heater with a 12 byte payload.  It is a request message that contains a command to the heater and the setpoints.  These are all contained within the first 3 bytes.  The message below is one where it is calling for heat to the spa setpoint (96).  I assume the other setpoint (pool) is set to 78.  
`[255, 0, 255][165, 0, 112, 16, 112, 12][2, 78, 96, 0, 0, 0, 0, 0, 0, 0, 0, 0][2, 81]`

The message is broken down like this.
`[255, 0, 255]` = Message preamble
`[165, 0, 112, 16, 112, 12]` = The first 2 bytes are the controller and protocol values followed by the destination  (112:Heater at address 1) and the source (16:OCP).  The next byte is the action byte (112:Request status) finally followed by the length of the payload in the next section.

So here is what we know of the payload starting with the first byte index.
`[2, 78, 96, 0, 0, 0, 0, 0, 0, 0, 0, 0]`
2 = Heat to the spa setpoint 96.
78 = Pool setpoint
96 = Spa setpoint

We know nothing about the remaining bytes.  These could be nothing or they could mean something.  The punchline is that they never change.

So that was the command message.  The response message is a larger message that has a payload 22 bytes long.  Here is what the changes look like for this message in the capture you sent.
[image]

Btw if you want to play around with the messages you have access to this tool as well.  The basic wiki instructions are here.  The wiki needs a couple of updates related to the filter icon on the header but essentially it will filter messages out of your current list by checking the box next to the message signature.
https://github.com/rstrouse/nodejs-poolController-dashPanel/wiki/Message-Manager

rstrouse, 2021-11-23:
I have also determined that the IntelliCenter status message is changed when the heater is a MasterTemp on RS485.  This is consistent with the way UltraTemp works.  I thought it was weird when I saw it with UltraTemp but it explains why the mode and status indicators in dashPanel are showing off.

rstrouse, 2021-11-23:
Also what heat mode was the pool in when you were capturing?

rstrouse, 2021-11-23:
I have added the code for MasterTemp so if you pull njsPC it should now show the proper selections in dashPanel as well as.  We may have some tweaks to do there based upon our collaboration.

bad350trk, 2021-11-23:
Sorry, I don't understand your question about heat mode.  If you mean Pool or Spa, it was in Spa mode.

bad350trk, 2021-11-23:
I will try a pull later this evening and see what shows up.

rstrouse, 2021-11-23:
The heat mode will be either Off or (MasterTemp or Heater) depending on how Pentair labeled this.  I am guessing that it was Off.

[image]

bad350trk, 2021-11-23:
Understand now.   Yes, the pool was set to off.

bad350trk, 2021-11-23:
Tried a pull in njsPC.  I get the following error:

error: cannot open .git/FETCH_HEAD: Permission denied

rstrouse, 2021-11-24:
Are you in the nodejs-poolController directory.
`cd ~/nodejs-poolController`
`git pull`

If you are then the issue is that the user you are logged in as does not have write access to the directory.  You can use
`sudo git pull` but I would check to see who the owner of the nodejs-poolController directory is.

I hate linux permissions.

rstrouse, 2021-11-24:
Run the following commands to set the ownership.
`cd ~/nodejs-poolController`
`sudo chown -R $(whoami) .git/`

bad350trk, 2021-11-24:
Sweet!  That worked!  Set ownership and then the pull worked.

bad350trk, 2021-11-28:
I have not had time to do a thorough run through to try and capture the errors on the RS-485 bus.  But I have noticed a few things while using the heater this weekend.

1 - There is no way for me to turn on the heater from either pool or spa selection. When I select the pop-up to turn on the heater, this is all I see...
[image]
When I click on the OFF radio button in this state, nothing happens.

2 - I turned on the heater using the IC web client.  This is how njsPC shows the heater.  I was expecting the Heat Mode to show - MasterTemp, and the Heater Status to show - ON.
[image]

With the heater running, I can turn it off using the pop-up OFF radio button.  That function works just fine.
[image]

3 - The heater type is inverted between the IC and njsPC.  If the IC has the heater type configured as MasterTemp, it shows up as Max E Therm in njsPC (see screen shot below).
[image]

To confirm this, I deleted the heater in njsPC and added a new heater and selected MasterTemp from the type drop down.  When I went to the IC web client, the heater shows up as Max E Therm.

I did not grab any captures for this, I certainly can if necessary, if the previous one does not have sufficient info.

Thanks,

--Jeff

rstrouse, 2021-11-28:
Please pull njsPC.  I fixed the differential between Max-E-Therm and MasterTemp several days ago.  This was inverted but it should not be inverted anymore.  This will also make it so that the heat modes and status are incorrect.  Then after you do that go to the hamburger menu and navigate to the System tab and hit Reload Config.  This will ask IntelliCenter to send its configuration again.

bad350trk, 2021-11-28:
Hmmm... Doesn't seem to change anything.  Just did a git pull on both njsPC and dashpanel, then reloaded the configuration from IC.  Still seeing the same issues as mentioned above.

rstrouse, 2021-11-28:
I will need a replay when it is reloading the config.  So select the Capture Replay then go to the Reload Config.

bad350trk, 2021-11-28:
I think I got it.  I selected the capture configuration reload option and then started the capture.  Once the system showed Ready again, I cancelled the capture.

[replay 28NOV2021.zip](https://github.com/tagyoureit/nodejs-poolController/files/7614312/replay.28NOV2021.zip)

--Jeff

rstrouse, 2021-11-28:
Pull njsPC then reload the config from the System Tab.  I switched the identifiers around.

bad350trk, 2021-11-28:
Alright.  Did that.  That corrected the MasterTemp / Max E Therm on the heater settings
[image]

Here's the screen before I turned the heater on.
[image]

The bottom radio button, I am assuming is supposed to be the ON button.  I attempted to turn on the heater using that button, but nothing happened.  After turning on the heater via the IC, here is what I see...
[image]

Here is the most recent capture loading the configuration after the last pull, in case you need that.

[replay 28NOV2021 (2).zip](https://github.com/tagyoureit/nodejs-poolController/files/7614580/replay.28NOV2021.2.zip)

Let me know if there is anything else you need me to do/look at.

Thanks so much for looking into this!  

--Jeff

rstrouse, 2021-11-28:
I need a capture of you changing the heat mode from off to on then back to off again.  I do not need the config.

bad350trk, 2021-11-29:
Started capture, turned heater on, about 45 seconds later turned it off, about 45 seconds turned it on, about 45 seconds later turned it off.  Stopped capture.

[replay Heater off_on.zip](https://github.com/tagyoureit/nodejs-poolController/files/7614677/replay.Heater.off_on.zip)

--Jeff

rstrouse, 2021-11-29:
Pull njsPC.  I added the sequences for the heat modes related to MasterTemp.

bad350trk, 2021-11-30:
Very nice!  That's it!  

Able to turn on/off the heater now, and heat mode shows up as MasterTemp.
[image]

Still planning on running through a sequence where I cause alarm trips.  Hopefully this upcoming weekend.

Thanks!

--Jeff

rstrouse, 2021-11-30:
Is the heater currently running?  I would have expected to see the flames when it is on.
[image]

rstrouse, 2021-11-30:
Pull dashPanel and hard refresh your browser.  The flames should appear for MasterTemp now when it is heating.

bad350trk, 2021-11-30:
Oh.  Didn't realize there was supposed to be a flame symbol.  Ok.  Did a pull on dashPanel.  Do not know what a hard refresh means on my browser, but completely closed out and reopened, and don't see the flame.  So restarted the computer and reopened and still don't see the flame.  Still looks like this.

[image]

--Jeff

rstrouse, 2021-11-30:
Ctrl + F5 on windows and something else on a Mac.  If this is an iPad you have to clear the browser cache.  But be careful it isn't very smart so you have to go in and delete the data for the individual web site.  Apple can be so stupid sometimes.

bad350trk, 2021-11-30:
Windows machine.  CTL + F5 didn't seem to have any effect.  Had to go into settings and clear the browser cache.  Showing up now!

[image]

[image]

Thanks!

--Jeff

rstrouse, 2021-12-14:
I am going to close this for now.  If you have any more issues please submit another issue.

closed 2021-12-14 by rstrouse